espressif/esp-idf #8673: Update docker images with modern version of git

BrianPugh, 2022-03-28:
## Problem Description

The `espressif/idf:release-***` docker images have a really old version of git installed: `2.17.1`. When using github-actions, this causes issues with `actions/checkout@`, because it expects at least git version `2.18`. I imagine a lot of CI uses this docker image + `actions/checkout`; this mostly becomes an issue when performing a recursive checkout, which is also pretty common for esp-idf projects.

## Solution
Add something like the following to the Dockerfile:

```
apt update
apt install -y software-properties-common
add-apt-repository -y ppa:git-core/ppa
apt update
apt install -y git
```

dobairoland, 2022-03-28:
Hi @BrianPugh. Are you sure you are using some recent version of the images. 

It is based on Ubuntu 20.04 LTS (https://github.com/espressif/esp-idf/blob/master/tools/docker/Dockerfile) which includes git 2.25 (https://packages.ubuntu.com/focal/git).

BrianPugh, 2022-03-28:
I'm using pretty recent builds of older versions of esp-idf.

Examples:
* `espressif/idf:release-v4.3` hash `e72b0c436522cf656c7f602fb5cb63f7b663e52208348df71f77272fe66c3ac0`; not sure when it was built.
* `espressif/idf:release-v4.2` hash `67a2e8eab6e68203709e1a027cb12c8f1ae7be9b141ab277a6df9f4d6cd8a0c6` was built 10 days ago.

Example terminal output:
```
MacBook-Air:~$ docker run -it --entrypoint /bin/bash espressif/idf:release-v4.2
Unable to find image 'espressif/idf:release-v4.2' locally
release-v4.2: Pulling from espressif/idf
11323ed2c653: Pull complete
6b9089b15a20: Pull complete
70441b0719cc: Pull complete
78603fa54015: Pull complete
03cf2c99a9a5: Pull complete
Digest: sha256:67a2e8eab6e68203709e1a027cb12c8f1ae7be9b141ab277a6df9f4d6cd8a0c6
Status: Downloaded newer image for espressif/idf:release-v4.2
root@4f50e4c50f76:/# git --version
git version 2.17.1
```

BrianPugh, 2022-03-28:
similarly:

```
MacBook-Air:~$ docker run -it --entrypoint /bin/bash espressif/idf:release-v4.3
Unable to find image 'espressif/idf:release-v4.3' locally
release-v4.3: Pulling from espressif/idf
11323ed2c653: Already exists
4a4c9b084b51: Pull complete
d43bbcad5be2: Pull complete
88c4897f6d60: Pull complete
06c637b2175f: Pull complete
Digest: sha256:c4418f7f2444cd6c6d2cc5dadad5d9c51b7224f4ac84cafc60b6f956bdd78c84
Status: Downloaded newer image for espressif/idf:release-v4.3
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
root@fdbaa0323f3d:/# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.6 LTS"
root@fdbaa0323f3d:/# git --version
git version 2.17.1
```

```
MacBook-Air:~$ docker run -it --entrypoint /bin/bash espressif/idf:release-v4.4
Unable to find image 'espressif/idf:release-v4.4' locally
release-v4.4: Pulling from espressif/idf
7c3b88808835: Pull complete
3fdc0b77fa26: Pull complete
22494b364e91: Pull complete
3b5a9db3ab7f: Pull complete
99f1c9124044: Pull complete
Digest: sha256:fc1548524e7a19980e8c4290deb81f0d77cb1348fdf8c3224960fe09e321354a
Status: Downloaded newer image for espressif/idf:release-v4.4
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
root@56129e1c0654:/# git --version
git version 2.25.1
root@56129e1c0654:/# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.4 LTS"
```

so it actually seems fine v4.4 onward

dobairoland, 2022-03-28:
Yes, the images for ESP-IDF v4.3 and previous versions are based on Ubuntu 18.04 LTS and git 2.17 is included there (https://packages.ubuntu.com/bionic/git). The update was done for ESP-IDF v4.4.

dobairoland, 2022-03-29:
BTW, the checkout action doesn't require git 2.18. 

> When Git 2.18 or higher is not in your PATH, falls back to the REST API to download the files.

I don't think we should make this upgrade for already released ESP-IDF versions. It could introduce breaking changes for others in their test queues.

I believe you are able to workaround the issue as the checkout action is not dependent on a very recent git. In the worst case, you can do the proposed update in your test jobs.

BrianPugh, 2022-03-29:
The issue is is that you cannot specify `actions/checkout` to recursively checkout your project with git <2.18

```
Error: Input 'submodules' not supported when falling back to download using the GitHub REST API. To create a local Git repository instead, add Git 2.18 or higher to the PATH.
```

Generally, many esp-idf projects contain submodules. This became broken (either by github, git, actions/checkout, or espressif) sometime within the last 3 months.

dobairoland, 2022-03-29:
Ok, this seems to be a legitimate reason. I reopen for further investigation. Maybe updating just `git` for v4.3 is a safer approach as it was originally proposed.

tomassebestik, 2022-04-20:
Hi @BrianPugh,
We tried to implement your suggestion in `v4.3`, but unfortunately a recent security update has appeared in the latest git release (`git v2.35.2`) which breaks our build in `release/v4.3`, see [example here](https://github.com/espressif/idf-extra-components/runs/6002229852?check_suite_focus=true).

This is probably related to this [vulnelabilty in git, which has been fixed in git 2.35.2](https://github.blog/2022-04-12-git-security-vulnerability-announced/.)
So this is breaking change, which is a problem in a closed release.

We tried to fix it by installing some older version into the Docker image from that mirror (ppa:git-core/ppa), but only these versions are available for Ubuntu 18:04:
```
root@623914593d3d:/# apt-cache madison git
       git | 1:2.35.2-0ppa1~ubuntu18.04.1 | http://ppa.launchpad.net/git-core/ppa/ubuntu bionic/main amd64 Packages
       git | 1:2.17.1-1ubuntu0.10 | http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages
       git | 1:2.17.1-1ubuntu0.10 | http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages
       git | 1:2.17.0-1ubuntu1 | http://archive.ubuntu.com/ubuntu bionic/main amd64 Packages

```

None of this is suitable - we need at least version `2.18` and lower than `2.35.2`. There might be a way to build it from source, but that can cause a lot of other problems and is not a long-term solution.

More information also here: https://github.com/actions/checkout/issues/760

We **had to revert it to the original state** and looking for another suitable solution. 
We apologize for the inconvenience and thank you for your understanding.
Tomas

closed 2023-01-11 by dobairoland