espressif/esp-idf #8674: I suggest that PPPOS routines include an introduction to the PPPOS protocol and how to get the account password

CAIJUN111, 2022-03-29:
I found that many people only know about PPPOE,but not PPPOS.By the way, I have a question, how to get the account password of PPPOS login.

chegewara, 2022-03-29:
Login to what?
https://en.wikipedia.org/wiki/Point-to-Point_Protocol

CAIJUN111, 2022-03-29:
I mean
 how do I `Set username for authentication` and `Set password for authentication` aboutPPPOS

 

 https://en.wikipedia.org/wiki/Point-to-Point_Protocol

AxelLin, 2022-03-29:
http://git.savannah.gnu.org/cgit/lwip.git/tree/doc/ppp.txt#n180

CAIJUN111, 2022-03-29:
Thank you for your help.

 
http://git.savannah.gnu.org/cgit/lwip.git/tree/doc/ppp.txt#n180

chegewara, 2022-03-29:
https://espressif.github.io/esp-protocols/esp_modem/index.html

```
#if !defined(CONFIG_EXAMPLE_MODEM_PPP_AUTH_NONE) && (defined(CONFIG_LWIP_PPP_PAP_SUPPORT) || defined(CONFIG_LWIP_PPP_CHAP_SUPPORT))
    esp_netif_ppp_set_auth(esp_netif, auth_type, CONFIG_EXAMPLE_MODEM_PPP_AUTH_USERNAME, CONFIG_EXAMPLE_MODEM_PPP_AUTH_PASSWORD);
#endif
```

CAIJUN111, 2022-03-29:
I want to know the CONFIG_EXAMPLE_MODEM_PPP_AUTH_USERNAME,CONFIG_EXAMPLE_MODEM_PPP_AUTH_PASSWORD,

I only have a regular SIM card, so if I  need password authentication？

chegewara, 2022-03-29:
It all depends on your SIM card provider/LTE service provider. Usually you can check GRPS setting on android smartphone, when you put that card and internet setting will be setup, but for example in my country or Australia there is no need to use any user/pass for PPPoS authentication, just the correct operator name in `"IP"` AT command.

CAIJUN111, 2022-03-29:
Thank you for your patient answer,

Why the routine I execute can't detect the provder,I skiped the verification,but I can not connnect net.

chegewara, 2022-03-29:
You should start from the very basics. Install on esp32 just a simple `SerialPassthrough` example from arduino. The use AT commands to diagnose your LTE device, networks etc. In meantime you may learn a lot about it. We cant help you with it, because we dont know what network you are using, what country you are living at, and even then we dont know what settings are used for that network. And most basic, you didnt provide any logs or example code you are using.

CAIJUN111, 2022-03-29:
This is the log of my project:
I (27) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (27) boot: compile time 14:06:37
I (27) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 3, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1a8c4h (108740) map
I (149) esp_image: segment 1: paddr=0002a8ec vaddr=3ffb0000 size=02a88h ( 10888) load
I (154) esp_image: segment 2: paddr=0002d37c vaddr=40080000 size=02c9ch ( 11420) load
I (160) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=68bbch (428988) map
I (319) esp_image: segment 4: paddr=00098be4 vaddr=40082c9c size=0748ch ( 29836) load
I (331) esp_image: segment 5: paddr=000a0078 vaddr=50000000 size=00010h (    16) load
I (337) boot: Loaded app from partition at offset 0x10000
I (337) boot: Disabling RNG early entropy source...
I (351) cpu_start: Pro cpu up.
I (351) cpu_start: Starting app cpu, entry point is 0x40081010
0x40081010: call_start_cpu1 at D:/A/esp-idf-v4.3.1/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (366) cpu_start: Pro cpu start user code
I (366) cpu_start: cpu freq: 160000000
I (366) cpu_start: Application information:
I (370) cpu_start: Project name:     4g_cj
I (375) cpu_start: App version:      1
I (379) cpu_start: Compile time:     Mar 29 2022 15:12:58
I (386) cpu_start: ELF file SHA256:  db1e44e2d31ceb3c...
I (391) cpu_start: ESP-IDF:          v4.3.1-dirty
I (397) heap_init: Initializing. RAM available for dynamic allocation:
I (404) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (410) heap_init: At 3FFB4530 len 0002BAD0 (174 KiB): DRAM
I (416) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (423) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (429) heap_init: At 4008A128 len 00015ED8 (87 KiB): IRAM
I (437) spi_flash: detected chip: gd
I (440) spi_flash: flash io: dio
I (445) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (455) uart: queue free spaces: 30
E (635) esp-modem: esp_dte_handle_line(114): handle line failed
W (635) pppos_example: Unknow line received: +COPS: 0

I (685) pppos_example: Module: SIMCOM_SIM800L
I (685) pppos_example: Operator: 
I (685) pppos_example: IMEI: 865472039096123
I (695) pppos_example: IMSI: 460019795514402
I (725) pppos_example: rssi: 9, ber: 0
I (765) pppos_example: Battery voltage: 3461 mV
I (925) pppos_example: Modem PPP Started
I (925) pppos_example: PPP state changed event 259
I (925) pppos_example: PPP state changed event 262
I (1205) pppos_example: PPP state changed event 263
I (1205) pppos_example: PPP state changed event 265
I (1365) pppos_example: PPP state changed event 262
I (7365) pppos_example: PPP state changed event 268
I (7365) pppos_example: PPP state changed event 256
I (7365) esp-netif_lwip-ppp: Connection lost
I (7365) pppos_example: Modem Disconnect from PPP Server

The attachment is the routine I run.
My sim card is unicom. (A regular mobile phone card) .
My 4G modem is sim800l.

chegewara, 2022-03-29:
Here is set of events for PPP state change, so you can see whats going on:
```
    NETIF_PPP_PHASE_DEAD         = NETIF_PP_PHASE_OFFSET +  0,
    NETIF_PPP_PHASE_MASTER       = NETIF_PP_PHASE_OFFSET +  1,
    NETIF_PPP_PHASE_HOLDOFF      = NETIF_PP_PHASE_OFFSET +  2,
    NETIF_PPP_PHASE_INITIALIZE   = NETIF_PP_PHASE_OFFSET +  3,
    NETIF_PPP_PHASE_SERIALCONN   = NETIF_PP_PHASE_OFFSET +  4,
    NETIF_PPP_PHASE_DORMANT      = NETIF_PP_PHASE_OFFSET +  5,
    NETIF_PPP_PHASE_ESTABLISH    = NETIF_PP_PHASE_OFFSET +  6,
    NETIF_PPP_PHASE_AUTHENTICATE = NETIF_PP_PHASE_OFFSET +  7,
    NETIF_PPP_PHASE_CALLBACK     = NETIF_PP_PHASE_OFFSET +  8,
    NETIF_PPP_PHASE_NETWORK      = NETIF_PP_PHASE_OFFSET +  9,
    NETIF_PPP_PHASE_RUNNING      = NETIF_PP_PHASE_OFFSET +  10,
    NETIF_PPP_PHASE_TERMINATE    = NETIF_PP_PHASE_OFFSET +  11,
    NETIF_PPP_PHASE_DISCONNECT   = NETIF_PP_PHASE_OFFSET +  12,
```
To me it looks ok. It is connected to network and then finally disconnected.

CAIJUN111, 2022-03-29:
Is there a possibility that the routine has configuration about 2G.Like that:

https://github.com/espressif/esp-idf/issues/7098#issue-909860034

Alvin1Zhang, 2022-04-11:
@CAIJUN111 Thanks for reporting and updates, would you please help share if any updates for the issue? Thanks.

closed 2022-05-06 by Alvin1Zhang