espressif/esp-idf #8675: identifier "CONFIG_LOG_MAXIMUM_LEVEL" is undefined C/C++(20)

chipweinberger, 2022-03-29:
branch: release/v4.4
commit: c29343eb94d2f2ca17b3a5b38c82452e556147f2

I keep seeing this intellisense error in VSC. Very annoying. Code compiles fine.

[image]

Looking at the code, it seems this is only used in the linker? I don't get how this is supposed to work.
https://github.com/espressif/esp-idf/search?q=LOG_MAXIMUM_LEVEL

c_cpp_properties.json looks like this:
[image]

chipweinberger, 2022-03-29:
I've edited the includePath to include `${workspaceFolder}/build/config/`, and that seems to have fixed it.

0xjakob, 2022-03-29:
@chipweinberger I'm afraid we can't help you here because this repository is about IDF problems. You'll probably receive more help if you move the issue over to the [vscode-esp-idf-extension](https://github.com/espressif/vscode-esp-idf-extension) repository.

Not everyone using IDF is also using VSCode (e.g. me).

chipweinberger, 2022-03-29:
Right. But there is still a possibility it is caused by an esp-idf build system issue.

chipweinberger, 2022-03-29:
I think you are right that this is not an ESP-IDF issue though. 

VSC seems to be picking up `/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/wear_levelling/test_wl_host/sdkconfig/sdkconfig.h ` as the sdkconfig.h to use for intellisense, so I think I need to change the order or includes.

chipweinberger, 2022-03-29:
https://esp32.com/viewtopic.php?t=23943

closed 2022-03-29 by chipweinberger