espressif/esp-idf #8676: Example code ABC, there is a strange problem. Maybe it's a bug.

SuhaylZhao, 2022-03-29:
Hello everyone，

I need help，I ran this example, and it's very normal.
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/ledc/ledc_basic
[image]

But when I changed gpio5 to gpio33, a strange problem occurred. The low level of PWM output is not 0V, but 1.65v, which is about 50% of VCC. Ask for help. Do I need any settings?

[image]

songruo, 2022-03-29:
Hi @SuhaylZhao,

Please provide more details about the module/chip you are using, and the Kit version (for WroverKit/PicoKit/DevKitC).

SuhaylZhao, 2022-03-29:
Hi @songruo ，

The Module is Wrover-E.

Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 44:17:93:e3:77:70

Thanks！ 

In addition, I found that this may be a hardware problem. I use another esp32 board and burn the same code. Gpio33 pin can also output normal PWM waveform. 

But I still feel confused. Why is there such a problem? In one of my products, I connected gpio33 to a buzzer, so I urgently want to know the answer to the problem.

songruo, 2022-03-29:
@SuhaylZhao Yes, I understand this is the module you are using. But what about the development kit (the one has the module attached on)? If there is a 32kHz crystal, it will be on the development kit board (like [this one](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html?highlight=esp32%20wrover%20kit#functional-description)). Or are you developing your own board for the module?

SuhaylZhao, 2022-03-29:
Hi @songruo ,

I use freenove's esp32-wrover-dev, and there is no crystal oscillator outside.

[image]
[image]

songruo, 2022-03-29:
@SuhaylZhao I tried but failed to find freenove esp32_wrover_dev board circuit diagram. But as you said, this should be a hardware problem with the development board. If you want to figure out the reason, you probably need to ask freenove for the schematic.

SuhaylZhao, 2022-03-30:
Hi @songruo ,

The circuit diagram can be found here: 
https://github.com/Freenove/Freenove_ESP32_WROVER_Board/blob/main/Datasheet/ESP32_Schematic.pdf

Alvin1Zhang, 2025-02-17:
Thanks for reporting and sorry for late reply, since this should be a hardware problem with the development board, you probably need to ask freenove for help, or find help on the forum https://esp32.com, will close the issue. Thanks.

closed 2025-02-17 by Alvin1Zhang