espressif/esp-idf #8679: Modbus Slave RTU ESP32S3 not responding

Archy404, 2022-03-29:
After setting up a modbus slave with the provided API, it does not recognize any input at the defined Rx Pin.

The Input is a correct Modbus Frame. Checked with a logic analyzer.
 Following Configurations have been made in sdkconfig

#define CONFIG_MB_UART_PORT_NUM 1
#define CONFIG_MB_UART_BAUD_RATE 9600
#define CONFIG_MB_UART_RXD 16
#define CONFIG_MB_UART_TXD 17
#define CONFIG_MB_UART_RTS 15
#define CONFIG_MB_COMM_MODE_RTU 1
#define CONFIG_MB_SLAVE_ADDR 1

The Slave does not react to modbus Frames at pin 16. (Or any other Pin declared as Rx). Baudrate and address are as described in the configuration.
It did not work with my custom code (which worked flawlessly on the esp32s2) so i tried the code example provided by espressif. The code example also does not work. It simply does not react to Modbus Frames recieved at the RXD Pin. Configuring a Master with identical Pin-layout does work. I attached a screenshot of the Bits recieved at Rx Pin.

It does not create any Error notifications. No Events are recognized by mbc_slave_check_event();
As mentioned above the code (custom and idf-example) worked flawlessly on the esp32s2.

[image]
.

Archy404, 2022-03-29:
Worked with a different esp32s3. Wanted to delete Issue but am not allowed.

closed 2022-03-29 by Archy404

alisitsyn, 2022-03-29:
@Archy404,

The issue has been checked and it also works on my side with my environment. The issue will be closed automatically. Feel free to reopen.