espressif/esp-idf #8680: BLE Advertising multiple packets at once

krupis, 2022-03-29:
I have been trying to implement such feature but without any luck. I have asked for help in ESP32 general discussion forums but no one was able to confirm whether this is possible or not.

I have started with gatt_server_service_table example from esp-idf:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server_service_table

I use this service for communication between the mobile phone APP and my device. 
```
        case ESP_GATTS_CREAT_ATTR_TAB_EVT: {
            ESP_LOGI(GATTS_TABLE_TAG, "The number handle = %x",param->add_attr_tab.num_handle);
            if (param->create.status == ESP_GATT_OK){
                if(param->add_attr_tab.num_handle == SERVICE_IDX_NB) {
                    memcpy(app_table, param->add_attr_tab.handles,
                    sizeof(app_table));
                    esp_ble_gatts_start_service(app_table[SERVICE1]);

                }else{
                    ESP_LOGE(GATTS_TABLE_TAG, "Create attribute table abnormally, num_handle (%d) doesn't equal to SERVICE_IDX_NB(%d)",
                         param->add_attr_tab.num_handle, SERVICE_IDX_NB);
                }
            }else{
                ESP_LOGE(GATTS_TABLE_TAG, " Create attribute table failed, error code = %x", param->create.status);
            }
        break;
    }
```

Now, on top of the above service, I would like to implement periodical iBeacon or Eddystone service. Is that possible or I can only have 1 service at a time?

I would like to have periodical iBeacon advertisement as well as advertisement for my custom BLE app

Weijian-Espressif, 2022-04-05:
@krupis  If you need to send multiple ADV at the same time, esp32C3/S3 support, ESP32 bluetooth version is 4.2, does not support multiple adv

krupis, 2022-04-05:
Thank you very much for your answer. So just to clarify:
This is not possible with the classic ESP32 wroom 32D modules. It only possible with the new S3 boards?

esp-cjh, 2024-02-27:
@krupis With the exception of the ESP32, other BLE chips support this feature. Such as ESP32C3, ESP32S3, ESP32C6, ESP32C2.