espressif/esp-idf #8681: LoadStoreError exception in closed source BLE library

MikitaMinau, 2022-03-29:
## Environment

- Development Kit:      none, custom board
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  ESP32
- IDF version: v4.2.2
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description
LoadStoreError exception happened on Core 0. 
Taking into account backtrace decoding the exception happened in closed source BLE library.
```
0x4001742a: ?? ??:0
0x4004a7b1: ?? ??:0
0x4014a630: r_lld_pdu_check at ??:?
0x40147592: r_lld_evt_move_to_slave at ??:?
0x400469b6: ?? ??:0
0x40019d11: ?? ??:0
0x40055b4d: ?? ??:0
0x4013f3bf: btdm_rom_ref_init at ??:?
0x4013f9d5: btdm_controller_task at ??:?
0x40097561: vPortTaskWrapper at C:/Projects/esp-idf/components/freertos/xtensa/port.c:143
```
The device had mobile client connected via BLE. The device were scanning for iBeacons.
Automatic light sleep was enabled.

### Expected Behavior

LoadStoreError exception should never happen.

### Actual Behavior

LoadStoreError exception happened in BLE library.

### Steps to reproduce

No specific steps. The exception was observed once after 1 hour of work.

### Code to reproduce this issue

Unfortunately, I am not able to share project code.

## Debug Logs

```
(3751987) Ota.c: Written image length 724992
(3752027) Ota.c: Written image length 727040
(3762037) Ota.c: Written image length 728163
(3767857) MotionSensor.c: Measured temperature is 38.41 Celsius
(3767857) TemperatureService.c: Ambient temperature (estimation): 73 Fahrenheit
(3768247) wagz_main.c: Timer event!
(3768247) wagz_main.c: Current date is Fri Mar 25 20:45:08 2022
(3768247) wagz_main.c: Assigned Pet ID: 3342
(3768247) KernelInterface.c: Heap size CAP 8bit 3934132
(3768257) KernelInterface.c: Heap size CAP 32bit 3935080
(3768257) KernelInterface.c: Heap size IRAM 948
(3768267) KernelInterface.c: Heap size EXT RAM 3868136
(3768277) KernelInterface.c: EXT RAM largest free block 3855780
(3768277) KernelInterface.c: EXT RAM law water mark 3816092
(3768287) KernelInterface.c: Heap size INT RAM 66944
(3768287) KernelInterface.c: INT RAM largest free block 57692
(3768297) KernelInterface.c: INT RAM law water mark 55284
(3768307) KernelInterface.c: External RAM heap is valid.
(3768307) KernelInterface.c: Internal RAM heap is valid.
(3784457) GnssManager.c: +QGPSCFG: "estimation_error",9.92,13.71,0.8,91.6
(3784457) GnssManager.c: Got cold fix. Starting accuracy correction
[0;32mI (3784487) BleGatt.c: Select Handle 0x4a: attribute index 34, char index 13, callback 0x0
[0;32mI (3784497) BleGatt.c: Send notification on char update, handle = 0x4a
[0;32mI (3784507) BleGatt.c: Select Handle 0x4a: attribute index 34, char index 13, callback 0x400eaa84
[0;32mI (3784507) LocationService.c: Location data was read by a BLE client and it's allowed not to upload them to the server
[0;32mI (3784627) GnssManager.c: $GPGSV,3,1,11,01,25,041,37,03,26,087,35,06,44,210,31,14,70,094,29,1*6C
[0;32mI (3784627) GnssManager.c: $GPGSV,3,2,11,17,63,343,36,19,53,298,29,24,11,316,31,30,22,181,27,1*6F
[0;32mI (3784627) GnssManager.c: $GPGSV,3,3,11,02,01,219,00,07,02,167,00,13,01,247,00,1*5A
[0;32mI (3784637) GnssManager.c: $GLGSV,2,1,08,70,02,060,00,86,55,007,00,77,38,326,00,76,84,203,00,1*79
[0;32mI (3784647) GnssManager.c: $GLGSV,2,2,08,75,27,153,00,88,04,232,00,87,43,252,00,85,13,037,00,1*71
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.

Core  0 register dump:
PC      : 0x4001742d  PS      : 0x00060833  A0      : 0x8004a7b4  A1      : 0x3ffb5630  
A2      : 0x3ffb2edc  A3      : 0x4002863e  A4      : 0x3ffb4200  A5      : 0x3ffb2ce8  
A6      : 0x3ffb8360  A7      : 0x00000000  A8      : 0x8008cc2c  A9      : 0x3ffb5610  
A10     : 0xffffffff  A11     : 0x3ffc2f48  A12     : 0x00000000  A13     : 0x00060823  
A14     : 0x000000fe  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000003  
EXCVADDR: 0x4002863e  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace:0x4001742a:0x3ffb5630 0x4004a7b1:0x3ffb5650 0x4014a630:0x3ffb5670 0x40147592:0x3ffb5690 0x400469b6:0x3ffb56e0 0x40019d11:0x3ffb5710 0x40055b4d:0x3ffb5730 0x4013f3bf:0x3ffb5750 0x4013f9d5:0x3ffb5770 0x40097561:0x3ffb57a0

```

## Other items if possible

- [ ] sdkconfig file
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8371255/sdkconfig.txt)

esp-cjh, 2024-02-27:
Hi, @MikitaMinau Does this problem still exist with the latest IDF branch?

Alvin1Zhang, 2024-09-06:
Thanks for reporting and sorry for slow turnaround, will close now, feel free to reopen with more updates.

closed 2024-09-06 by Alvin1Zhang