espressif/esp-idf #8682: ESP32-D0W: Low IPerf thoughput with some devices

barkov00, 2022-03-29:
Hi!
I found, that  IPerf shows  low  thoughput and TCP disconnects with some android devices.
This smartphones works nice  with Wi-Fi routers.

IDF v 4.4

Lets see on IPerf  output:

****XIaomi Mi A2 Lite <> ESP32:**
      Interval Bandwidth
   0-   3 sec       0.82 Mbits/sec
   3-   6 sec       0.00 Mbits/sec
   6-   9 sec       0.00 Mbits/sec
   9-  12 sec       0.01 Mbits/sec
  12-  15 sec       2.43 Mbits/sec
  15-  18 sec       2.74 Mbits/sec
  18-  21 sec       0.00 Mbits/sec
  21-  24 sec       0.00 Mbits/sec
[1B][0;33mW (80719) iperf: tcp server recv error, error code: 11, reason: No more processes[1B][0m
[1B][0;32mI (80720) iperf: TCP Socket server is closed.[1B][0m**

**XIaomi Mi A2 Lite <> Zyxel Router:**
```
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0- 3.0 sec  10.9 MBytes  30.4 Mbits/sec
[  4]  3.0- 6.0 sec  11.4 MBytes  31.9 Mbits/sec
[  4]  6.0- 9.0 sec  2.88 MBytes  8.05 Mbits/sec
[  4]  9.0-12.0 sec  8.53 MBytes  23.8 Mbits/sec
[  4] 12.0-15.0 sec  10.5 MBytes  29.2 Mbits/sec
[  4] 15.0-18.0 sec  11.0 MBytes  30.7 Mbits/sec
[  4] 18.0-21.0 sec  11.9 MBytes  33.4 Mbits/sec
[  4] 21.0-24.0 sec  12.0 MBytes  33.5 Mbits/sec
[  4] 24.0-27.0 sec  10.8 MBytes  30.2 Mbits/sec
[  4] 27.0-30.0 sec  11.3 MBytes  31.7 Mbits/sec
```

**Huawei Y5P <> ESP32**
        Interval Bandwidth
   0-   3 sec       59.89 Mbits/sec
   3-   6 sec       61.41 Mbits/sec
   6-   9 sec       65.47 Mbits/sec
   9-  12 sec       65.65 Mbits/sec
  12-  15 sec       66.18 Mbits/sec
  15-  18 sec       63.14 Mbits/sec
  18-  21 sec       66.26 Mbits/sec
  21-  24 sec       66.06 Mbits/sec
  24-  27 sec       66.01 Mbits/sec
  27-  30 sec       65.28 Mbits/sec
   0-  30 sec       64.53 Mbits/sec

Are there any ways to solve this problem, maybe some settings in the menuconfig need to be corrected?

MaxwellAlan, 2022-03-31:
Hi @barkov00 
Can you try disable WiFi AMPDU TX and RX in the Menuconfig for ESP32 and retest.
Thanks.

jack0c, 2022-05-20:
@MaxwellAlan @freakyxue we can provide BA patch here.

barkov00, 2022-05-20:
> @MaxwellAlan @freakyxue we can provide BA patch here.
Hi. We are very interested to know what this patch is.
Does it fix wifi bandwidth issues?

barkov00, 2022-05-23:
> hi @barkov00 Please provide the commit of idf you use.
> 
> I will provide a fix patch based on your commit.
> 
> thanks.

Now i'm use 4.3, but this problem reproduced in this version too.
commit: 62cc246d5fa22960569366c23e11a5609555ccd2
https://github.com/espressif/esp-idf/commit/62cc246d5fa22960569366c23e11a5609555ccd2

MaxwellAlan, 2022-05-23:
Hi @barkov00 
Please replace this patch *.a lib: [esp32_ba_fix_patch_v4.3.zip](https://github.com/espressif/esp-idf/files/8752700/esp32_ba_fix_patch_v4.3.zip)  to your $IDF_PATH/components/esp_wifi/lib/esp32 ，while the libphy.a & librtc.a do not replace. Have a try.
Thanks.

MaxwellAlan, 2022-07-27:
Hi @barkov00

You can get this fix in: https://github.com/espressif/esp-idf/commits/cd13a62952d2a4165c721954b1dfca895c35f68a

Thanks.

Alvin1Zhang, 2022-08-12:
Thanks for reporting, fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/cd13a62952d2a4165c721954b1dfca895c35f68a, feel free to reopen.

closed 2022-08-12 by Alvin1Zhang