espressif/esp-idf #8683: esp32-c3 sd card example error

innomatica, 2022-03-29:
## Environment

- Development Kit:      ESP32-C3-DevKitC-02
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-C3-WROOM-02
- IDF version: v4.4 stable
- Build System:  idf.py
- Compiler version: riscv32-esp-elf-gcc 8.4.0
- Operating System:  Linux
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

`/examples/storage/sd_card/sdspi` puts out following error occasionally.

```
V (997) sdspi_transaction: r1 = 0x11 hw_cmd.r[0]=0xffffffff
I (997) sdspi_transaction: cmd=52, R1 response: function number error
D (1007) sdmmc_cmd: cmd=52, sdmmc_req_run returned 0x102
V (1007) sdmmc_io: sdmmc_io_rw_direct: sdmmc_send_cmd returned 0x102
E (1017) sdmmc_io: sdmmc_io_reset: unexpected return: 0x102
D (1027) sdmmc_init: sdmmc_card_init: sdmmc_io_reset returned 0x102
E (1027) vfs_fat_sdmmc: sdmmc_card_init failed (0x102).
I (1037) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1047) APP: Failed to initialize the card (ESP_ERR_INVALID_ARG). Make sure SD card lines have pull-up resistors in place.

```

### Expected Behavior

Runs example code without error every time.

### Actual Behavior

Roughly one in five cases, it failed to recognize the card. It looks identical to [#6785](https://github.com/espressif/esp-idf/issues/6785) but happens on a different SOC. 

Chances vary. Sometimes more frequently than others. If I switch to a different card (higher capacity), 
it happens a lot frequently, like almost everytime.

### Steps to reproduce

Here is my settings

1. SanDisk Edge 8GB
2. Micro SD card breakout board with SPI connection (3V3 power)
3. ESP32-C3-DevKitC-02
4. Circuit laid out on a breadboard with minimal distance in between (60mm max)
5. NO PULL-UPs (as far as I know ESP32-C3 does not require one)
6. `/storage/sd_card/sdspi` example with following  modification for the GPIOs

GPIO0 - SCLK
GPIO1 - MISO
GPIO2 - MOSI
GPIO3 - CS

### Code to reproduce this issue

See above

## Debug Logs

Most of the time, it works without issue
```
I (888) APP: Mounting filesystem
D (888) sdspi_host: sdspi_host_init_device: SPI2 cs=3 cd=-1 wp=-1
V (898) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (898) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x80E
D (908) intr_alloc: Connected src 19 to int 9 (cpu 0)
V (918) bus_lock: device registered on bus 1 slot 5.
D (918) spi_hal: eff: 400, limit: 80000k(/0), 0 dummy, -1 delay
D (928) spi_master: SPI2: New device added to CS5, effective clock: 400kHz
I (928) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
V (938) sdmmc_cmd: sending cmd slot=1 op=52 arg=80000c08 flags=1c00 data=0x0 blklen=0 datalen=0 timeout=1000
V (948) sdspi_host: sdspi_host_start_command: slot=1, CMD52, arg=0x80000c08 flags=0x40, data=0x0, data_size=0 crc=0x4f
V bus_lock: dev 5 acquired.
D (968) spi_master: device5 locked the bus
V bus_lock: SPI dev changed from -1 to 5
V (968) spi_master: polling trans
V (978) spi_master: polling trans done
V (978) spi_master: polling trans
V (988) spi_master: polling trans done
D (988) spi_master: device5 release bus
V (988) bus_lock: dev 5 released.
V (998) sdspi_transaction: r1 = 0x04 hw_cmd.r[0]=0xffffffff
I (998) sdspi_transaction: cmd=52, R1 response: command not supported
D (1008) sdmmc_cmd: cmd=52, sdmmc_req_run returned 0x106
V (1008) sdmmc_io: sdmmc_io_rw_direct: sdmmc_send_cmd returned 0x106
V (1018) sdmmc_cmd: sending cmd slot=1 op=0 arg=0 flags=20 data=0x0 blklen=0 datalen=0 timeout=1000
V (1028) sdspi_host: sdspi_host_start_command: slot=1, CMD0, arg=0x00000000 flags=0x100, data=0x0, data_size=0 crc=0x4a
V bus_lock: dev 5 acquired.
V (1038) spi_master: polling trans
V (1048) spi_master: polling trans done
V (1048) bus_lock: dev 5 released.
V bus_lock: dev 5 acquired.
D (1058) spi_master: device5 locked the bus
V (1058) spi_master: polling trans

... (omitted in the middle) ...

V (6678) bus_lock: dev 5 released.
V (6688) sdspi_transaction: r1 = 0x00 hw_cmd.r[0]=0xffffff00
V (6688) sdmmc_cmd: cmd response 000000ff 00000000 00000000 00000000 err=0x0 state=0
V (6698) sdmmc_cmd: sending cmd slot=1 op=13 arg=0 flags=1c00 data=0x0 blklen=0 datalen=0 timeout=1000
V (6708) sdspi_host: sdspi_host_start_command: slot=1, CMD13, arg=0x00000000 flags=0x8, data=0x0, data_size=0 crc=0x06
V bus_lock: dev 5 acquired.
D (6718) spi_master: device5 locked the bus
V (6728) spi_master: polling trans
V (6728) spi_master: polling trans done
V (6728) spi_master: polling trans
V (6738) spi_master: polling trans done
D (6738) spi_master: device5 release bus
V (6738) bus_lock: dev 5 released.
V (6748) sdspi_transaction: r1 = 0x00 hw_cmd.r[0]=0xffffff00
V (6748) sdmmc_cmd: cmd response 000000ff 00000000 00000000 00000000 err=0x0 state=0
I (6758) APP: Read from file: 'Hello SS08G!'
I (6768) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (6778) APP: Card unmounted
V (6778) intr_alloc: esp_intr_free: Disabling int, killing handler
I (6788) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6788) gpio: GPIO[1]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6798) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (6808) gdma: del rx channel (0,0)
D (6818) gdma: del tx channel (0,0)
D (6818) gdma: del pair (0,0)
D (6818) gdma: del group 0

```

However, roughly one out of five cases, it produces following log
```
I (857) APP: Initializing SD card
I (857) APP: Using SPI peripheral
D (867) gdma: new group (0) at 0x3fc931b8
D (867) gdma: new pair (0,0) at 0x3fc931e4
D (867) gdma: new tx channel (0,0) at 0x3fc93190
D (877) gdma: new rx channel (0,0) at 0x3fc93204
D (877) spi: SPI2 use gpio matrix.
I (887) APP: Mounting filesystem
D (887) sdspi_host: sdspi_host_init_device: SPI2 cs=3 cd=-1 wp=-1
V (897) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (897) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x80E
D (907) intr_alloc: Connected src 19 to int 9 (cpu 0)
V (917) bus_lock: device registered on bus 1 slot 5.
D (917) spi_hal: eff: 400, limit: 80000k(/0), 0 dummy, -1 delay
D (927) spi_master: SPI2: New device added to CS5, effective clock: 400kHz
I (927) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
V (937) sdmmc_cmd: sending cmd slot=1 op=52 arg=80000c08 flags=1c00 data=0x0 blklen=0 datalen=0 timeout=1000
V (947) sdspi_host: sdspi_host_start_command: slot=1, CMD52, arg=0x80000c08 flags=0x40, data=0x0, data_size=0 crc=0x4f
V bus_lock: dev 5 acquired.
D (967) spi_master: device5 locked the bus
V bus_lock: SPI dev changed from -1 to 5
V (967) spi_master: polling trans
V (977) spi_master: polling trans done
V (977) spi_master: polling trans
V (987) spi_master: polling trans done
D (987) spi_master: device5 release bus
V (987) bus_lock: dev 5 released.
V (997) sdspi_transaction: r1 = 0x11 hw_cmd.r[0]=0xffffffff
I (997) sdspi_transaction: cmd=52, R1 response: function number error
D (1007) sdmmc_cmd: cmd=52, sdmmc_req_run returned 0x102
V (1007) sdmmc_io: sdmmc_io_rw_direct: sdmmc_send_cmd returned 0x102
E (1017) sdmmc_io: sdmmc_io_reset: unexpected return: 0x102
D (1027) sdmmc_init: sdmmc_card_init: sdmmc_io_reset returned 0x102
E (1027) vfs_fat_sdmmc: sdmmc_card_init failed (0x102).
I (1037) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1047) APP: Failed to initialize the card (ESP_ERR_INVALID_ARG). Make sure SD card lines have pull-up resistors in place.
```

igrr, 2022-03-29:
Hi @innomatica, it's a pretty interesting issue, thanks for reporting it!

The log says that the command response in the failure case is r1 = 0x11. According to the SD spec, the bits in the response have the following meaning:

bit 0: in idle state
bit 1: erase reset
bit 2: illegal command
bit 3: command CRC error
bit 4: erase sequence error
bit 5: address error
bit 6: parameter error

So for an SD card, 0x11 status indicates "idle state" + "erase sequence error".

The "erase sequence error" may be reported by the card if an erase command is issued in a state where it can't be used. Looks like somehow the card considers the command which is sent by ESP to be related to an erase operation. There are 3 erase-related SD protocol commands (for SPI mode): 32 (0b100000 binary), 33 (0b100001 binary), 38 (0b100110 binary). Command 52 is 0b110100 binary. So it would take at least two bit errors to confuse command 52 with command 38. It doesn't look very likely, but I can't think of any other explanation why an SD card would report such status.

Could you please examine the SPI bus using a logic analyzer or oscilloscope, ideally also capturing the signals in analog mode? This is to rule out any possible voltage levels issue or a signal integrity issue.

---

The other possibility is that the SD card controller recognizes CMD52 (which is an SDIO command) and instead of responding with "illegal command" (r1 = 0x4) it responds with r1 = 0x11, which in SDIO protocol means "function number error". That is, it recognises command 52, but complains that function 0 is an unknown function. I guess it makes some sense (SDIO function 0 indeed doesn't exist for an SD card) but according to SD specification, an SD memory card should simply respond to an SDIO command with an "illegal command" response.

To check this possibility, could you please try changing this part of the code:
https://github.com/espressif/esp-idf/blob/97fb98a91b308d4f4db54d6dd1644117607e9692/components/sdmmc/sdmmc_io.c#L68-L72
as follows:
```c++
     if (err == ESP_ERR_TIMEOUT || (host_is_spi(card) && (err == ESP_ERR_NOT_SUPPORTED || err == ESP_ERR_INVALID_ARG))) { 
         /* Non-IO cards are allowed to time out (in SD mode) or 
          * return "invalid command" error (in SPI mode). 
          */ 
     } else if (err == ESP_ERR_NOT_FOUND) { 
```

(i.e. allow `ESP_ERR_INVALID_ARG` to be returned for SPI cards)

igrr, 2022-03-30:
Could you try adding a pull-up on MISO line? Some cards set MISO as open-drain output. I know the board we test SD on C3 has pull-up resistors on all lines except CLK, perhaps that's the difference.
(The pin you are using as MISO on an esp32 board may already have a pull-up resistor on the PCB or in the module.)

innomatica, 2022-03-30:
I did put 47K pull up on all 4 signals: DAT0-3

innomatica, 2022-03-30:
Actually if I skipped the pull ups, especially on MISO, the signal looks like this:
[image]

igrr, 2022-03-30:
That looks pretty odd. The MISO line is driven by the card, could you try to probe it somewhere closer to the card? Perhaps there is a loose connection somewhere. I'm struggling to see how something on esp32-c3 driver side could affect the MISO line, since it's an input into esp32-c3 (output of a card).

innomatica, 2022-03-30:
It is a bit confusing but the card is not initialized properly I think, so the MISO line is in high impedance at its both ends. So it is high when the pull up is installed, otherwise it is low. What is really odd to me is the MOSI line. If you look closely you will notice that it becomes floating in the half way as if the GPIO has no enough sink current.

seulater, 2022-04-04:
> It is a bit confusing but the card is not initialized properly I think, so the MISO line is in high impedance at its both ends. So it is high when the pull up is installed, otherwise it is low. What [...]

Sunlu2333, 2022-11-17:
Hello, did you find the solution? I have encountered similar problems.

Sunlu2333, 2022-11-17:
> @Sunlu2333 No, I didn't. I decided to move away from this SoC and chose esp32 instead. The same circuit with the software worked there.

Thank you for your reply, actually, I have tried on ESP32 WROOM 32D and ESP32 WROVOR IE, but I failed. I always got errors as follows. Could you share with me your setup? 
[image]

chegewara, 2022-11-17:
> Micro SD card breakout board with SPI connection (3V3 power)

make sure this is 3.3V module. I have sd card adapter which requires 5V. (actually you could try to use 5V to supply yours)
for example this one
https://github.com/espressif/arduino-esp32/issues/524#issuecomment-647181887

Sunlu2333, 2022-11-18:
> Thank you for your reply, actually, I have tried on ESP32 WROOM 32D and ESP32 WROVOR IE, but I failed. I always got errors as follows. Could you share with me your setup? [image]

Thank you very much for your help, it is really useful to me.

Sunlu2333, 2022-11-18:
> > Micro SD card breakout board with SPI connection (3V3 power)
> 
> make sure this is 3.3V module. I have sd card adapter which requires 5V. (actually you could try to use 5V to supply yours) for example this one [espressif/arduino-esp32#524 (comment)](https://github.com/espressif/arduino-esp32/issues/524#issuecomment-647181887)

Thank you very much for your help, it is really useful to me.

adokitkat, 2024-04-15:
Hi. I tested this example on latest release branch for v4.4 `release/v4.4` (and up to `release/v5.2`) and the example seems to work properly.

Hardware: ESP32-C3-DevKitM-1, SD card breakout board (3.3V), 2GB MicroSD card with SD card adapter

This particular log is from `release/v4.4`:
```log
I (293) example: Initializing SD card
I (293) example: Using SPI peripheral
I (293) example: Mounting filesystem
I (303) gpio: GPIO[1]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (353) sdspi_transaction: cmd=5, R1 response: command not supported
I (463) example: Filesystem mounted
Name: SU02G
Type: SDSC
Speed: 20 MHz
Size: 1886MB
I (463) example: Opening file /sdcard/hello.txt
I (503) example: File written
I (513) example: Renaming file /sdcard/hello.txt to /sdcard/foo.txt
I (513) example: Reading file /sdcard/foo.txt
I (523) example: Read from file: 'Hello SU02G!'
I (523) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (523) example: Card unmounted
I (533) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (543) gpio: GPIO[6]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (553) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
```

Are any of you still experiencing this issue? 

I will close this now and if someone still has this problem, feel free to reopen. Thank you.

closed 2024-04-15 by adokitkat