espressif/esp-idf #8684: Nimble blecent and bleprph coded phy support

iamcalledusman, 2022-03-29:
## Environment

Development Kit: [ESP32-C3-DevKitM-1 v1.0]
Kit version (DevKitM): [v1]
Module or chip used: [ESP32-C3-MINI-1]
IDF version (ESP-IDF v4.4-dev-3224-g23a03a3bd)
Build System: [idf.py]
Compiler version (xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0)
Operating System: [Windows]
(Windows only) environment type: [ESP Command Prompt].
Using an IDE?: [Yes (VSCODE Studio)]
Power Supply: [USB]

## Problem Description

I have questions, I tried (https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/blecent/main/main.c) and https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/bleprph

Above repo's are working with 1M_Phy able but when I set it to coded_phy its behavior is the same as of setting it to 1M_Phy

//Detailed problem description goes here.

1)when I set the default Phy's
[image]

Then I am able to receive the legacy (on nrf connect andriod app )+ extended advertisement packets(using nrf52833 scanner to receive)

2)But when I set this setting, with coded PHY as shown in the image below

[image]

Then still the same behavior as in the previous case. Unable to receive the coded PHY packets. For coded PHY packets reception, I am using nrf52833.

3)In last , i set params.legacy_pdu =0;

then following stack dump error message as mentioned in the image

[image]

Can you please help me in this regard?

Thanks in advance.

### Expected Behavior

I think it should receive the coded phy messages on coded phy scanner when set 

 **params.primary_phy = BLE_HCI_LE_PHY_CODED**

      as mentioned above in the pictures 

### Actual Behavior

Above repo's are working with 1M_Phy able but when I set it to coded_phy its behavior is the same as of setting it to 1M_Phy
 

### Steps to reproduce

1. load the following example into esp32c3 https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/bleprph
2. Set params.primary_phy = BLE_HCI_LE_PHY_CODED and params.secondary_phy = BLE_HCI_LE_PHY_CODED
3.Use the Coded phy scanner app from andriod phone which supports BLE5.0 or use the nrf52833 or any other another BLE5.0 supported controller and load a coded phy scanner app into it.
4.Messages will not be receive on that but when set the default phy's  BLE_HCI_LE_PHY_1M then can view the extended 1M packets 

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

code is the default as in the orignal Repo , only changes i have made is as mentioned above .

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

rahult-github, 2022-04-05:
Hi @iamcalledusman ,

 The sample application provided in sdk has basic settings configured for 1M to work. When coded phy is to be selected, there are different configuration values that need to be modified. i.e. only setting primary / secondary phy will not suffice.  For e.g.  BT pec 5.3 , Vol 4, Part E , Section 7.8.53 cites various conditions like : 
 
 
[image]

 Such conditions would need to be taken care of in the application.  The same is true for the value of legacy_pdu. 

So, in general there are many parameters that are passed to set extended advertising parameters command. There are certain, combinations of values allowed depending upon what configuration you choose. You can refer to the BT spec for detailed information for same. 

For sample, you can try attached patch that will advertise coded phy PDU. But you may need more changes in it for your use case.
[coded_changes.txt](https://github.com/espressif/esp-idf/files/8416727/coded_changes.txt)

Thanks,
 Rahul

iamcalledusman, 2022-04-05:
Hi @rahult-github 

Thank you so much for the details

rahult-github, 2022-05-17:
I am closing this issue, if you still have this issue, feel free to re-open

closed 2022-05-17 by rahult-github

iamcalledusman, 2022-05-17:
Thank you again!

Wasabinary, 2022-07-05:
Hi @iamcalledusman , did you manage to switch from 1M to coded phy?

I tried tuning a few parameters in extended advertising as mentioned above, but my ESP32-S3 still uses BLE_HCI_LE_PHY_1M...

iamcalledusman, 2022-07-05:
HI @Wasabinary !

Yes i am able to switch to coded phy but not using NimBLE . But by using the example provided by esp-idf . I usedd esp32-c3

Wasabinary, 2022-07-05:
@iamcalledusman So you are using Bluedroid instead of NimBLE? I also tried Bluedroid ble_50 examples, but I'm now trying to adapt NimBLE examples to Coded Phy

iamcalledusman, 2022-07-05:
@Wasabinary!
Yes bluedriod working fine with coded phy! After that stopped working on NimBLE because of requirements

Wasabinary, 2022-07-05:
Ok, thanks a lot for the info! I'll keep trying with NimBLE, hopefully I can make coded phy work with it

iamcalledusman, 2022-07-05:
Best of Luck! if possible share your implementation which may help others!