espressif/esp-idf #8685: Radios crash dive when ble and wifi both connected and streaming data

bill1davis, 2022-03-29:
Development Kit: [N/A] Custom board
Kit version : [N/A]
Module or chip used: [ESP32-WROVER v3]
IDF version : V4.3.1
Build System: [idf.py]
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
Operating System: [UBUNTU LTS]
environment type: [ESP Command Prompt].
Using an IDE?: [No]
Power Supply: [External 5v]

**Description of issue**
Ble is configured as GATT server, connected and streaming data, both via notifications and polled reads.
Wifi is configured as AP, connected to WAN and streaming data in a socket.
Neither stream is near tx limit, ble maybe about 50% of expected bandwith, wifi well below.

Streams on both interfaces stop abruptly when the topmost error appears, in other words this isn't a GATT timeout.

**Expected Behavior**
BLE and Wifi should remain connected, streaming data.

**Log**
```
W (1306494) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x1f
W (1306494) ct5_ble: ESP_GATTS_DISCONNECT_EVT, reason = 31 ini_disc=0
W (1306504) ct5_ble: gatts_profile_event_handler Changing state from BLE_CONNECTED to BLE_ORPHANED
I (1306504) ct5_ble: Set name to VS Orphan 1234567890:20
W (1306514) ct5_ble: start_advertising Changing state from BLE_ORPHANED to BLE_ADVERTISING_ORPHANED
W (1306524) BT_BTM: BTM_BleWriteScanRsp, Partial data write into ADV
I (1306544) ct5_ble: advertising start [A8:03:2A:30:FA:A2]
I (1308194) processSample: Last time 100 Current time AVE 100.000000 max 45800 min 0
bcn_timout,ap_probe_send_startI 
(1310124) ct5_wifi: Unhandled wifi event 21
pm stop, total sleep time: 970358907 us / 1302929888 us
W (1312634) ct5_wifi: WIFI_EVENT_STA_DISCONNECTED expected = 0 reason = 200
E (1312644) ct5_wifi: Restarting Wifi
```

Espressif-liuuuu, 2022-05-23:
@bill1davis Hi, thanks for your report.   
Just to confirme at first, according to your log, the part below seems came from a station instead of AP. Did you configure your ESP32 as STA mode?
`(1310124) ct5_wifi: Unhandled wifi event 21
pm stop, total sleep time: 970358907 us / 1302929888 us`  
`W (1312634) ct5_wifi: WIFI_EVENT_STA_DISCONNECTED expected = 0 reason = 200`

If you configured it with STA mode, it would be appreciated if you could provide us with your reproducible project, which we may need to reproduce your problm, otherwise it actually did happen and the part AP/APSTA with BLE is in progress. We would finish it as soon as possible.

bill1davis, 2022-06-01:
@Espressif-liuuuu I created a simulation of our app with no proprietary code in it with ble and wifi streaming sinks other than our android app and the code ran for days without a single radio issue.  Something is going on on the android side I guess.  Closing.

closed 2022-06-01 by bill1davis