espressif/esp-idf #8688: I met the case of that value of "entry->state" in function, dns_check_entry(u8_t i)  was 58 or 59.

bini2007, 2022-03-30:
## Environment

- Development Kit:    ESP32 module based custom board.
- Kit version (for WroverKit/PicoKit/DevKitC): [v4.2]
- Module or chip used:  [ESP32-WROOM-32UE]
- IDF version (run ``git describe --tags`` to find it): v4.2.2-396-gccef7c1712
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes (vscode and ESP32 extension)
- Power Supply:        external 3.3V

## Problem Description

When we tested our target without no antenna (it seems to be without wifi connection.),
the following happened.

So after we debugged with JTAG debugger, we found the following debugged fact.

The debugged fact is as followings:

I met the case of that value of "entry->state" 
in dns_check_entry(u8_t i) function in the source, dns.c [(](https://github.com/CHERTS/esp32-devkit/blob/master/Espressif/ESP32_RTOS_SDK/third_party/lwip/core/dns.c) was 58 or 59.

So our program went to the default in case statement and met the LWIP_ASSERT() and fianally paniced.

In the end, we didn't know why
but first of all, we added the following codes in the case of the value, 58 and 59 in the function, dns_check_entry(u8_t i).

 `                  case 58:
		    case 59:
		      {
		        LWIP_DEBUGF(DNS_DEBUG, ("dns_check_entry: \"%s\": flush\n", entry->name));
		        /* flush this entry, there cannot be any related pending entries in this state */
		        entry->state = DNS_STATE_UNUSED;
		      }
		      break;`

As you know, we added the same code as the case, DNS_STATE_DONE in the switch statement.
Then the above refered problem didn't happened.

But we still don't know why the case of  the value 58 or 59 happend when without anntenna (wifi connection) 
so anyone, please explain to us about this problem.

jack0c, 2022-05-20:
@bini2007 How can I reproduce the issue. I think the problem looks like a memory corrupted issue.

bini2007, 2022-05-26:
We are going to make our own custom product based on esp32.
Let us know what and how you need to reproduce the issue.

Thanks,

xueyunfei998, 2022-06-06:
hi @bini2007 

How did you reproduce this problem? Can you provide a reproduced demo? I want to reproduce this problem locally.

xueyunfei998, 2022-06-06:
Do you mean that the value of entry->state has two values of 58 and 59?

bini2007, 2022-06-06:
yes, it is right.

bini2007, 2022-06-06:
> hi @bini2007 
> 
> How did you reproduce this problem? Can you provide a reproduced demo? I want to reproduce this problem locally.

Is it OK if we send oue elf file?
then you can reproduce rhe problem.

thanks,

xueyunfei998, 2022-06-07:
hi @bini2007 

`entry->state`This value is scheduled by lwip, and its range of change is `DNS_STATE_NEW = 1 ` `DNS_STATE_ASKING = 2` ,`DNS_STATE_DONE = 3` , `DNS_STATE_UNUSED = 0`  that the application code will not change this value.

If such a problem occurs, I guess your system should have stepped on the memory.
 

Then do the following: ping espressif.com. See if it works properly. This example uses dns address resolution function.
location : `examples/protocols/icmp_echo/main/echo_example_main.c`

xueyunfei998, 2022-06-09:
https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-reference/system/heap_debug.html

You can follow this document, which describes how to debug stepping on memory.

closed 2023-05-23 by Alvin1Zhang