espressif/esp-idf #8690: Wrong nchan when set country code with "TW"

AxelLin, 2022-03-30:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v5.0-dev-2281-ge3c0229d2e
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

With below call:
esp_wifi_set_country_code("TW", true);

I got below message:
I (2064) wifi:set country: cc=TW schan=1 nchan=13 policy=0

### Expected Behavior

Above message seems wrong.
I would expect the nchan=11 if country code is TW.

ElementXs, 2022-03-31:
nsml

0xjakob, 2022-04-01:
@AxelLin I have trouble accessing Taiwan's regulatory information from Shanghai (...), but I was able to access some database from the Linux Kernel, containing the regulatory data: https://mirrors.edge.kernel.org/pub/software/network/wireless-regdb/.

Is it possible that the frequencies changed for Taiwan in recent years?
I'm not sure if I read all this data correctly, but in the most recent database, I see that the frequencies are 
```
...
country TW: DFS-FCC
	# 2.4g band, LP0002 section 3.10.1
	(2400 - 2483.5 @ 40), (30)
...
```
This looks similar to other countries with 13 channels (like Germany, e.g.).

But in an older database from 2011, the numbers are different:
```
...
country TW:
	(2402 - 2472 @ 40), (3, 27)
...
```
In this older database, the numbers are the same as from the US, which only has 11 channels.

Maybe you can check yourself, these are the regulatory sources for that database (I myself can not access these from China):
```
# Source:
# Table of Frequency Allocations of Republic of China (Taiwan) / Feb 2017:
#   https://www.motc.gov.tw/websitedowndoc?file=post/201702221012200.doc& \
#	filedisplay=Table%2Bof%2Bradio%2Bfrequency%2Ballocation.doc
# LP0002 Low-power Radio-frequency Devices Technical Regulations / 23 Aug 2016:
#   http://www.ncc.gov.tw/english/show_file.aspx?table_name=news&file_sn=681
```

Please let us know if you still find that the regulatory information is incorrect, it should be fixed.

kapilkedawat, 2022-04-01:
@AxelLin We follow the info from latest linux kernel.
Please the the channel list for TW.
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt#n1575

AxelLin, 2022-04-01:
Hi @0xjakob @kapilkedawat 

I think you are both right, I'm closing this issue.
Somehow all the 2.4G wifi access points around my environment only support ch1-11 so I thought only ch1-11 is supported in Taiwan.

closed 2022-04-01 by AxelLin