espressif/esp-idf #8694: esp32s3 usb host example cdc_acm_bg96 does not compile

versamodule, 2022-03-30:
I am using the master branch code.
Using VSC, and the command palette for "Show Example Projects" option.  
I let it create the project for the host example cdc_acm_bg96. 

Made no changes other than to select the ESP32-S3 and Uart. 
When I compile it fails with: 

../main/cdc_acm_host_bg96.cpp:18:10: fatal error: nmea.h: No such file or directory
 #include "nmea.h"
          ^~~~~~~~
compilation terminated.
[9/15] Completed 'bootloader'

tore-espressif, 2022-03-31:
Hello @versamodule 

Thank you for reaching out to us.
This project is using [idf-component-manager](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s2/api-guides/tools/idf-component-manager.html) so this `nmea.h` header file should be fetched automatically, if you are using our [VSCode extension](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s2/api-guides/tools/idf-component-manager.html#activating-the-component-manager).

At the moment it is not clear whether this is a bug or you system is misconfigured. Could you please double check that you are using latest version of Espressif's VSCode extension? If yes, could you try rerunning esp-idf install script (`install.sh/.bat`) ?

During configuration step of the build you should see something like this:
```
...
-- Building ESP-IDF components for target esp32
Solving dependencies requirements
Updating lock file at C:/Users/tore/esp/esp-idf/examples/peripherals/usb/host/cdc/cdc_acm_bg96\dependencies.lock
Processing 2 dependencies:
[1/2] idf (5.0.0)
[2/2] igrr/libnmea (0.1.1)
...
```
That is idf-component-manager resolving and fetching defined dependencies

tore-espressif, 2022-03-31:
Fair enough, could you please share full build logs? I'll investigate further

tore-espressif, 2022-03-31:
There are 2 problems in your log:
1. target is set to `esp32` instead of `esp32s3` (run `idf.py set-target esp32s3`)
2. 
```
CMake Warning at J:/Espressif/frameworks/esp-idf-master/esp-idf/tools/cmake/build.cmake:482 (message):
  "idf_component.yml" file was found for components:

  	J:/Espressif/frameworks/esp-idf-master/workspace/cdc_acm_bg96/main

  However, the component manager is not enabled.
  ```
  
  The component manager should be enabled by default. To quick fix this you can set `IDF_COMPONENT_MANAGER` environmental variable to `1`.
  
  I'll forward this issue to VSCode extension team

versamodule, 2022-03-31:
Oh.... I forgot to mention that when the project first opened I saw that the target was not set properly. I though I changed it. I changed it to ESP32-S3 and now I get the header issue. 

Thanks for the info. But isn't it strange that for the other USB examples it compiles ok but this one. 
[Logs.txt](https://github.com/espressif/esp-idf/files/8389916/Logs.txt)
[image]

brianignacio5, 2022-04-01:
> There are 2 problems in your log:
> 
>     1. target is set to `esp32` instead of `esp32s3` (run `idf.py set-target esp32s3`)
> 
> 
> ```
> CMake Warning at J:/Espressif/frameworks/esp-idf-master/esp-idf/tools/cmake/build.cmake:482 (message):
>   "idf_component.yml" file was found for components:
> 
>   	J:/Espressif/frameworks/esp-idf-master/workspace/cdc_acm_bg96/main
> 
>   However, the component manager is not enabled.
> ```
> 
> The component manager should be enabled by default. To quick fix this you can set `IDF_COMPONENT_MANAGER` environmental variable to `1`.
> 
> I'll forward this issue to VSCode extension team

Just need to set the `idf.enableIdfComponentManager` to `true` in  your project's `.vscode/settings.json` to enable the idf component manager which is not enabled  by default.

tore-espressif, 2022-06-02:
@versamodule is your problem solved? can we close this issue?

closed 2022-06-02 by versamodule