espressif/esp-idf #8695: Heap Corruption after a simples MALLOC to SPIRAM and try to clean it.

filzek, 2022-03-30:
ESP-IDF v4.4
xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
Esp32 v3.3Wrover 8MB /8MB

I (31) boot: ESP-IDF v4.2-dev-792-g6330b3345-dirty 2nd stage bootloader
I (31) boot: compile time 01:02:27
I (31) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 8MB
I (50) boot: Enabling RNG early entropy source...

We found a terrible problem on the current ESP-IDF v4.4

Steps to reproduce:

in a global part do:
typedef struct queue_command_timmer_t;
{
	char sCommand[5];
	char sPort[5];
	char sData[1025];
	uint32_t 		epoch_now;
	uint32_t 		epoch_exe;
	uint32_t 		serial;
	bool 				save_to_rom;												
	} queue_command_timmer_t;

char *debug_string;
queue_command_timmer_t *timmer_commands = NULL;

in the initial task do:
debug_string = (char *)malloc(1000000);
bzero(debug_string,1000000);
strcat(debug_string,"INITIAL DEBUG OK - MALLOC SUCESS!\n");
heap_caps_check_integrity(MALLOC_CAP_INVALID, true);

Heap is okay. 

timmer_commands = heap_caps_malloc(50*sizeof(queue_command_timmer_t),MALLOC_CAP_SPIRAM);
heap_caps_check_integrity(MALLOC_CAP_INVALID, true);

Heap is okay. 

If we try to clean the structure above than the CRASH happen in a check:

	for (int z=0;z<50;z++) {
			bzero(timmer_commands[z-1].sCommand,5);
			bzero(timmer_commands[z-1].sPort,5);
			bzero(timmer_commands[z-1].sData,1025);

			//OR 
                        //memset(timmer_commands[z-1].sCommand,0,5);
			//memset(timmer_commands[z-1].sPort,0,5);
			//memset(timmer_commands[z-1].sData,0,1025);

			timmer_commands[z-1].epoch_now = 0;
			timmer_commands[z-1].epoch_exe = 0;
			timmer_commands[z-1].serial = 0;
			timmer_commands[z-1].save_to_rom = 0;
	}

heap_caps_check_integrity(MALLOC_CAP_INVALID, true);

CRASH HAPPEN HERE!

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4009c147  PS      : 0x00060c33  A0      : 0x8009c189  A1      : 0x3ffbca30
0x4009c147: block_size at D:/Dropbox/esp-idf/components/heap/heap_tlsf_block_functions.h:75
 (inlined by) block_is_last at D:/Dropbox/esp-idf/components/heap/heap_tlsf_block_functions.h:86
 (inlined by) tlsf_walk_pool at D:/Dropbox/esp-idf/components/heap/heap_tlsf.c:581

A2      : 0x94e59afc  A3      : 0x4009bfb8  A4      : 0x3ffbca50  A5      : 0x3f9045ac
0x4009bfb8: integrity_walker at D:/Dropbox/esp-idf/components/heap/heap_tlsf.c:499

A6      : 0xfffffffc  A7      : 0x3f802e60  A8      : 0x8009c161  A9      : 0x3ffbca10
A10     : 0x3f9045a4  A11     : 0x55555554  A12     : 0x00000000  A13     : 0x3ffbca50
A14     : 0x007b6d18  A15     : 0x003fffff  SAR     : 0x00000001  EXCCAUSE: 0x0000001c
EXCVADDR: 0x94e59b00  LBEG    : 0x4009360e  LEND    : 0x40093619  LCOUNT  : 0x00000000
0x4009360e: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150

0x40093619: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace:0x4009c144:0x3ffbca300x4009c186:0x3ffbca50 0x4009ccda:0x3ffbca80 0x400d654a:0x3ffbcaa0 0x400f231c:0x3ffbcac0 0x401d19c7:0x3ffbccc0
0x4009c144: block_size at D:/Dropbox/esp-idf/components/heap/heap_tlsf_block_functions.h:75 (discriminator 4)
 (inlined by) block_is_last at D:/Dropbox/esp-idf/components/heap/heap_tlsf_block_functions.h:86 (discriminator 4)
 (inlined by) tlsf_walk_pool at D:/Dropbox/esp-idf/components/heap/heap_tlsf.c:581 (discriminator 4)

0x4009c186: tlsf_check_pool at D:/Dropbox/esp-idf/components/heap/heap_tlsf.c:607

0x4009ccda: multi_heap_check at D:/Dropbox/esp-idf/components/heap/multi_heap.c:299

0x400d654a: heap_caps_check_integrity at D:/Dropbox/esp-idf/components/heap/heap_caps.c:498

0x400f231c: app_main at d:\user_main.c:30

0x401d19c7: main_task at D:/Dropbox/esp-idf/components/freertos/port/port_common.c:129

BUT  besides if we use CALLOC it runs perfect!
timmer_commands = calloc(50, sizeof(queue_command_timmer_t));

and then 
	for (int z=0;z<50;z++) {
			bzero(timmer_commands[z-1].sCommand,5);
			bzero(timmer_commands[z-1].sPort,5);
			bzero(timmer_commands[z-1].sData,1025);
			timmer_commands[z-1].epoch_now = 0;
			timmer_commands[z-1].epoch_exe = 0;
			timmer_commands[z-1].serial = 0;
			timmer_commands[z-1].save_to_rom = 0;
	}

heap_caps_check_integrity(MALLOC_CAP_INVALID, true);

IT PASS and works okay.

But, why this is happening now? There some fault on the heap/spiram on somehow.

filzek, 2022-03-30:
after a more closer debug we found out that the problem is with a BOOL type in a strutct, something with a problem with the GCC and Linker.

Please correct it ASAP! problem is real and serious.

o-marshmallow, 2022-03-31:
Your loop starts at `z = 0` and you are using `timmer_commands[z-1]`, for sure this will have a bad side effect as this will erase your block's metadata.
Please have a try again with `for (int z=1;z<=50;z++)`

chegewara, 2022-03-31:
I had similar behavior with bool, when ive been using C++ class with uninitialized class member variable.
Adding initialization to that variable solved the problem. It may be some weird compiler optimization.

Alvin1Zhang, 2022-04-29:
Thanks for sharing updates.