espressif/esp-idf #8696: RTC Registers from source code are different from reference

costa-victor, 2022-03-30:
## Environment

- Development Kit:    NodeMCU ESP32 DevKit V1
- Module:  ESP32-WROOM-32
- IDF version: master - Hash [e3c0229d2ec9bfd0203a3cab18d6642798457578]
- Build System:   idf.py
- Operating System:     macOS
- Using an IDE?:  No
- Power Supply:         USB

## Problem Description

I'm looking at the ULP Coprocessor programming for ESP32, then I notice that the registers mapped into the [ESP32 Technical reference](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) is different from the [source code](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/rtc_io_reg.h#L1290). The image below shows the bits I'm talking about.

[image]

Commented below the part of the source code where it [appears](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/rtc_io_reg.h#L1290).
```c
#define RTC_IO_TOUCH_PAD2_REG          (DR_REG_RTCIO_BASE + 0x9c)

/* ================ These mapping below isn't mapped into the reference   ================ */
/* RTC_IO_TOUCH_PAD2_HOLD : R/W ;bitpos:[31] ;default: 1'd0 ; */
/*description: hold the current value of the output when setting the hold to Ò1Ó*/
#define RTC_IO_TOUCH_PAD2_HOLD  (BIT(31))
#define RTC_IO_TOUCH_PAD2_HOLD_M  (BIT(31))
#define RTC_IO_TOUCH_PAD2_HOLD_V  0x1
#define RTC_IO_TOUCH_PAD2_HOLD_S  31
/* RTC_IO_TOUCH_PAD2_DRV : R/W ;bitpos:[30:29] ;default: 2'd2 ; */
/*description: the driver strength of the pad*/
#define RTC_IO_TOUCH_PAD2_DRV  0x00000003
#define RTC_IO_TOUCH_PAD2_DRV_M  ((RTC_IO_TOUCH_PAD2_DRV_V)<<(RTC_IO_TOUCH_PAD2_DRV_S))
#define RTC_IO_TOUCH_PAD2_DRV_V  0x3
#define RTC_IO_TOUCH_PAD2_DRV_S  29
/* RTC_IO_TOUCH_PAD2_RDE : R/W ;bitpos:[28] ;default: 1'd1 ; */
/*description: the pull down enable of the pad*/
#define RTC_IO_TOUCH_PAD2_RDE  (BIT(28))
#define RTC_IO_TOUCH_PAD2_RDE_M  (BIT(28))
#define RTC_IO_TOUCH_PAD2_RDE_V  0x1
#define RTC_IO_TOUCH_PAD2_RDE_S  28
/* RTC_IO_TOUCH_PAD2_RUE : R/W ;bitpos:[27] ;default: 1'd0 ; */
/*description: the pull up enable of the pad*/
#define RTC_IO_TOUCH_PAD2_RUE  (BIT(27))
#define RTC_IO_TOUCH_PAD2_RUE_M  (BIT(27))
#define RTC_IO_TOUCH_PAD2_RUE_V  0x1
#define RTC_IO_TOUCH_PAD2_RUE_S  27

/* ================ Bit 23-25 is mapped into the reference ================*/
/* RTC_IO_TOUCH_PAD2_DAC : R/W ;bitpos:[25:23] ;default: 3'h4 ; */
/*description: touch sensor slope control. 3-bit for each touch panel  default 100.*/
#define RTC_IO_TOUCH_PAD2_DAC  0x00000007
#define RTC_IO_TOUCH_PAD2_DAC_M  ((RTC_IO_TOUCH_PAD2_DAC_V)<<(RTC_IO_TOUCH_PAD2_DAC_S))
#define RTC_IO_TOUCH_PAD2_DAC_V  0x7
#define RTC_IO_TOUCH_PAD2_DAC_S  23
```

## Expected behavior
Expecting the same information from the technical reference into the source code.

## Actual behavior
Registers in the source code are different from the documentation.

## Conclusion
Am I doing something wrong? Where can I find informations about these bits 25-31? I get this reference from [official](https://www.espressif.com/en/support/documents/technical-documents?keys=&field_type_tid%5B%5D=492) page, as showed below:
[image]

ESP-Marius, 2022-04-01:
@costa-victor 

You are right, there is a mismatch here. I'll check with our doc team to see if these bits are something that should be added to the TRM.

costa-victor, 2022-04-01:
Thanks @ESP-Marius for the quick response.

Okay, I'll wait for this.

KaeLL, 2022-04-01:
Code > Docs
☹️

ESP-Marius, 2022-04-02:
Turns out in this case the docs are actually correct. These bits (RTC_IO_TOUCH_PAD_RUE/RDE/DRV) do not work in ESP32. If a pin is configured as an RTC GPIO pin, its drive strength, pull-up resistor and pull-down resistor can only be controlled by the bits listed in column "Normal Execution OR SLP_SEL = 0", instead of "RUE/RDE/DRV" bits.

We'll clean up the code to reflect this.

AxelLin, 2022-12-17:
@ESP-Marius how is the status? (It takes so much time to clean up the code?)

ESP-Marius, 2022-12-18:
@AxelLin The time it takes is not necessarily related to the size of the issue. It goes into our backlog until someone responsible for that area picks it up, which time will depend on the issue's priority.