espressif/esp-idf #8698: esp_netif_set_ip_info always returning ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED

chipweinberger, 2022-03-31:
Reopening https://github.com/espressif/esp-idf/issues/4893

**Look at that issue first^^**

I am hitting this issue using v4.4, with ESP32-S3. Trying to create an AP with a specific IP using APSTA mode.

My DHCP status is "Init", then I call "stop", but then the status is "Started", the complete opposite of stop.

I must be getting the chain of events wrong. Can someone share/describe a working example?

```
I (383) pd_wifi_start()
I (383) pd wifi: esp_netif_init()
I (393) pd wifi: esp_event_loop_create_default()
I (393) pd wifi: esp_netif_create_default_wifi_ap()
I (403) pd wifi: esp_wifi_init()
I (503) pd wifi: esp_event_handler_instance_register(ESP_EVENT_ANY_ID, ...)
I (513) pd wifi: esp_event_handler_instance_register(IP_EVENT_STA_GOT_IP, ...)
I (523) pd wifi: esp_wifi_set_mode(WIFI_MODE_APSTA)
I (533) pd wifi: esp_wifi_set_config(...)
I (1383) pd wifi: esp_wifi_start()
I (1473) pd wifi:  Wifi Event: STA Mode - Start
I (1483) pd wifi: Wifi Event: AP Mode - Start
I (1483) pd wifi: dhcp status: Init
I (1483) pd wifi: esp_netif_dhcpc_stop(...)
I (1483) pd wifi: Sleeping 5 seconds...
I (6493) pd wifi: dhcp status: Started
I (6493) pd wifi: esp_netif_set_ip_info(...)
E (6493) : e1 : [pd wifi] esp_netif_set_ip_info() failed, ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED
```

But If I try to call `esp_netif_dhcpc_stop`() twice, the second time it says it is already stopped. So `esp_netif_dhcpc_stop` and `esp_netif_set_ip_info` do not seem to agree if DHCP is stopped or not. 

```
...
I (1473) pd wifi: Wifi Event: STA Mode - Start
I (1483) pd wifi: dhcp status: Init
I (1483) pd wifi: Wifi Event: AP Mode - Start
I (1493) pd wifi: esp_netif_dhcpc_stop(0x3fce3308)
I (1493) pd wifi: dhcp status: Started
I (1503) pd wifi: Sleeping 5 seconds...
I (6503) pd wifi: dhcp status: Started
I (6503) pd wifi: esp_netif_dhcpc_stop(0x3fce3308)
E (6503) : e1 : [pd wifi] esp_netif_dhcpc_stop() failed, ESP_ERR_ESP_NETIF_DHCP_ALREADY_STOPPED
E (6503) : e1 : [pd wifi] esp_netif_set_ip_info() failed, ESP_ERR_ESP_NETIF_DHCP_NOT_STOPPED
```

<details>
  <summary>Lots of code...</summary>

```
// create AP
wifi_config_t create_hotspot_config(pd_error_t* error) {

    PDArgAssert(error, TAG);
    pd_err_unset(error);

    wifi_config_t config;
    memset(&config, 0, sizeof(wifi_config_t));
    config.ap.channel = pd_wifi_default_ap_channel();
    config.ap.max_connection = 5;
    config.ap.authmode = WIFI_AUTH_WPA_WPA2_PSK;

    ESP_LOGI(TAG, "switching to 'Create Wifi AP Hotspot' mode");

    wifi_ap_config_t* ap = &config.ap;

    if (strlen(ssid) == 0){
        pd_err_fail(error, TAG, "ssid is NULL");
        return config;
[... 241 lines cut ...]
            return;
        }

        // log dhcp status
        ESP_LOGI(TAG, "dhcp status: %s", netif_dhcp_status_to_string(status));

        esp_netif_ip_info_t ip_info;

        IP4_ADDR(&ip_info.ip, 192, 168, 1, 1);
        IP4_ADDR(&ip_info.gw, 192, 168, 1, 1);
        IP4_ADDR(&ip_info.netmask, 255, 255, 255, 0);

        ESP_LOGI(TAG, "esp_netif_set_ip_info()");
        err = esp_netif_set_ip_info(init.netif_handle, &ip_info);
        if (err != ESP_OK) {
            pd_err_fail(error, TAG, "esp_netif_set_ip_info() failed, %s", esp_err_to_name(err));
            return;
        }
    }

    pd_err_success(error);
    return;
}
```
</details>

sharkx, 2022-03-31:
For what it's worth... I'm still using the changes mentioned in [this comment](https://github.com/espressif/esp-idf/issues/4893#issuecomment-633580003), and they work as expected.

david-cermak, 2022-04-01:
Hi @chipweinberger 

I think the main trouble is in your reconfiguration code and the fact that you're trying to use one network interface handle from both STA and AP. It is possible, yes, but a proper cleanup is necessary. If you're trying to use AP+STA, though, you'd have to use two network interfaces at the same time (see the [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html#wifi-default-initialization))

I've tried to run the code you've shared but failed with too many undefined references. But, I'm afraid I didn't understand what you're trying to achieve from the code. Was it the static IP on the STA interface, but still running DHCP server on AP interface?

chipweinberger, 2022-04-01:
Hmm, it’s not clear from my logs but I’m only calling esp_set_config(ESP_IF_AP,..). I don’t use the STA portion. I only use APSTA to allow esp_wifi_scan()

I want static IP on the AP interface, while allowing for a wifi scan to be done at some point (maybe do the scan first?).

david-cermak, 2022-04-04:
@chipweinberger You have to stop the server, not the client! (`dhcpc` == DHCP client, `dhcps` == DHCP server)

```patch
-err = esp_netif_dhcpc_stop(ap);
+err = esp_netif_dhcps_stop(ap);
```

just tested with:
```cpp
    esp_err_t err = esp_netif_dhcps_stop(ap);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_netif_dhcpc_stop() failed, %s", esp_err_to_name(err));
        return;
    }
    log_dhcp_status(ap);
    err = esp_netif_dhcps_stop(ap);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_netif_dhcpc_stop() failed, %s", esp_err_to_name(err));
        return;
    }
    log_dhcp_status(ap);
    ESP_LOGI(TAG, "wifi_init_sta finished.");
```

and works as expected:
```
I (804) wifi:state: init -> auth (b0)
I (804) wifi station: dhcp status: 2
E (804) wifi station: esp_netif_dhcpc_stop() failed, ESP_ERR_ESP_NETIF_DHCP_ALREADY_STOPPED
I (814) wifi:state: auth -> assoc (0)
I (824) wifi:state: assoc -> run (10)
```

chipweinberger, 2022-04-04:
Wow! You are right! I did not even know there was a client server architecture. 

Thank you for noticing this. 

I based my code off of [ETHClass::config](https://github.com/espressif/arduino-esp32/search?q=ETHClass%3A%3Aconfig) from the [adruino-esp32](https://github.com/espressif/arduino-esp32) library.

Hopefully others that made the same mistake will find this thread =)

chipweinberger, 2022-04-05:
Tested. Using `esp_netif_dhcps_stop()` && `esp_netif_dhcps_start()` fixed it!

closed 2022-04-05 by chipweinberger