espressif/esp-idf #8703: ESP32-S3 BLE cannot be scanned by IOS provisioning app

KonssnoK, 2022-04-01:
If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-S3 Devkit C + ESP32-C3devKitC02
- IDF version v4.4 c29343eb94d2f2ca17b3a5b38c82452e556147f2

## Problem Description

ios provisioning app is not able to scan S3 but successfully scans C3.
Using ios 15.4 on Iphone SE + Galaxy S21

I picked C3 as comparisong because they  both are BLE 5.0 chipsets.

### Steps to reproduce

simply compile from 
`esp-idf\examples\provisioning\wifi_prov_mgr`
for both C3 and S3, then flash devices.

C3 will popup in the BLE app for ios, S3 will not.
Both are visible in android BLE app.

KonssnoK, 2022-04-04:
Generated an app-related issue on the proper repo since it's unknown if issue is coming from chipset or app.

Weijian-Espressif, 2022-04-05:
@KonssnoK  Do you mean that the IOS app can't scan the adv packets of S3?

KonssnoK, 2022-04-05:
@Weijian-Espressif  exactly.
Sadly i don't own a swift dev environment so i cannot test what is going on the provisioning app side.

But trying to check the ScanResult of the android app i couldn't find any difference between C3 and S3.

Weijian-Espressif, 2022-04-05:
Please test gatt_server example with your phone SE ,  check whether S3 BLE can be scaned and connect

KonssnoK, 2022-04-05:
- I managed to see it (that's new) once, 
- but after i tried to connect with the provisioning app it gave a "pop" error.
- Then I reset the board, 
- tried to scan with the Bluetooth of IOS -> no success
- Android still visible tho.
- Reflash with same file
- Still not visible on IOS.

If i try with android it just tells me "an app is needed to pair with this device"

KonssnoK, 2022-04-05:
More info:
- i restarted the iphone
- device visible in the IOS provisioning app
- device NOT visible in the IOS bluetooth section
- log when failing to connect to the ios provisioning
`I (628) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44

I (638) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44

I (96208) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 7e:dc:3d:5f:56:8e:
I (96208) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote 7e:dc:3d:5f:56:8e:
I (96238) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 185
I (96238) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 185
I (96458) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 30,latency = 0, timeout = 400
I (96868) GATTS_DEMO: GATT_READ_EVT, conn_id 0, trans_id 2, handle 43

I (116068) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
`
- Log when connecting with android
`
I (234568) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 5b:1f:5d:f4:18:06:
I (234568) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote 5b:1f:5d:f4:18:06:
I (235748) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 6,latency = 0, timeout = 500
I (236798) GATTS_DEMO: GATT_READ_EVT, conn_id 0, trans_id 1, handle 47

I (237008) GATTS_DEMO: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 32,latency = 0, timeout = 400
`
- Via the provisioning app now i'm able too see it if i reset the board

KonssnoK, 2022-04-05:
More info:
- I put back the provisioning FW on the S3.
- I removed the prefix from the provisioning app
- Android sees PROV_*
- IOS sees ESP_GATTS_DEMO

This is same issue as 
#7647

KonssnoK, 2022-04-06:
@Weijian-Espressif do you have any idea on this? 
i tried to dig a bit in the BLE stack, but it's not that easy.
i reached `btu_hcif_send_cmd` which is called at BLE setup to set the Scan response raw. 

However these are the initial settings, I'm still trying to find the function that is called when the actual scan response is sent.

Weijian-Espressif, 2022-04-06:
@KonssnoK  You can download the NRF Connect app, Test connect and pairing. Check if provisioning App is OK

Weijian-Espressif, 2022-04-06:
Or you can test the `gatt_security_server` example on your iPhone

KonssnoK, 2022-04-06:
@Weijian-Espressif 

Using the `gatt_security_server ` app I'm always able to scan and connect to the device via IOS nRF Connect, even by changing the device name and reflashing.

If i move to  `gatt_server `  the device is not visible anymore for some minutes, then appears with the new name.
Also in this case i can change the device name, reflash, and see it with the correct name in the scan.

Finally i move to  `wifi_prov_mgr`. In the iOS nRF Connect the device is still seen with the old name, in Android it's seen with the correct name.

So,
this is an issue of the BLE implementation of the provisioning system, which affects only iOS.

What can we do as next steps?

Weijian-Espressif, 2022-04-06:
his is not a bug, iPhone has a cache, it shows the old name, you connect and then disconnect, iPhone will update the cache, it will show your new name, you can try it

KonssnoK, 2022-04-06:
> this is not a bug, iPhone has a cache, it shows the old name, you connect and then disconnect, iPhone will update the cache, it will show your new name, you can try it

if this is true then why with `gatt_security_server ` and  `gatt_server` the name gets updated instantly (without connecting) and with  `wifi_prov_mgr` is not?

All this started because my FrontEnd colleague didn't manage to see a device that he never seen before with its own Iphone, therefore he couldn't have any cache of the BLE device.

Weijian-Espressif, 2022-04-06:
You can do a test with your iPhone, send adv, see the adv name, connect and disconnect, change the adv name and restart the adv, and then check the  adv name with your iphone

Weijian-Espressif, 2022-04-06:
please refer https://developer.apple.com/forums/thread/19381

KonssnoK, 2022-04-06:
@Weijian-Espressif , as expected, the issue is related to the protocomm component.

By comparing code of `simple_ble.c` and `example_ble_sec_gatts_demo.c` i was able to find which is the discrepancy which is causing this issue.

In `example_ble_sec_gatts_demo.c` the code relies on `esp_ble_gap_config_adv_data ` for configuring the scan response.
Instead, in `protocomm_ble.c` your colleagues used raw_adv_data for both advertising and scan response.
The raw data is then set in `simple_ble.c` using `esp_ble_gap_config_scan_rsp_data_raw`.

Apparently, something in the raw data is missing -> bug.

Please check out these changes:
https://github.com/KonssnoK/esp-idf/commit/a9e06bc7d88b42c07eab16b54d5ec08e1d3ff364

In this code, instead of using a raw data, function `esp_ble_gap_config_adv_data` is used. In this way the name of the device gets updated immediately also with provisioning app.

Please forward to your colleagues for fixing.

Thanks!

Alvin1Zhang, 2022-05-16:
Thanks for reporting, and sorry for slow turnaround, fix on master branch is available at https://github.com/espressif/esp-idf/commit/f7f6a929f2608cbcb67eafa240369ec4987971d1, we are back porting the fix to release/4.4 and release/4.3. Thanks.

Alvin1Zhang, 2022-07-11:
Thanks for reporting, and sorry for slow turnaround. Fix on release/4.4 branch is available at https://github.com/espressif/esp-idf/commit/74dd39d8b07fc64abc69a4094af5745d4d846096 and fix on release/4.3 branch is available at https://github.com/espressif/esp-idf/commit/fab70ef3262ce70b5cc79f024ca966abe6fb9fcd, feel free to reopen.

closed 2022-07-11 by Alvin1Zhang