espressif/esp-idf #8705: Webserver does not work when stopped/started in the middle of the program

krupis, 2022-04-01:
I have configured my device in the AP_STA mode and I start a webserver. The functions I use:
```
    WIFI_`START_AP_STA();
    ESP_ERROR_CHECK(start_file_server("/spiffs"));
```

Function to initialise WIFI
```
void WIFI_INIT_AP_STA(void)
{
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    ap_netif_g = esp_netif_create_default_wifi_ap();
    assert(ap_netif_g);

    sta_netif_g = esp_netif_create_default_wifi_sta();
    assert(sta_netif_g);

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    wifi_config_t wifi_config = {
        .ap = {
            //.ssid = EXAMPLE_ESP_WIFI_SSID_AP,
            //.ssid_len = strlen(EXAMPLE_ESP_WIFI_SSID_AP),
            .ssid_len = strlen(device_info.elstat_id),
            .channel = EXAMPLE_ESP_WIFI_CHANNEL,
            .password = EXAMPLE_ESP_WIFI_PASS_AP,
            .max_connection = EXAMPLE_MAX_STA_CONN,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        },
    };
    strcpy((char*)(wifi_config.ap.ssid),device_info.elstat_id);
    if (strlen(EXAMPLE_ESP_WIFI_PASS_AP) == 0) {
        wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    }
    

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI("WIFI AP", "wifi_init_softap finished. SSID:%s password:%s channel:%d",
             EXAMPLE_ESP_WIFI_SSID_AP, EXAMPLE_ESP_WIFI_PASS_AP, EXAMPLE_ESP_WIFI_CHANNEL);
}
```

Function to start the webserver
```
esp_err_t start_file_server(const char *base_path)
{
    struct file_server_data *server_data = NULL;
    /* Validate file storage base path */
    if (!base_path || strcmp(base_path, "/spiffs") != 0) {
        ESP_LOGE("START_SERVER", "File server presently supports only '/spiffs' as base path");
        return ESP_ERR_INVALID_ARG;
    }

    if (server_data) {
        ESP_LOGE("START_SERVER", "File server already started");
        return ESP_ERR_INVALID_STATE;
    }

    /* Allocate memory for server data */
    server_data = calloc(1, sizeof(struct file_server_data));
    if (!server_data) {
        ESP_LOGE("START_SERVER", "Failed to allocate memory for server data");
        return ESP_ERR_NO_MEM;
    }
    strlcpy(server_data->base_path, base_path,
            sizeof(server_data->base_path));

    //httpd_handle_t server = NULL; //use global instead

    httpd_config_t config = HTTPD_DEFAULT_CONFIG();
    config.max_uri_handlers = 15;
    config.stack_size = 8096;
    /* Use the URI wildcard matching function in order to
     * allow the same handler to respond to multiple different
     * target URIs which match the wildcard scheme */
    config.uri_match_fn = httpd_uri_match_wildcard;

    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, &server_global));

    ESP_LOGI("START_SERVER", "Starting HTTP Server on port: '%d'", config.server_port);
    if (httpd_start(&server_global, &config) != ESP_OK) {
        ESP_LOGE("START_SERVER", "Failed to start file server!");
        return ESP_FAIL;
    }

    //URI handler for getting uploaded files 
    httpd_uri_t file_download = {
        .uri       = "/*",  // Match all URIs of type /path/to/file
        .method    = HTTP_GET,
        .handler   = index_page_handler,
       .user_ctx  = server_data    // Pass server data as context
    };
    httpd_register_uri_handler(server_global, &file_download);
return ESP_OK;
}
```

My WIFI and Webserver is working fine. I would like to achieve something a little bit more advanced:
I would like to be able to completely stop and deinitilise wifi in the middle of a program and then enable it once again when I need to.

When I want to stop the WIFI, I call the following functions:
```
        stop_webserver(server_global);
        esp_wifi_disconnect();            // break connection to AP
        esp_wifi_stop();                 // shut down the wifi radio
        esp_wifi_deinit();              // release wifi resources   
        esp_netif_destroy_default_wifi(ap_netif_g);
        esp_netif_destroy_default_wifi(sta_netif_g);
        esp_event_loop_delete_default();
        esp_netif_deinit();
```

When I want to enable the WIFI and the webserver again, I simply call:

```
WIFI_START_AP_STA();
ESP_ERROR_CHECK(start_file_server("/spiffs"));
```

I Can see the AP being created and I can connect to it, but the webserver does not seem to be working.

Console log when I connect to the AP initially:
```
I (8443) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (8443) wifi:station: cc:6b:1e:9b:36:43 join, AID=1, bgn, 40U
[1B][0;32mI (8453) webserver: station cc:6b:1e:9b:36:43 join, AID=1[1B][0m
[1B][0;32mI (8453) webserver: connected clients = 1 [1B][0m
[1B][0;32mI (8463) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2[1B][0m
```

Console log when I disable and enable the WIFI and webserver and then try to connect to the AP:
```
I (51843) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (51843) wifi:station: cc:6b:1e:9b:36:43 join, AID=1, bgn, 40U
[1B][0;32mI (51853) webserver: station cc:6b:1e:9b:36:43 join, AID=1[1B][0m
[1B][0;32mI (51853) webserver: connected clients = 1 [1B][0m
```

Notice a difference.  After disabling and enabling the device, the ESP32 does not assign IP address to my computer when I connect to the AP. I cannot see the log:
```
[1B][0;32mI (8463) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2[1B][0m
```

Could someone help me understand what could be the issue here?

nishanth-radja, 2022-04-04:
@krupis can you pls mention the IDF version you are using?

krupis, 2022-04-04:
Hey. I am using v4.4

I have been further thinking about it.. Could it be because the DHCP lease time? If that is the case, I dont think there is any way around this? I hope I am wrong about this

krupis, 2022-04-04:
I have been testing it in both AP_STA and only AP mode. It seems to be working in AP mode but when I use AP_STA mode, It does not.

krupis, 2022-04-04:
I have created a little repository where this issue can be easily recreated:
https://github.com/krupis/wifi_test/blob/master/main/blink_example_main.c

You can enable/disable WiFi using 2 UART commands:
"stop_wifi_ap"
and
"start_wifi_ap"

nishanth-radja, 2022-05-09:
@krupis The issue seems to be fixed on latest 4.4, can you pull the latest 4.4 IDF and try. "4222e93351bd387b87676b6a43ccf44397958fb5"

Note:
The issue was fixed on this commit "ef7ce17dbd4a0585d4c10549d1788f40f9aa02d5"

closed 2022-08-22 by nachiketkukade