espressif/esp-idf #8706: Backtrace of `abort`ed/`exit`ed tasks don't show on core dumps

KaeLL, 2022-04-01:
## Environment
- Module or chip used:  ESP32
- IDF version: master - 97fb98a91b308d4f4db54d6dd1644117607e9692
- Compiler version: esp-2021r2-patch3 8.4.0 + [new esp-gdb](https://github.com/espressif/binutils-gdb/releases/tag/esp-gdb-v11.1_20220318) (behavior is the same on python2 based esp-gdb as well)

### Expected Behavior
`abort`ed/`exit`ed tasks should have their stack shown correctly like any other task and like it's shown by idf monitor backtrace decoding.

### Steps to reproduce

1. Run code to repro the issue
2. Decode coredump
3. Compare decoded backtrace (which works) to `main` task stack on coredump (which doesn't)

### Code to reproduce this issue

```c
#include <stdlib.h>

void app_main(void)
{
    exit(EXIT_FAILURE);
}
```

## Debug Logs

<details>
<summary>Log + decoded core dump</summary>

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:2124
ho 0 tail 12 room 4
load:0x40078000,len:15484
load:0x40080400,len:3760
0x40080400: _init at ??:?

entry 0x4008060c
I (33) cpu_start: Pro cpu up.
I (33) cpu_start: Starting app cpu, entry point is 0x40081020
0x40081020: call_start_cpu1 at /esp-idf/components/esp_system/port/cpu_start.c:152

I (0) cpu_start: App cpu up.
I (47) cpu_start: Pro cpu start user code
I (47) cpu_start: cpu freq: 160000000 Hz
I (47) cpu_start: Application information:
I (52) cpu_start: Project name:     test_minimal
I (57) cpu_start: App version:      1
I (62) cpu_start: Compile time:     Apr  1 2022 17:00:57
I (68) cpu_start: ELF file SHA256:  860110c87e986694...
I (73) cpu_start: ESP-IDF:          97fb98a-dirty
I (79) heap_init: Initializing. RAM available for dynamic allocation:
I (86) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (92) heap_init: At 3FFB4D20 len 0002B2E0 (172 KiB): DRAM
I (98) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (104) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (111) heap_init: At 4008B714 len 000148EC (82 KiB): IRAM
I (118) spi_flash: detected chip: generic
I (122) spi_flash: flash io: qio
I (126) esp_core_dump_uart: Init core dump to UART
I (131) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

abort() was called at PC 0x400d4b3b on core 0
0x400d4b3b: syscall_not_implemented_aborts at /esp-idf/components/newlib/syscalls.c:35

[... 149 lines cut ...]
.dram0.heap_start 0x3ffb4d20 0x0 RW  
.coredump.tasks.data 0x3ffb74e0 0x158 RW 
.coredump.tasks.data 0x3ffb7260 0x278 RW 
.coredump.tasks.data 0x3ffb763c 0x158 RW 
.coredump.tasks.data 0x3ffb7c70 0x124 RW 
.coredump.tasks.data 0x3ffb7d9c 0x158 RW 
.coredump.tasks.data 0x3ffb83d0 0x124 RW 
.coredump.tasks.data 0x3ffaf818 0x158 RW 
.coredump.tasks.data 0x3ffaf650 0x1c0 RW 
.coredump.tasks.data 0x3ffb6380 0x158 RW 
.coredump.tasks.data 0x3ffb61c0 0x1b8 RW 
.coredump.tasks.data 0x3ffafd78 0x158 RW 
.coredump.tasks.data 0x3ffafbd0 0x1a0 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
Coredump checksum='7142bb09'
I (1280) esp_core_dump_uart: Core dump has been written to uart.
CPU halted.

Done
```

</details>

igrr, 2022-04-01:
I think this is currently tracked in https://github.com/espressif/esp-idf/issues/6124. We have tried fixing this in esp-2021r2 release, but that caused a regression, so in esp-2021r2-patch3 the "fix" was reverted. We plan to investigate how to fix the issue again, please follow the updates in the linked ticket.
(For the technical explanation of the issue, please see commits linked from that ticket.)

KaeLL, 2022-04-01:
Great. Feel free to close it.

Alvin1Zhang, 2022-04-02:
Thanks for reporting, let us track in https://github.com/espressif/esp-idf/issues/6124.

closed 2022-04-02 by Alvin1Zhang