espressif/esp-idf #8711: Esp32 WiFi/Ble True IP Nat Mesh Network

filzek, 2022-04-03:
Hi,

Why not make the true mesh api to work as IP Nat for a tcp/ip network?

It can use the voting system of the current mesh to create the links among each to another and use the lwip nat as it will perfect make it worth as the Espressif Mesh network will sustain the best path and can work as a Spanning Tree Protocol, so mesh supply the best route to the Nat network and it will flow thru.

Why this is not considerable for the true and transparent mesh network?

Esp32 carry external ram so it can handle the packets and retransmission over any scenario.

Isn't this the best and more affordable way to do it so?

david-cermak, 2022-04-06:
Hi @filzek 

Here's the attempt to implement the IP mesh using NAT:

https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network

This example, however, doesn't seem to attract too many customers, so it stayed in the example folder and didn't make it to an official solution. Is this something, that could be used as a starting point for your usecase?

Also, we can use this issue for other users to comment and/or up/down vote this feature.

filzek, 2022-05-18:
> Hi @filzek
>
> Here's the attempt to implement the IP mesh using NAT:
>
> https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network
>
> This example, however, doesn't seem to attract [...]

Did not work as true ip mesh so doesn't work as intent

filzek, 2022-05-18:
The mesh nodes should.be able to post/rest to an utl in the internet directly.  The example doesn't do that.

david-cermak, 2022-05-18:
> Did not work as true ip mesh so doesn't work as intent

> The mesh nodes should.be able to post/rest to an utl in the internet directly. 

Could you please elaborate? The nodes should be able access internet without an issue, the example doesn't send any post/rest request, but simply connects to a public mqtt broker.

sramrajkar, 2022-09-02:
Hi,
I have this example working when the root connects to a WiFi AP router. Can the same feature be made possible where the root node has a PPPoS serial connection and provides that as a source of internet to the MESH devices?

david-cermak, 2024-10-01:
Closing this older issue as "won't do", since we already have an example which demonstrates this functionality:
https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network

@filzek FYI: please reopen if you feel it still doesn't meet all your requirements, but AFAICT everything mentioned in the comment is supported in the example

@sramrajkar Yes, it is possible (and quite simple) to modify the IP mesh example to make the root node access the internet via PPP connection. You just need to combine the mesh example with [ap_to_pppos](https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/ap_to_pppos). In fact replace the station part from:

https://github.com/espressif/esp-idf/blob/46acfdce969f03c02b001fe4d24fa9e98f6adc5e/examples/mesh/ip_internal_network/main/mesh_netif.c#L268-L276

to use `esp_modem` connection instead.

closed 2024-10-01 by david-cermak