espressif/esp-idf #8712: 缺少部分网络函数宏定义？

CaineLew, 2022-04-04:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-2349-g4350e6fef8
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): riscv32-esp-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [macOS]
- Using an IDE?: [No] 

## Problem Description

我在移植一个网络组件时提示NI_NUMERICHOST 和NI_NUMERICSERV未定义。
### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.

// 简化的测试代码，以下代码无意义，仅用于测试编译错误，调用getnameinfo函数时需要传递NI_NUMERICHOST 和NI_NUMERICSERV
#include <stdio.h>
#include <netdb.h>
#include <sys/socket.h>

void test()
{
    int s = 0;
    char host[64];
    char port[16];
    struct sockaddr_storage sockaddr;

    if(getnameinfo((struct sockaddr*)&sockaddr, sizeof(sockaddr), host, sizeof(host), port, sizeof(port), NI_NUMERICHOST | NI_NUMERICSERV))
    {
        printf("get name info failed.");
    }
}
void app_main(void)
{
    test();
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

lhespress, 2022-04-07:
@CaineLew 目前是缺少这部分的宏定义，从功能上可以使用 ```getaddrinfo``` 接口替换。

CaineLew, 2022-04-12:
> 

getaddrinfo 是在已知主机名称的情况下，获取地址信息，getnameinfo是在已知地址信息的情况下，获取主机名称（IP地址），功能不能说相似，只能说正好相反。getnameinfo函数在lwip/port/esp32/include/netdb.h中有申明，但没有在源文件中定义实现，现在的解决办法只有自己定义相关的宏，并在代码中用inet_ntop和nstoh这两个函数来实现，只是多出一些额外的代码，影响美观。

MaxwellAlan, 2022-04-18:
@CaineLew 
目前 LwIP 源码中没有实现 getaddrinfo 这个API，因此没有相关的宏定义。
可以尝试提交一个 PR。
Thanks.

closed 2023-02-17 by Alvin1Zhang