espressif/esp-idf #8715: ESP32-S3 WROOM-2 Flash Error, Suggested idf.py menuconfig options (sdkconfig file)?

TGiles1998, 2022-04-04:
## Environment

- Development Kit: ESP32-S3 WROOM-2
- Kit version (DevKitC): 
- Module or chip used: ESP32-S3 WROOM-2
- IDF version: V4.4
- Build System:  idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: macOS
- Using an IDE?: No just Terminal
- Power Supply: USB

## Problem Description

When uploading the the hello world to the devkit, it does not output the desired result from the flash monitor.

### Actual Behavior

idf.py set-target esp32s3
idf.py menuconfig

idf.py build

idf.py --port /dev/cu.usbserial-1460 flash monitor

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd0108,len:0x1648
ets_loader.c 78 

It just keeps repeating this ^^^^

What would the suggested config options be for the octal flash to work?

Many thanks,

Thomas

dobairoland, 2022-04-04:
Maybe the solution is here: https://github.com/espressif/esp-idf/issues/8505#issuecomment-1060367460

TGiles1998, 2022-04-04:
@dobairoland Unfortunately that didn't seem to work. This was the result:

esptool.py v3.2-dev
Serial port /dev/cu.usbserial-1460
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 60:55:f9:f5:55:70
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0003dfff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20896 bytes to 13019...
Writing at 0x00000000... (100 %)
Wrote 20896 bytes (13019 compressed) at 0x00000000 in 0.5 seconds (effective 307.0 kbit/s)...
Hash of data verified.
Compressed 186032 bytes to 97829...
Writing at 0x00010000... (16 %)
Writing at 0x0001b2d3... (33 %)
Writing at 0x00020a80... (50 %)
Writing at 0x0002696d... (66 %)
Writing at 0x0002e827... (83 %)
Writing at 0x00037314... (100 %)
Wrote 186032 bytes (97829 compressed) at 0x00010000 in 2.9 seconds (effective 519.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 521.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor 
Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory "" ' '--port' '/dev/cu.usbserial-1460'"...
--- idf_monitor on /dev/cu.usbserial-1460 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd0108,len:0x1648
ets_loader.c 78 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd0108,len:0x1648
ets_loader.c 78 
ESP-ROM:esp32s3-20210327

This was the config file:

[sdkconfig.pdf](https://github.com/espressif/esp-idf/files/8412027/sdkconfig.pdf)

Any help would be amazing thanks!

igrr, 2022-04-04:
It seems like the Octal mode eFuse might not be burned on your module yet. Could you please run `espefuse.py --port YOUR_PORT summary` and check if `FLASH_TYPE` efuse is set to 1?

(Ref. https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html#error-handling)

TGiles1998, 2022-04-04:
Hi, Yes burning the efuses has solved it, thanks!

closed 2022-04-04 by TGiles1998