espressif/esp-idf #8716: There is no PCNT (pulce counter) driver in library for ESP32-C3

asier70, 2022-04-04:
## Environment

- Development Kit:      [ESP32-DevKitC v4]
- Module or chip used:  [ESP32-C3-WROOM-02]
- IDF version:  v4.4
- Build System:         [idf.py]
- Operating System:     [Windows]

## Problem Description

The Pulse Counter (PCNT) peripheral is shown in datasheet for ESP32-C3 (Fig.1 Block diagram, page 2):
[https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf)
There is no other description nor registers for it in manual.
There is no PCNT header files for ESP32-C3. For other versions they are present.
Espressif forgot about this peripherial in idf library..

### Expected Behavior

It is desirable for a library for PCNT for ESP32-C3 to appear.
I need it for my project.
I trusted the datasheet information.

### Actual Behavior

I use software pulse counting, but for short pulse periods IRQ routine is too slow and some pulses are missed.

igrr, 2022-04-04:
For your project you may also consider using the RMT peripheral in RX mode. It isn't as convenient as PCNT specifically for pulse counting, but it might be still better than doing pulse counting in software with GPIO interrupts.

asier70, 2022-04-04:
Thanks for help.
I think a good microcontroller should have timer with input for external clock.

closed 2022-04-04 by asier70