espressif/esp-idf #8722: ESP32 Crash Report is  not properly formatted

MartinMueller2003, 2022-04-04:
## Environment

- Development Kit:      LoLin D32 PRO / ESP32-CAM and more
- Kit version NA
- Module or chip used:  [ESP32-WROOM-32]
- PLATFORM: Espressif 32 (3.3.1+sha.33ff413)  WEMOS LOLIN D32 PRO
- IDF version (run ``git describe --tags`` to find it):
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0
- Build System:         [platformio]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
unknown
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [Yes platformio]
- Power Supply:         [USB]

## Problem Description

I am experiencing an RMT related crash (This ticket is not for the crash) which results in a back trace being printed to the console. The back trace fails to decode properly using the decoder in platformio (March 2022 (version 1.66)). I have manually decoded the trace and can see that there are issues that need to be resolved (separate question will be generated for that). Her is the defective back trace:

Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0xbad00bad  PS      : 0x00060033  A0      : 0x80081aa2  A1      : 0x3ffbf1e0  
A2      : 0x3ffe8da4  A3      : 0x3ff56000  A4      : 0x01000000  A5      : 0x3ffb7e40
A6      : 0x00000003  A7      : 0x00060f23  A8      : 0x80081a0a  A9      : 0x00000040
A10     : 0x3ffe8bd4  A11     : 0x08000000  A12     : 0x08000000  A13     : 0x00000060
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000014
EXCVADDR: 0xbad00bac  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x7ad00baa:0x3ffbf1e00x40081a9f:0x3ffbf200 0x40085865:0x3ffbf220 0x40087859:0x3ffbf240 0x4008415f:0x3ffb7e90 0x40087da4:0x3ffb7eb0
  #0  0x7ad00baa:0x3ffbf1e00 in ?? ??:0

The issue is a missing space in this text: "0x7ad00baa:0x3ffbf1e00x40081a9f:0x3ffbf200"
                                                                                                     ^^
The characters 0x3ffbf1e0 should have a space before 0x40081a9f:0x3ffbf200

### Expected Behavior
Back Trace string can be decoded

### Actual Behavior
Decoder fails to decode the back trace.

### Steps to reproduce

RMT is transmitting and I update a web page at the same time 100% Crash on RMT interrupt.

// If possible, attach a picture of your setup/wiring here.
NA

MartinMueller2003, 2022-04-05:
Thanks. I will move the issue to pio

closed 2022-04-05 by MartinMueller2003

MitchBradley, 2022-06-10:
I think the resolution is incorrect, and the problem is in esp-idf.  I have observed the "missing space" problem with various ordinary terminal emulators.  Here is the broken code, in components/esp_system/port/arch/xtensa/debug_helpers.c:

```c
esp_err_t IRAM_ATTR esp_backtrace_print_from_frame(int depth, const esp_backtrace_frame_t* frame, bool panic)
{
    //Check arguments
    if (depth <= 0) {
        return ESP_ERR_INVALID_ARG;
    }

    //Initialize stk_frame with first frame of stack
    esp_backtrace_frame_t stk_frame = { 0 };
    memcpy(&stk_frame, frame, sizeof(esp_backtrace_frame_t));

    print_str("\r\n\r\nBacktrace:", panic);
    print_entry(esp_cpu_process_stack_pc(stk_frame.pc), stk_frame.sp, panic);
!!!!!! There should be ***    print_str(" ", panic);    *** right here
    //Check if first frame is valid
    bool corrupted = !(esp_stack_ptr_is_sane(stk_frame.sp) &&
                       (esp_ptr_executable((void *)esp_cpu_process_stack_pc(stk_frame.pc)) ||
                        /* Ignore the first corrupted PC in case of InstrFetchProhibited */
                        (stk_frame.exc_frame && ((XtExcFrame *)stk_frame.exc_frame)->exccause == EXCCAUSE_INSTR_PROHIBITED)));

    uint32_t i = (depth <= 0) ? INT32_MAX : depth;
    while (i-- > 0 && stk_frame.next_pc != 0 && !corrupted) {
        if (!esp_backtrace_get_next_frame(&stk_frame)) {    //Get previous stack frame
            corrupted = true;
        }
        print_entry(esp_cpu_process_stack_pc(stk_frame.pc), stk_frame.sp, panic);
        print_str(" ", panic);
    }
```
In the old version of esp-idf, the corresponding code called a routine named putEntry which put a space before each entry.  The new code puts the space after the entry, but neglects to do it after the first entry, thereby splicing together the first and second entries with no intervening space.

idf.py monitor just papers over the problem by coping with the bad backtrace line.