espressif/esp-idf #8724: Memory leak in DPP

jkent, 2022-04-05:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      none/ODROID-GO
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // v5.0-dev-2349-g4350e6fef8
- Operating System:     Linux
- Using an IDE?: vscode
- Power Supply:         USB/Battery

## Problem Description

DPP has a bad memory leak when you de-init it and re-init it.

### Expected Behavior

Heap to be nearly the same between multiple calls.

### Actual Behavior

Crash once the heap reaches ~8K free.

### Steps to reproduce

Try minimal demo code provided.

### Code to reproduce this issue

https://gist.github.com/jkent/b1e613cce212780ded0ca731f2385ba3

## Debug Logs

```
I (3931) wifi:mode : sta (b4:e6:2d:95:df:f9)
I (3931) wifi:enable tsf
I (4031) wifi dpp-enrollee: <qrcode here>
I (4031) wifi dpp-enrollee: esp_supp_dpp_start_listen()
I (4031) wifi:flush txq
I (4031) wifi:stop sw txq
I (4041) wifi:lmac stop hw txq
I (4041) wifi dpp-enrollee: heap:      14640
I (4051) wifi:mode : sta (b4:e6:2d:95:df:f9)
I (4051) wifi:enable tsf
I (4151) wifi dpp-enrollee: <qrcode here>
I (4151) wifi dpp-enrollee: esp_supp_dpp_start_listen()
I (4151) wifi:flush txq
I (4151) wifi:stop sw txq
I (4151) wifi:lmac stop hw txq
I (4151) wifi dpp-enrollee: heap:       7244
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40087d6c
0x40087d6c: _esp_error_check_failed at /home/jkent/usr/esp-idf/components/esp_system/esp_err.c:42

file: "../main/main.c" line 66
func: dpp_enrollee_init
expression: esp_supp_dpp_init(dpp_enrollee_event_cb)

abort() was called at PC 0x40087d6f on core 0
0x40087d6f: _esp_error_check_failed at /home/jkent/usr/esp-idf/components/esp_system/esp_err.c:43

Backtrace:0x40081b96:0x3ffba4b00x40087d79:0x3ffba4d0 0x4008eb16:0x3ffba4f0 0x40087d6f:0x3ffba560 0x400d8194:0x3ffba580 0x400d8217:0x3ffba660 0x4014fd69:0x3ffba680 0x4008b165:0x3ffba6a0 
0x40081b96: panic_abort at /home/jkent/usr/esp-idf/components/esp_system/panic.c:409

0x40087d79: esp_system_abort at /home/jkent/usr/esp-idf/components/esp_system/esp_system.c:135

0x4008eb16: abort at /home/jkent/usr/esp-idf/components/newlib/abort.c:46

0x40087d6f: _esp_error_check_failed at /home/jkent/usr/esp-idf/components/esp_system/esp_err.c:43

0x400d8194: dpp_enrollee_init at /home/jkent/Projects/espressif/esp32/odroid-go/dpptest/build/../main/main.c:66 (discriminator 1)

0x400d8217: app_main at /home/jkent/Projects/espressif/esp32/odroid-go/dpptest/build/../main/main.c:98

0x4014fd69: main_task at /home/jkent/usr/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:127 (discriminator 2)

0x4008b165: vPortTaskWrapper at /home/jkent/usr/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

jkent, 2022-04-07:
There is a serious regression as well, which I didn't know about at the time.

I've tried the wifi_easy_connect demo with a previous version of IDF (no idea what version it was at the time) and it was working.  Trying the wifi_easy_connect demo with my Pixel 3 does not seem to do anything any longer.  The Pixel 3 says "Make sure the device has been plugged in, charged, and turned on."

```
I (673) wifi_init: rx ba win: 6
I (683) wifi_init: tcpip mbox: 32
I (683) wifi_init: udp mbox: 6
I (683) wifi_init: tcp mbox: 6
I (693) wifi_init: tcp tx win: 5744
I (693) wifi_init: tcp rx win: 5744
I (703) wifi_init: tcp mss: 1440
I (703) wifi_init: WiFi IRAM OP enabled
I (703) wifi_init: WiFi RX IRAM OP enabled
I (813) wifi dpp-enrollee: Scan below QR Code to configure the enrollee:

I (813) QRCODE: Encoding below text with ECC LVL 0 & QR Code Version 10
I (813) QRCODE: DPP:C:81/6;M:b4:e6:2d:95:df:f9;K:MDkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDIgAD6nYeG3MgCSFvTIjMVNVdYSQA6e/O463cHOB2fFz8KV4=;;
<snip>
I (1153) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1253) wifi:mode : sta (b4:e6:2d:95:df:f9)
I (1253) wifi:enable tsf
I (1263) wifi dpp-enrollee: Started listening for DPP Authentication
```

nachiketkukade, 2022-04-14:
Thanks @jkent for reporting this issue. The memory leak is caused by failure in deleting the DPP task, which will be fixed soon.
For the second issue, support for DPP (WiFi Easy Connect) was added in IDF v4.3, while on Android side, it was added in Android 10. So please confirm if you're using softwares above these versions.

jkent, 2022-04-15:
I am using the master branch of IDF5.0, and I have a Google Pixel 3 running Android 12.  I previously had DPP running with this device.  The QR code decodes with the camera app fine, but when scanning the QR code from within WiFi settings it times out.

I've also tested this with a Pixel 4 with the same result.  I don't have any other devices to test scanning with at the moment.

nachiketkukade, 2022-04-18:
@jkent , I've just tested the feature with a Pixel 4A running Android 12 and it seems to be working fine. Are you by any chance using the "**Add network**" option from "**Internet**" menu for scanning QR code? The correct way is to go inside additional settings of the connected AP, then use the "**Add device**" for scanning the QR code.

closed 2022-04-19 by espressif-bot

jkent, 2022-04-21:
@nachiketkukade Sorry, yes it does work.  I was trying from the QR code icon next to "Add network."

Thanks for the resolution.