espressif/esp-idf #8726: the idf.py monitor baudrate setting in sdkconfig has no effect or is ignored

moefear85, 2022-04-05:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version : v4.4-c29343eb94d2f2ca17b3a5b38c82452e556147f2
- Build System:         idf.py
- OS: Linux
- Using an IDE?: Yes (VSCODE IDF PLUGIN]
- Power Supply:      external 3.3V

## Problem Description
setting the idf.py monitor baudrate in sdkconfig has to effect. All entries still cause the serial output to be at 115200. So even if I choose 921600bps, the output remains at 115200. If I use an external monitor that I set to 115200, I can see the output normally. But when launching the idf.py monitor, since it expects 921600, all I get is junk. There is absolutely nothing in my code that is modifying the baudrate or touching the uart in any way. I only use ESP_LOG and printf functions. I never set any environment variables that would change the baudrate. MONITORBAUD is not set. It seems the chip doesn't respect this sdkconfig entry and ignores it.

moefear85, 2022-04-05:
I'm setting CONFIG_ESPTOOLPY_MONITOR_BAUD to 921600. But this is not having an effect. When I open any serial console @115200, it works fine. When I do so with @921600, I see jibberish. So when idf.py tries to open the monitor @921600, I see jibberish. I don't understand why setting CONFIG_ESPTOOLPY_MONITOR_BAUD is being ignored by the esp chip/app/uart (while idf.py monitor correctly opens in 921600).

However, CONFIG_ESP_CONSOLE_UART_BAUDRATE is set to 115200 in my sdkconfig, even though I never set it anywhere in menuconfig. In menuconfig, under "serial flasher config->idf.py monitor baud rate" I'm choosing "921600". Thus the variable CONFIG_ESPTOOLPY_MONITOR_BAUD is correctly being set to 921600, but when I inspect the sdkconfig file itself, I realize CONFIG_ESP_CONSOLE_UART_BAUDRATE is set to 115200, even though I never chose this. Could this be the source of the problem? I'm going to test it right now.

moefear85, 2022-04-05:
@dobairoland 

I just realized, CONFIG_ESP_CONSOLE_UART_BAUDRATE is automatically being set to 115200 in the sdkconfig file everytime I flash the application. Even if I delete it or comment it out, once I hit the flash button, the file gets instantly modified and it reappears. So i have no way to test things with this option off. Is there a logical reason for this?

dobairoland, 2022-04-05:
> I don't understand why setting CONFIG_ESPTOOLPY_MONITOR_BAUD is being ignored by the esp chip/app/uart

Because changing the monitor baud rate doesn't imply or change the console baud rate.

dobairoland, 2022-04-05:
So I think the solution for your problem is to set the monitor baud rate to follow the console baud rate; and change the desired baud rate for the console only.

This is the default behavior for ESP-IDF v5.0. You will not have to set it there.

dobairoland, 2022-04-05:
First, you have to set custom UART. Then it will appear. If you don't enable it then custom baud-rate is hidden in the menu (if you set it manually then it will ignore it).

moefear85, 2022-04-05:
@dobairoland 

I tried that... that only affects CONFIG_ESPTOOLPY_MONITOR_BAUD. It doesn't change CONFIG_ESP_CONSOLE_UART_BAUDRATE. What I need is a way to persistently change CONFIG_ESP_CONSOLE_UART_BAUDRATE. What exactly is changing it back to 115200 on every flash?

dobairoland, 2022-04-05:
```
CONFIG_ESP_CONSOLE_UART_CUSTOM=y
CONFIG_ESP_CONSOLE_UART_BAUDRATE=921600
```

dobairoland, 2022-04-05:
> What exactly is changing it back to 115200 on every flash?

... is what I mentioned earlier. If you don't set custom UART then reconfiguration will ignore the baud-rate change.

moefear85, 2022-04-05:
@dobairoland that gives an error. the build system doesn't accept when I define it manually. it just overwrites both lines

moefear85, 2022-04-05:
there must be some other option that has to be set first to allow CONFIG_ESP_CONSOLE_UART_CUSTOM to be set to "y".

moefear85, 2022-04-05:
@dobairoland 
can you please tell me the exact path from the top of menuconfig do I have to navigate to reach the option that allows me to set this from menuconfig? because doing so directly in the sdkconfig file.... just to have it throw errors in the menuconfig window, then overwrite stuff, then allow it strangely to work, is ofcourse not a systematic solution.

dobairoland, 2022-04-05:
"Component config" -> "ESP system settings" -> "Channel for console output (Default: UART0)"

dobairoland, 2022-04-05:
> only then do submenu options appear that let you keep or change the uart number, its pins, and also set its output baudrate.

Yes, that is what I was trying to tell you. I'm sorry if it wasn't clear enough.