espressif/esp-idf #8728: Build amazon_freertos  esp32 "aws_demo" into Espressif IDE for esp32

WilliamFrasson, 2022-04-05:
I'm encountering problems in configure, generate and compile the esp32 aws_demo into Espressif IDE.

here below when i try to compile from aws_demo from the Espressif IDE; i guess the problem is due to some set into CMakeLists.txt

here my steps:

here my steps:

folder with amazon-freertos is : "D:\esp_test4\aws\amazon-freertos"
i have create a new project by import the it into Espressif IDE as "Existing IDF project" importing from "D:\esp_test4\aws\amazon-freertos"
for this project, in edit configuration, I have added the "Additional CMake arguments" : -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32
into the workspace environment i have changed the IDF_PATH to "D:\esp_test4\aws\amazon-freertos\vendors\espressif\esp-idf"
into the workspace environment i have changed the P
[build_failed.txt](https://github.com/espressif/esp-idf/files/8421473/build_failed.txt)
ATH to the same as achieved calling from a shell:
vendors\espressif\esp-idf\export.bat
now i build the project using the build button
this is the log:
[build_failed.txt](https://github.com/espressif/esp-idf/files/8421484/build_failed.txt)

can someone help in fix the cmakefiles?

AvinashSinghPES, 2022-04-28:
how to configuration "Additional CMake arguments" : -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 ? describe

WilliamFrasson, 2022-04-28:
here below when i try to compile from aws_demo from the Espressif IDE

i guess the problem is how to compile using the CMake;

here my steps:

folder with amazon-freertos is : "D:\esp_test4\aws\amazon-freertos"
i have create a new project by import the it into Espressif IDE as "Existing IDF project" importing from "D:\esp_test4\aws\amazon-freertos"
for this project, in edit configuration, I have added the "Additional CMake arguments" : -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32
into the workspace environment i have changed the IDF_PATH to "D:\esp_test4\aws\amazon-freertos\vendors\espressif\esp-idf"
into the workspace environment i have changed the PATH to the same as achieved calling from a shell:
vendors\espressif\esp-idf\export.bat
now i build the project

closed 2022-04-28 by WilliamFrasson