espressif/esp-idf #8730: [v4.4] bootloader_common_erase_part_type_data("nvs", false) is failing

chipweinberger, 2022-04-05:
ESP32-S3
ESP-IDF release/v4.4 (https://github.com/espressif/esp-idf/commit/c29343eb94d2f2ca17b3a5b38c82452e556147f2) from a few days ago.

`bootloader_common_erase_part_type_data()` does not seem to be working. It looks like it is just reseting the CPU without doing any work?

I have some simple code in my bootloader:

```
static void factory_reset() {

    ESP_LOGI(TAG, "Performing Factory Reset");
        
    bool ota_data_erase_no = false;
    const char *list_erase = "nvs";

    ESP_LOGI(TAG, "Factory Reset: Data partitions to erase: %s", list_erase);

    // delete nvs
    if (bootloader_common_erase_part_type_data(list_erase, ota_data_erase_no) == false) {
        ESP_LOGE(TAG, "Factory Reset: Not all partitions were erased");
    }

    beep_factory_reset();

    ESP_LOGI(TAG, "Factory Reset Complete");

    // wait until user lets go of the button
    gpio_wait_while_low(PD_GPIO_USER_BUTTON);
}
```

```
I (10509) boot: Performing Factory Reset
I (10510) boot: Factory Reset: Data partitions to erase: nvs
I (10516) boot_comm: ## Label            Usage Offset   Length   Cleaned
I (10523) boot_comm:  0 otadata          data  00009000 00002000 [no]
I (10530) boot_comm:  1 phy_init         data  0000b000 00001000 [no]
I (10538) boot_comm:  2 coredump         data  0000c000 00010000 [no]
I (10545) boot_comm:  3 coredumpInfo     data  0001c000 00004000 [no]
I (10552) boot_comm:  4 hardwareInfo     data  00020000 00002000 [no]
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1a0c
load:0x403b6000,len:0xdb8
load:0x403ba000,len:0x324c
entry 0x403b6398
I (25) boot: ESP-IDF v4.4-367-gc29343eb94-dirty 2nd stage bootloader
I (25) boot: compile time 15:32:28
I (26) boot: chip revision: 0
```

And a custom partition table

```
# Name,          Type, SubType,  Offset,   Size,  Flags
otadata,         data, ota,      ,           8K
phy_init,        data, phy,      ,           4K
coredump,        data, coredump, ,          64K
coredumpInfo,    data, undefined,,          16K
hardwareInfo,    data, undefined,,           8K
nvs,             data, nvs,      ,         128K
nvs_key,         data, nvs_keys, ,           4K
mainApp,    app,  ota_0,    ,        2280K
jfuApp,         app,  ota_1,    ,        1400K
extra1,          data, undefined,,           8K
extra2,          data, undefined,,           8K
```

chipweinberger, 2022-04-06:
Hmm. I added some more logs statements to `bootloader_common_erase_part_type_data` and now its restarting even earlier. Is there some bootloader timeout?

```
I (9422) boot: User Button held 9 second
I (10422) boot: User Button held 10 second
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1b5c
```

chipweinberger, 2022-04-06:
This seems to be the cuplrit:

[image]

KonstantinKondrashov, 2022-04-20:
Hi @chipweinberger!
Yes, you are right. You got `RTCWDT_RTC_RST` because the erasing is exceeded Timeout for RTC watchdog (default BOOTLOADER_WDT_TIME_MS=9000ms) which is started in the init of bootloader and stopped in the app.

KonstantinKondrashov, 2022-04-20:
@chipweinberger 
Have you resolved this issue?

We can fix it by reseting RTC timer before erasing a partition.

```diff
diff --git a/components/bootloader_support/src/bootloader_common.c b/components/bootloader_support/src/bootloader_common.c
index 709504d..8208d8b 100644
--- a/components/bootloader_support/src/bootloader_common.c
+++ b/components/bootloader_support/src/bootloader_common.c
@@ -18,6 +18,7 @@
 #include "bootloader_flash_priv.h"
 #include "bootloader_common.h"
 #include "bootloader_utility.h"
+#include "hal/wdt_hal.h"
 #include "soc/gpio_periph.h"
 #include "soc/rtc.h"
 #include "soc/efuse_reg.h"
@@ -119,6 +120,12 @@ bool bootloader_common_erase_part_type_data(const char *list_erase, bool ota_dat
                 // partition->label is not null-terminated string.
                 strncpy(label, (char *)&partition->label, sizeof(label) - 1);
                 if (fl_ota_data_erase == true || (bootloader_common_label_search(list_erase, label) == true)) {
+#ifdef BOOTLOADER_BUILD
+                    wdt_hal_context_t rtc_wdt_ctx = {.inst = WDT_RWDT, .rwdt_dev = &RTCCNTL};
+                    wdt_hal_write_protect_disable(&rtc_wdt_ctx);
+                    wdt_hal_feed(&rtc_wdt_ctx);
+                    wdt_hal_write_protect_enable(&rtc_wdt_ctx);
+#endif // BOOTLOADER_BUILD
                     err = bootloader_flash_erase_range(partition->pos.offset, partition->pos.size);
                     if (err != ESP_OK) {
                         ret = false;
```

chipweinberger, 2022-04-20:
Thanks for the code snippet! I might use that.

Ah yes I see;
rst:0x10 (RTCWDT_RTC_RST)

Is it possible to log before reset too? "Resetting due to bootloader wdt timeout..."

chipweinberger, 2022-04-20:
also, would be more clear to end users to rename it to

reset reason:0x10 (RTCWDT_RTC_RST)

instead of rst.

chipweinberger, 2022-04-20:
Appreciate the additional comments.

Yes the documentation change seems worthwhile :)

I'll close this issue since there's not much else that can be done!

closed 2022-04-20 by chipweinberger