espressif/esp-idf #8736: undefined reference function in static library

Eran125, 2022-04-06:
Hi,
I create project that include NVS function and build it. The functions that use NVS located in separate folder so I can use the .a file as static library in other project. I copy the .a file to a new project and add the library in CMakeLists.txt file.
The build fail with the error "undefined reference to `nvs_get_i8'" - this function is called in my library (there were no errors when I build the library.

igrr, 2022-04-06:
Hi @Eran125, 

>  I copy the .a file to a new project and add the library in CMakeLists.txt file.

Could you please attach your CMakeLists.txt file here?

Also please do provide the information requested in the issue template you were presented with when opening a new issue: ESP-IDF version, build system, chip you are using, operating system and so on.

Eran125, 2022-04-06:
Hi,
This is my CMakeLists.txt file:

idf_component_register(
    SRCS "main.c"
    INCLUDE_DIRS "."
    REQUIRES nvs_flash
    REQUIRES mbedtls
)

add_library(libInternal_flash STATIC IMPORTED GLOBAL)
set_property(TARGET libInternal_flash PROPERTY IMPORTED_LOCATION ${COMPONENT_DIR}/../main/../Internal_Flash/libInternal_flash.a)
target_link_libraries(${COMPONENT_TARGET} PUBLIC libInternal_flash)

igrr, 2022-04-06:
I see, the error makes sense then! The `libInternal_flash` CMake target doesn't yet have a dependency on `nvs_flash` library, so CMake doesn't know that it needs to place `libInternal_flash` ahead of `nvs_flash` on the linker command line. 

You need to let CMake know that `libInternal_flash.a` itself has some dependencies.

For this purpose, ESP-IDF build system provides a helper function, `add_prebuilt_library` ([docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#using-prebuilt-libraries-with-components)), which allows you to import such a prebuilt library and also specify its dependencies. Please refer to the [add_prebuilt example](https://github.com/espressif/esp-idf/blob/4350e6fef859b94f3efae6ceced751911c9d2cbc/examples/build_system/cmake/import_prebuilt/main/CMakeLists.txt#L4-L7).

Eran125, 2022-04-07:
Hi,
I just fix the Cmake file as you explain and it works great.
Thank you.

0xjakob, 2022-04-07:
@Eran125 Happy to see that it works for you! May we close this issue or is there anything else related to this problem?

Eran125, 2022-04-07:
Yes, you may.

closed 2022-04-07 by Eran125