espressif/esp-idf #8738: Misleading name of USB-console example

JVKran, 2022-04-06:
**Describe the solution you'd like**

USB-log is more appropriate since the esp-console component isn't supported as of writing this request. Hence, one isn't able to issue commands.

**Describe alternatives you've considered**

Not using USB and just going for uart. What's the point of using USB as replacement for flashing and logging if the console doesn't work?

igrr, 2022-04-07:
I see, agree the better name for this example would be something like tusb_stdout or such.

But aside from the example name, do I understand it well that your feature request is that you want to use `console` component on ESP32-S3 with USB_OTG peripheral? (I.e. [this example](https://github.com/espressif/esp-idf/tree/master/examples/system/console_usb) which is only supported on ESP32-S2 now?)

JVKran, 2022-04-07:
Well, yes; that's right. Wasn't directly my intention, but you indeed managed to find the cause for my adventures regarding usb and consoles. 😜 

Ideally I would like to use the [basic console example](https://github.com/espressif/esp-idf/blob/master/examples/system/console/basic) in combination with USB. Hence, instead of using ```ESP_CONSOLE_DEV_UART_CONFIG_DEFAULT``` at [main.c:96](https://github.com/espressif/esp-idf/blob/master/examples/system/console/basic/main/console_example_main.c#L69) the ```ESP_CONSOLE_DEV_CDC_CONFIG_DEFAULT``` which is provided as empty structure in esp_console.h.

igrr, 2022-04-07:
Got it, indeed we have a task open internally for this. We need to add support for USB_CDC as system console for ESP32-S3 (`CONFIG_ESP_CONSOLE_USB_CDC` option). There are a couple fixes required to get this to work which aren't finished yet.

Once this is done, it should be possible to use `esp_console_new_repl_usb_cdc` on ESP32-S3 as well. Currently this works only on ESP32-S2.

We'll update this issue once the problem is solved.

udem1234, 2022-11-18:
Hi !  Any update on this ?  Thank you very much in advance !

closed 2023-04-18 by espressif-bot