espressif/esp-idf #8741: I2C communication failures on ESP32-S3 (works on ESP32)

CSHaze, 2022-04-07:
## Environment

- Development Kit:  ESP32-S3-DevKitC-1
- Kit version DevKitC: v1
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version: v4.4
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:  Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes, VSCode ESP-IDF extension
- Power Supply: USB

## Problem Description

I2C device communication that works on the standard ESP32 does not work on the ESP32-s3.

### Expected Behavior

I would expect I2C functionality to work similarly on both devices.

### Actual Behavior

I2C writes work correctly on the standard ESP32 but do not work on the ESP32-s3.

### Steps to reproduce

1. Attach an I2C device (I am using an HTU21d)
2. Run the scan program on a standard ESP32 (https://github.com/UncleRus/esp-idf-i2cscan)
3. Observe the scan shows the correct output.
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
4. Run the scan program on an ESP32-s3
5. Observe the scan does not find the device

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

**Note: I have tried other libraries, communication calls, and configurations (pullups, clk freq, gpio, etc) through IC2, all of which are unable to write to the device. I figured this was the simplest example.**

### Code to reproduce this issue

```cpp
#include <driver/i2c.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>

#ifndef APP_CPU_NUM
#define APP_CPU_NUM PRO_CPU_NUM
#endif

#define SDA_PIN 21
#define SCL_PIN 22

static const char *TAG = "i2cscanner";

void task(void *ignore)
{

    i2c_config_t conf;
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = SDA_PIN;
    conf.scl_io_num = SCL_PIN;
    conf.sda_pullup_en = false;
    conf.scl_pullup_en = false;
    conf.master.clk_speed = 400000;
    i2c_param_config(I2C_NUM_0, &conf);

   i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0);

    while (1)
    {
        esp_err_t res;
        printf("     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f\n");
        printf("00:         ");
        for (uint8_t i = 3; i < 0x78; i++)
        {
            i2c_cmd_handle_t cmd = i2c_cmd_link_create();
            i2c_master_start(cmd);
            i2c_master_write_byte(cmd, (i << 1) | I2C_MASTER_WRITE, 1 /* expect ack */);
            i2c_master_stop(cmd);

            res = i2c_master_cmd_begin(I2C_NUM_0, cmd, 10 / portTICK_PERIOD_MS);
            if (i % 16 == 0)
                printf("\n%.2x:", i);
            if (res == 0)
                printf(" %.2x", i);
            else
                printf(" --");
            i2c_cmd_link_delete(cmd);
        }
        printf("\n\n");
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}

void app_main()
{
    // Start task
    xTaskCreatePinnedToCore(task, TAG, configMINIMAL_STACK_SIZE * 8, NULL, 5, NULL, APP_CPU_NUM);
}
```

ginkgm, 2022-04-07:
Hi @CSHaze ,
Have you checked what's on the lines by a logic analyzer?
And are the lines pulled up by resistors correctly?

Michael

CSHaze, 2022-04-07:
Hi @ginkgm, unfortunately I do not have access to a logic analyzer.

The [Htu21d](https://www.sparkfun.com/products/retired/12064?__hstc=77938635.2ea4a41c0c09b1d7d2bf704e17e9b70e.1484956800128.1484956800129.1484956800130.1&__hssc=77938635.1.1484956800131&__hsfp=528229161) board I am using has built in pull up resistors. I have also tried unsoldering them and using the esp32-s3 internal pull ups and neither seems to work on the esp32-s3. The weird thing is everything works correctly on the old esp32 I have. I am trying to swap to the newer board.

I should be getting a bme280 board in the next couple days and I will test with that.

Are there differences in the i2c config or calls between the esp32 and esp32-s3? I did not see any differences in the docs.

CSHaze, 2022-04-08:
Hi @ginkgm and @igrr. I just received a couple bme280 modules and i2c seems to be working fine. 

Maybe I just have a couple of shoddy HTU21d modules. It is a little weird they seemed to work on the old esp32 though. Maybe due to differences in the internal clocks between the new and old esp32s?

dhalbert, 2022-05-09:
We are also seeing some I2C issues on ESP32-S3, though it's in comparison with ESP32-S2. Specifically, LC709203, an I2C battery meter, does not work reliably. It can work longer at slow I2C bus speeds (tested at 10kHz) but eventually produces I2C protocol errors. Other devices such as MSA301 or BNO055 (which can do clock stretching) seen OK at normal speeds. We're still trying to characterize the problem.

dhalbert, 2022-05-12:
LC709203F comparison: something holds SCL down after a successful write on S3, but not S2. SCL eventually rises after 3.8ms

dhalbert, 2022-05-12:
I cannot reproduce https://github.com/espressif/esp-idf/issues/8741#issuecomment-1124457602 in Arduino on S3, using https://github.com/adafruit/Adafruit_LC709203F/tree/master/examples/LC709203F_demo and removing the delay in `loop()`. There are clock-stretches, but they don't cause issues above. The example above is from CircuitPython, which has an extra task and some other complications.

lebossejames, 2023-01-16:
Hello,

I have same issue with ESP32-S3 Arduino IDE, i2C plug to lcd screen.

Papirrruqui, 2023-02-04:
Hello, I just got a ESP32-S3HDevKitv1, and can't connect a I2C device to it. Can any one point me an example of how to configure the pins for I2C communication

riker65, 2023-02-25:
same for me with heltec wifi kit 32 V3, no platformio module found for i2c but in arduino it is working with V3

illuminati-vs, 2023-03-17:
Hey how did you configure pin 15&16 as i2c pins in ESP32-S3-Wroom-1, or are they by default i2c pins. Please help!!

riker65, 2023-03-20:
Hi, on the heltec I used

`i2c:

  sda: GPIO17
  scl: GPIO18

#  sda: GPIO4
#  scl: GPIO15
  scan: false
#  frequency: 700khz
  frequency: 50khz`

Martius108, 2023-04-05:
Hi riker65,

could you please post a full code example for Heltec boards? I tried to establish I2C connections to a BME280 sensor on CubeCell HTTC AB-02 and ESP32 LoRa boards (both V3) but nothing works. I always get "no I2C device found" I tried the second I2C channel which should be SDA = pin9 and SCL = pin8

riker65, 2023-04-05:
Hi 

see here, 

just my prototype to get it working

hope this helps

`#https://forum.hassiohelp.eu/d/536-display-oled-su-esp32-con-esphome
#https://github.com/Heltec-Aaron-Lee

esphome:
  name: heltec2
  platform: ESP32
#   platform: ESP32-S3
#  board: heltec_wifi_kit_32(V3)
#  board: heltec_wifi_kit_32  
  board: esp32-s3-devkitc-1
#  restore_from_flash: true
#  board:  esp32s3

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Oled2 Fallback Hotspot"
    password: "Zf9oas35HAIZ"
captive_portal:
# Enable logging
logger:
# Enable Home Assistant API

api:
  reboot_timeout: 0s
  
ota:

# Example configuration entry
web_server:
  port: 80
  
   
  
#https://esphome.io/components/switch/gpio.html
switch:
  - platform: gpio
    pin:
      number: 36
    id: mode
#      inverted: true
    restore_mode: RESTORE_DEFAULT_OFF 
      
 #     https://esphome.io/components/switch/gpio.html
#switch: 
  - platform: gpio    
    name: "LED_Int"
    pin: 
      number: 35
    id: LED_Int
    icon: "mdi:gate"    
#    name: "LED_Int"
    inverted: false
    restore_mode: RESTORE_DEFAULT_OFF 

i2c:

  sda: GPIO17
  scl: GPIO18
  scan: false
#  frequency: 700khz
  frequency: 50khz

  
font:
- file: 'arialblk.ttf'
  id: font1
  size: 30
- file: 'arial.ttf'
  id: font4
  size: 12
- file: 'arial.ttf'
  id: font2
  size: 12
- file: 'arial.ttf'
  id: font3
  size: 10

time:
  - platform: sntp
    id: sntp_time   
    

    
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: GPIO21
#    reset_pin: D0    
    address: 0x3C
    setup_priority: 0
    id: my_display
    pages:
      - id: page1
        lambda: |-
           it.printf(0, 0, id(font3), "WP WT:%s",id(WPTemp).state.c_str());
           it.printf(70, 0, id(font3), "-%s",id(WPWT_TS).state.c_str());
           it.printf(0, 36, id(font3),"TS: %s",id(WPWT_TS).state.c_str());
           it.printf(0, 12, id(font3), "LWTLOL:%s",id(WPLOLLWT).state.c_str());
           it.printf(0, 24, id(font3), "LWTSICH:%s",id(WPSICHLWT).state.c_str());
           it.printf(100, 12, id(font4), TextAlign::TOP_CENTER, "TR");
           it.strftime(0, 52, id(font2), "%Y-%m-%d %H:%M:%S", id(sntp_time).now());  

  
# Example configuration entry
#https://esphome.io/components/mqtt.html
mqtt:
  broker: 192.168.0.1
  username: !secret mqtt_user
  password: !secret mqtt_password

  
text_sensor:
  - platform: mqtt_subscribe
    name: "mqttWPTemp"
    id: WPTemp
    topic: WP/Was_temp 
  - platform: mqtt_subscribe
    name: "WPWT_TS"
    id: WPWT_TS
    topic: WP/LOL/WT_PS 
  - platform: mqtt_subscribe
    name: "mqttWPLOLLWT"
    id: WPLOLLWT
    topic: WP/LOL/LWT     
`

riker65, 2023-04-05:
Hi

see here,

just my prototype to get it working

hope this helps

pasting it as code leads to strange formating

here without code tags

-----------------start
#https://forum.hassiohelp.eu/d/536-display-oled-su-esp32-con-esphome
#https://github.com/Heltec-Aaron-Lee

esphome:
  name: heltec2
  platform: ESP32
#   platform: ESP32-S3
#  board: heltec_wifi_kit_32(V3)
#  board: heltec_wifi_kit_32  
  board: esp32-s3-devkitc-1
#  restore_from_flash: true
#  board:  esp32s3

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Oled2 Fallback Hotspot"
    password: "Zf9oas35HAIZ"
captive_portal:
# Enable logging
logger:
# Enable Home Assistant API

api:
  reboot_timeout: 0s
  
ota:

# Example configuration entry
web_server:
  port: 80
  
   
  
#https://esphome.io/components/switch/gpio.html
switch:
  - platform: gpio
    pin:
      number: 36
    id: mode
#      inverted: true
    restore_mode: RESTORE_DEFAULT_OFF 
      
 #     https://esphome.io/components/switch/gpio.html
#switch: 
  - platform: gpio    
    name: "LED_Int"
    pin: 
      number: 35
    id: LED_Int
    icon: "mdi:gate"    
#    name: "LED_Int"
    inverted: false
    restore_mode: RESTORE_DEFAULT_OFF 

i2c:

  sda: GPIO17
  scl: GPIO18
  scan: false
#  frequency: 700khz
  frequency: 50khz

  
font:
- file: 'arialblk.ttf'
  id: font1
  size: 30
- file: 'arial.ttf'
  id: font4
  size: 12
- file: 'arial.ttf'
  id: font2
  size: 12
- file: 'arial.ttf'
  id: font3
  size: 10

time:
  - platform: sntp
    id: sntp_time   
    

    
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: GPIO21
#    reset_pin: D0    
    address: 0x3C
    setup_priority: 0
    id: my_display
    pages:
      - id: page1
        lambda: |-
           it.printf(0, 0, id(font3), "WP WT:%s",id(WPTemp).state.c_str());
           it.printf(70, 0, id(font3), "-%s",id(WPWT_TS).state.c_str());
           it.printf(0, 36, id(font3),"TS: %s",id(WPWT_TS).state.c_str());
           it.printf(0, 12, id(font3), "LWTLOL:%s",id(WPLOLLWT).state.c_str());
           it.printf(0, 24, id(font3), "LWTSICH:%s",id(WPSICHLWT).state.c_str());
           it.printf(100, 12, id(font4), TextAlign::TOP_CENTER, "TR");
           it.strftime(0, 52, id(font2), "%Y-%m-%d %H:%M:%S", id(sntp_time).now());  

  
# Example configuration entry
#https://esphome.io/components/mqtt.html
mqtt:
  broker: 192.168.0.1
  username: !secret mqtt_user
  password: !secret mqtt_password

  
text_sensor:
  - platform: mqtt_subscribe
    name: "mqttWPTemp"
    id: WPTemp
    topic: WP/Was_temp 
  - platform: mqtt_subscribe
    name: "WPWT_TS"
    id: WPWT_TS
    topic: WP/LOL/WT_PS 
  - platform: mqtt_subscribe
    name: "mqttWPLOLLWT"
    id: WPLOLLWT
    topic: WP/LOL/LWT     

-----------------end

slimcdk, 2023-06-28:
@riker65 Not relevant here. ESPHome currently uses Arduino as core unless you explicitly define otherwise. Your question should be on their project instead.

wBrhy2, 2023-09-21:
I simply want to comment that I have observed similar issues as well where certain i2c chips behave differently between Wroom-32 and Wroom-S2/S3 as well. Specifically VL6180X chips and i2c touch panel. It renders those devices completely unusable on newer ESP32 so I look forward to seeing what you find.

iyansrvc, 2025-01-11:
Same probelm I2C Not working on YD-ESP32-S3-N16-R8 (DeveKit-c1 clone)

mythbuster5, 2025-01-13:
Which version are you use @iyansrvc ? With same stuff?

Shravani-03, 2025-01-20:
> Same probelm I2C Not working on YD-ESP32-S3-N16-R8 (DeveKit-c1 clone)

I have this exact board with the same problem, any help would be appreciated

mythbuster5, 2025-01-21:
@Shravani-03 May I see your i2c related code and board?

iyansrvc, 2025-01-28:
my board YD-ESP32-S3 (WROOM-1_N16R8) I2C not working.