espressif/esp-idf #8744: Default AP SSID name not working

filgra96, 2022-04-07:
## Environment

- Development Kit:      ESP32-S2 MINI-1 custom board
- Module or chip used:  ESP32-S2 MINI-1
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:         idf.py
- Operating System:     Windows 10
- (Windows only) ESP Command Prompt
- Using an IDE?: VSCode
- Power Supply:         USB

## Problem Description

Using an AP SSID where ssid[0] is 0xFF and ssid[1] is 0xFF I do not obtain the default SSID generated by the ESP32.
Here my code

```cpp
esp_err_t
wifi_ap_init (const uint8_t ap_ssid[SSID_LENGTH],
              const uint8_t ap_password[PSWD_LENGTH])
{
    int valid_credential = 0;
    esp_err_t ret = ESP_OK; 

    esp_netif_ip_info_t ipInfo;

	ESP_ERROR_CHECK(esp_netif_dhcps_stop(g_p_wifi_ap));
    IP4_ADDR(&ipInfo.ip, 192, 168, 1, 1);
    IP4_ADDR(&ipInfo.gw, 192, 168, 1, 1);
	IP4_ADDR(&ipInfo.netmask, 255, 255, 255, 0);

	ESP_ERROR_CHECK(esp_netif_set_ip_info(g_p_wifi_ap, &ipInfo));

	ESP_ERROR_CHECK(esp_netif_dhcps_start(g_p_wifi_ap));
    ESP_LOGI(g_p_tag, "AP - IP: " IPSTR, IP2STR(&ipInfo.ip));
    ESP_LOGI(g_p_tag, "AP - GW: " IPSTR, IP2STR(&ipInfo.gw));
    ESP_LOGI(g_p_tag, "AP - Mask: " IPSTR, IP2STR(&ipInfo.netmask));

    g_wifi_event_group = xEventGroupCreate();

    ESP_ERROR_CHECK(esp_wifi_stop());

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_NULL));

    wifi_config_t wifi_access_point_config = {
        .ap = {
            .max_connection = MAX_STA_CONNECTED,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        },
    };

    valid_credential = wifi_check_ssid_validity((char *) ap_ssid);
    if (0 == valid_credential)
    {
        ESP_LOGI(g_p_tag, "SSID AP OK");
        (void) strcpy((char *) wifi_access_point_config.ap.ssid,
                      (char *) ap_ssid);
    }
    else
    {
        ESP_LOGW(g_p_tag, "SSID AP non ok");
        wifi_access_point_config.ap.ssid[0] = 0XFF;
        wifi_access_point_config.ap.ssid[1] = 0XFF;
    }
    ESP_LOGI(g_p_tag, "AP SSID = %s", wifi_access_point_config.ap.ssid);

    valid_credential = wifi_check_password_validity((char *) ap_password);
    if (0 == valid_credential)
    {
        ESP_LOGI(g_p_tag, "PASSWORD AP OK");
        (void) strcpy((char *) wifi_access_point_config.ap.password,
                      (char *) ap_password);
    }
    else if (1 == valid_credential)
    {
        ESP_LOGW(g_p_tag, "PASSWORD AP non inserita");

        wifi_access_point_config.ap.authmode = WIFI_AUTH_OPEN;
    }
    else
    {
        ESP_LOGW(g_p_tag, "PASSWORD AP non ok");
        (void) strcpy((char *) wifi_access_point_config.ap.password,
                      "defaultvalue");
    }
    ESP_LOGI(g_p_tag, "AP PSWD = %s", wifi_access_point_config.ap.password);
    
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));

    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_access_point_config));

    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI(g_p_tag, "Created ap with SSID:%s password:%s",
             wifi_access_point_config.ap.ssid,
             wifi_access_point_config.ap.password);

    return ret;
} 
```

It's not important how I check if the SSID is valid, because after I see a not valid SSID it set the first two bytes.

**Here the problem:** I saw that, instead of creating the default ESP32's SSID, it used the previous (valid) stored SSID.
Below I attach what I see on terminal.
You can see the two question marks that are the 0xFF.
On my PC I look at the APs detected and I see ABCDEFGH that correspond to the previous (saved in NVS?)  SSID. The password corresponds to what I choose.

### Expected Behavior

From the official description of the Wi-Fi driver [here](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-guides/wifi.html#ap-basic-configuration), I read:
> SSID of AP; if the ssid[0] is 0xFF and ssid[1] is 0xFF, the AP defaults the SSID to ESP_aabbcc, where “aabbcc” is the last three bytes of the AP MAC.

## Debug Logs

```
W (1483) wifi_manage: SSID AP non ok
I (1493) wifi_manage: AP SSID = ��
W (1493) wifi_manage: PASSWORD AP non ok
I (1503) wifi_manage: AP PSWD = 1234ABCD
....
I (5603) wifi_manage: Created ap with SSID:�� password:1234ABCD
```

esphelp, 2022-04-07:
I was just having this same problem. I think the problem is the compiler optimizing the wifi_config_t struct with null values or some mysterious stack related issue. I used: wifi_config_t wifi_config = {0}; and then filled in each parameter manually like this: 
wifi_config.ap.channel = EXAMPLE_ESP_WIFI_CHANNEL; 
 wifi_config.ap.ssid_len = asprintf(&SSID, "esp32c3%x%x%x", mac[3], mac[4], mac[5]); 
memcpy(wifi_config.ap.ssid, SSID,wifi_config.ap.ssid_len);
and so on.
Make sure to flash the NVS before changing the SSID or it won't broadcast properly. 
ESP_ERROR_CHECK(nvs_flash_erase());
If you will be changing the SSID often and need to keep the rest of the NVS intact find the storage tag that the WIFI API uses and erase it like that instead.

filgra96, 2022-04-08:
> I was just having this same problem. I think the problem is the compiler optimizing the wifi_config_t struct with null values or some mysterious stack related issue. I used: wifi_config_t [...]

So you force an SSID in a manner it behaves like the default one. But I think, seen what's happening, that this is a bug!

> ESP_ERROR_CHECK(nvs_flash_erase()); If you will be changing the SSID often and need to keep the rest of the NVS intact find the storage tag that the WIFI API uses and erase it like that instead.

I can't erase the entire NVS because I use it to store other variables. Yes, I could delete only the Wi-Fi reserved area of NVS. However I'm pretty sure there is a bug..

chegewara, 2022-04-08:
How about:
```
wifi_access_point_config.ap.ssid[0] = 0X0;
```

filgra96, 2022-04-08:
> wifi_access_point_config.ap.ssid[0] = 0X0;

I can confirm that, by doing in this way, it sets the following SSID! `ESP_E45453`

So it doesn't follow the specs in the documentation, there is a different behaviour. **However now it's fine**, thanks!

filgra96, 2022-04-08:
> you have to get config: `esp_wifi_get_config`

It works, here the log:
`I (6234) wifi_manage: Created ap with SSID:ESP_E45453 password:1234ABCD`

filgra96, 2022-04-08:
I'm encountering another bug related to the previous problem: ESP32 creates the SSID expected BUT I'm not able to connect to it anymore.
I tried to switch back to a known SSID and I can connect to it with the provided password `1234ABCD`.
Connecting to `ESP_E45453` my PC tries some connections but nothing to do.. Turning off the ESP I see an error message on my PC such as "Not valid credentials". But I didn't change them.

nishanth-radja, 2022-04-11:
@filgra96 can you try to forget the network on your PC and try connecting again.Pls provide the console logs of the DUT when your are trying to connect

EN20M, 2022-05-03:
I'm having similar connection issues. With connections beeing unstable or not even possible to connet. Another symtom I see is that the SSID is not picked up by Windows and its showen as if it was hidden, while on android everything is semi fine (still having problems with stabilitiy...).

Alvin1Zhang, 2022-07-11:
Thanks for reporting, fix is available https://github.com/espressif/esp-idf/commit/88f79ca9caed2b3f2fc673995c1269ef6d1fd744, feel free to reopen. Thanks.

closed 2022-07-11 by Alvin1Zhang

diplfranzhoepfinger, 2023-07-01:
@Alvin1Zhang 

should this code work ? 

or should we do the Story with the 2 times FF like in the Documentation ? 

```
    wifi_config_t wifi_config = {
        .ap = {
            .ssid = "",
            .ssid_len = 0,
```

diplfranzhoepfinger, 2023-07-01:
OK, i use ESP-IDF 5.1, and there is still a Bug: 

if i use it with 

```
 .ssid = "",
  .ssid_len = 0,
```

i generates the Default SSID, but i still cannot connect, like https://github.com/espressif/esp-idf/issues/8744#issuecomment-1092555592

if i make a 

```
    wifi_config_t wifi_config = {
        .ap = {
            .ssid = "XX",
            .ssid_len = 2,
            .channel = EXAMPLE_ESP_WIFI_CHANNEL,
            .password = EXAMPLE_ESP_WIFI_PASS,
            .max_connection = EXAMPLE_MAX_STA_CONN,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK,
            .pmf_cfg = {
                    .required = false,
            },
        },
    };

    wifi_config.ap.ssid[0] = 0XFF;
    wifi_config.ap.ssid[1] = 0XFF;
```

it is fine.

Aditi-Lonkar, 2023-07-03:
Hi @diplfranzhoepfinger, 
Setting the ssid to blank and ssid_len to 0 will also set the ssid to default. We tried reproducing the issue at our end, but we are unable to do so. 

Can you please send the example code you are using along with debug logs showing default ssid is generated? please also share sniffer capture.

Here is how to enable debug logs from menuconfig.
In "idf.py menuconfig" goto Component config->Log output and select default log verbosity to debug/verbose.

To capture sniffer, you can go through [this](https://github.com/espressif/esp-idf/issues/8687#issuecomment-1105070843).

Thanks.

diplfranzhoepfinger, 2023-07-03:
i will do so, as soon as i can. 

Thanks for your Answer.