espressif/esp-idf #8746: mere existence of ESP_LOG_BUFFER_HEX_LEVEL() line causes 10x wifi slowdown

moefear85, 2022-04-07:
## Environment

- Development Kit:  AiThinker  ESP-12K
- Module or chip used:  esp32s2
- IDF version (run ``git describe --tags`` to find it): release/v4.4
- Build System:   idf.py
- Operating System:   Linux
- Using an IDE?: VSCode IDF Plugin
- Power Supply:  external 3.3V

## Problem Description

I have a send_packet() function that sends wifi packets using esp_wifi_80211_tx(). The function does many things, such as compute crc32, fragment large packets, does some logging, properly configures the mac header, and so on.
One line present is:

>ESP_LOG_BUFFER_HEX_LEVEL(tag_send_packet,packet->array,PACKET_HEADER_SIZE+payload_size,ESP_LOG_VERBOSE);

But that line never actually gets triggered, since I globally set the log level to ESP_LOG_WARN. Despite this, when running the code on an esp32, I get around ~900KB/s. I'm happy with that. With some tweaking I can get up to 5.8MB/s (jaw dropping). But surprisingly, when I compile for an esp32s2, with all sdkconfig settings remaining as equivalent as possible, the speed drops to 157.83 KB/s in a test setup, causing it to drop even further (~20KB/s) in a field-app, which is unbearably slow. After alot of testing, I localized the issue to that log line, simply its presence. If I comment it out, the speed jumps to 900KB/s, just like for the esp32. By uncommenting it, even though it never generates any output, the speed tanks as mentioned. Is this a bug or is there some caveat I'm not aware of?

My initial assumption was perhaps it's a caching issue. But if I place IRAM_ATTR before the definition of send_packet(), that actually makes the speed drop to about 80 KB/s. If I keep the IRAM_ATTR but again comment out ESP_LOG_BUFFER_HEX_LEVEL(), then the speed rises again to 900KB/s and is thus unaffected by the pack/presence of IRAM_ATTR.

negativekelvin, 2022-04-07:
Maybe there is some LOG_LOCAL_LEVEL define in one of the esp32s2 header files that is overriding your global log level

MaxwellAlan, 2022-04-08:
Hi @moefear85 
Do you mean just adding below line will cause lower performace in ESP32S2, right?
> > ESP_LOG_BUFFER_HEX_LEVEL(tag_send_packet,packet->array,PACKET_HEADER_SIZE+payload_size,ESP_LOG_VERBOSE);
Seems like cache issue, can try different configration of ICache config:

- Instruction cache size : 8KB or 16KB;
- instruction cache line size: 16Bytes or 32 Bytes
And do you enable PSRAM?
If yes, you can try with different configration of DCache config:

- Data Cache size: 8KB or 16KB;
- Data Cache line size: 16Bytest or 32Bytes.
And you can refer to s2 iperf config: https://github.com/espressif/esp-idf/blob/master/examples/wifi/iperf/sdkconfig.defaults.esp32s2
Thanks.

moefear85, 2022-04-08:
I already have the cache settings at their maximums. No PSRAM enabled. I already tried those s3 iperf config settings.

But yeah, I guess it's a cache issue. I just commented out all occurances of that function. Performance in a test scenario is perfect. In a real-life scenario, such as SLIP over wireless, I was initially getting 200kbit/s. After tweaking everything I can think of, I bumped this up to 470kbit/s. That's good enough for me, that's half of what I get over a direct serial line. Any further improvement will require deep analysis how parts of the code are moved in/out of cache and which parts should be kept where. But now I have other priorities.

moefear85, 2022-04-08:
btw as far as I understand, PSRAM runs at max 80MHz, ie the max that flash can achieve. So I see PSRAM effectively as slow as flash accesses on cache misses. That's why I generally don't use it unless I really need the space.

closed 2022-04-08 by moefear85

moefear85, 2022-04-08:
I started caching the results of some relatively intensive calculations on each packet send. I managed to bump up wireless SLIP speed to 700Kbit/s. That's almost on par with serial, and somewhat faster than what I would get over traditional wifi (station+lwIP).

MaxwellAlan, 2022-04-11:
hi @moefear85 
>  CPU is always at 240MHz. I assume CPU accesses to memory run on the order of ~240MHz?

Yes.

> Also worth noting, even in the bulk/unacknowledged case, esp_wifi_80211_tx returns alot of ESP_ERR_NO_MEM values. So for accurate results, I create an inner loop around it, that endlessly retries until ESP_OK is returned, and that counts as a single outer loop.

This depends on the wireless air traffic. And you can try with esp_wifi_set_ps(0) to disable Modem sleep to improve the opportunity of 80211_tx to send. Also you can use esp_err_t esp_wifi_config_80211_tx_rate(wifi_interface_t ifx, wifi_phy_rate_t rate); to config 80211_tx pkt rate to improve performace, but this feature still only in master branch, will be backported to release/v4.4 & v4.3 soon.

moefear85, 2022-04-11:
> This depends on the wireless air traffic. And you can try with esp_wifi_set_ps(0) to disable Modem sleep to improve the opportunity of 80211_tx to send. Also you can use esp_err_t [...]

but i'm already calling those functions. do you mean they are currently just place-holders that don't actually do anything?

also, is i use default sdkconfig, but disable power saving, does this mean the modem is still able to sleep between transfers & does it sporadically do so? if yes, for how many micro/milliseconds on average? does it sleep until the next transfer? i'm sure the channel i'm on is practically empty

MaxwellAlan, 2022-04-20:
Hi @moefear85 

> but i'm already calling those functions. do you mean they are currently just place-holders that don't actually do anything?

No, it can works, bug in release/v4.4, it can only works in 20MHz bandwidth, if you can capture wireless pkts, you can upload some pkts for me to analysis.

> also, if i use default sdkconfig, but disable power saving, does this mean the modem is still able to sleep between transfers & does it sporadically do so?

No, if you set esp_wifi_set_ps(0) ,  then esp32 modem will never turn off to sleep.