espressif/esp-idf #8747: Virtual address not enough for PSRAM!

TGiles1998, 2022-04-07:
Development Kit: ESP32-S3
Kit version (DevKitC):
Module or chip used: ESP32-S3
IDF version: V5.0
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
Operating System: macOS
Using an IDE?: No just Terminal
Power Supply: USB

Actual Behavior

SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0110,len:0x18d0
load:0x403b6000,len:0xb70
load:0x403b9000,len:0x31e4
entry 0x403b6240
I (37) boot: ESP-IDF v5.0-dev-2349-g4350e6fef8 2nd stage bootloader
I (38) boot: compile time 19:01:32
I (38) boot: chip revision: 0
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : SLOW READ
I (51) boot.esp32s3: SPI Flash Size : 32MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=07d20h ( 32032) map
I (115) esp_image: segment 1: paddr=00017d48 vaddr=3fc91220 size=02dd0h ( 11728) load
I (119) esp_image: segment 2: paddr=0001ab20 vaddr=40374000 size=054f8h ( 21752) load
I (130) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=17130h ( 94512) map
I (155) esp_image: segment 4: paddr=00037158 vaddr=403794f8 size=07d20h ( 32032) load
I (164) esp_image: segment 5: paddr=0003ee80 vaddr=50000000 size=00010h (    16) load
I (170) boot: Loaded app from partition at offset 0x10000
I (170) boot: Disabling RNG early entropy source...
I (184) opi psram: vendor id    : 0x0d (AP)
I (185) opi psram: dev id       : 0x03 (generation 4)
I (185) opi psram: density      : 0x07 (256 Mbit)
I (189) opi psram: good-die     : 0x01 (Pass)
I (194) opi psram: Latency      : 0x01 (Fixed)
I (199) opi psram: VCC          : 0x00 (1.8V)
I (204) opi psram: SRF          : 0x01 (Fast Refresh)
I (210) opi psram: BurstType    : 0x01 (Hybrid Wrap)
I (216) opi psram: BurstLen     : 0x01 (32 Byte)
I (221) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (227) opi psram: DriveStrength: 0x00 (1/1)
I (232) spiram: Found 32MB SPI RAM device
I (237) spiram: Speed: 40MHz
I (240) spiram: Initialized, cache is in normal (1-core) mode.
E (247) spiram: Virtual address not enough for PSRAM!

How can I get 32MB of external RAM to work? The docs say up to 1GB is possible.

Many thanks,

Thomas

Icarus113, 2022-04-08:
@TGiles1998 
Hi, on ESP32S3, 
- The maximum virtual size for external memory (both Flash and PSRAM) is 32MB. Flash and PSRAM share these virtual addresses.
- On current Master, only 16MB is used for PSRAM.
- 1GB is the maximum physical address that can be mapped to the MMU translation unit.

One solution is:
The firmware can map as much. The rest of the PSRAM physical region will not be used. For example, 16MB of your PSRAM is mapped, and the rest 16MB can't be accessed.

FYI, there will be a commit improving the mapping scheme. Flash rodata will be mapped to the vaddr region first, then the rest will be used by the PSRAM. An easy example, Flash rodata takes 1MB, then 31MB can be used by the PSRAM.

Another way is to have a dynamic mapping scheme, but this is not in our plan yet.

TGiles1998, 2022-04-08:
@Icarus113 Thank you for your reply! I look forward to the commit!

TGiles1998, 2022-04-08:
@Icarus113 By any chance is there a rough date as to when this update might come out? Many thanks, Thomas

TGiles1998, 2022-04-18:
Hi are there any updates on this? Thanks, Thomas

TGiles1998, 2022-04-25:
@Icarus113 Just following up on the status of the 32MB mapping, is there anything I can do to help also?
Thanks in advance,
Thomas

Icarus113, 2022-04-25:
Hi @TGiles1998 ,
> there will be a commit improving the mapping scheme

This commit is merged. 

We don't support PSRAM larger than 8MB yet. So you may need to update the code a bit. Here are the code that need to be updated:

For [this block](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/port/esp32s3/spiram.c#L103), would you help do this:
```
if (vaddr_start + psram_available_size > DRAM0_CACHE_ADDRESS_HIGH) {
  psram_available_size = DRAM0_CACHE_ADDRESS_HIGH - vaddr_start - 1;
}
```
Also change [this line](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/port/esp32s3/spiram.c#L114) to:
```
cache_bus_mask_t bus_mask = cache_ll_l1_get_bus(0, vaddr_start, actual_mapped_len - 1);
```
These are the modifications that I can think of, but we don't have such device so I can't tell if this can 100% work...
Another note is, after above changes, you can't find an extra piece of vaddr region that is available for Flash.

TGiles1998, 2022-04-25:
Many thanks fro your reply, I made the changes but unfortunately it didnt work.

Rebooting...
??ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403758d4
0x403758d4: esp_restart_noos at /Users/thomasgiles/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0110,len:0x18d8
load:0x403b6000,len:0xb70
load:0x403b9000,len:0x31f4
entry 0x403b6240
I (37) boot: ESP-IDF v5.0-dev-2563-g45c1d1cba2-dirty 2nd stage bootloader
I (38) boot: compile time 21:12:53
I (38) boot: chip revision: 0
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : SLOW READ
I (51) boot.esp32s3: SPI Flash Size : 32MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=07da0h ( 32160) map
I (115) esp_image: segment 1: paddr=00017dc8 vaddr=3fc913d0 size=02f20h ( 12064) load
I (119) esp_image: segment 2: paddr=0001acf0 vaddr=40374000 size=05328h ( 21288) load
I (130) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=172a8h ( 94888) map
I (156) esp_image: segment 4: paddr=000372d0 vaddr=40379328 size=080a0h ( 32928) load
I (165) esp_image: segment 5: paddr=0003f378 vaddr=50000000 size=00010h (    16) load
I (171) boot: Loaded app from partition at offset 0x10000
I (171) boot: Disabling RNG early entropy source...
I (185) opi psram: vendor id    : 0x0d (AP)
I (185) opi psram: dev id       : 0x03 (generation 4)
I (186) opi psram: density      : 0x07 (256 Mbit)
I (190) opi psram: good-die     : 0x01 (Pass)
I (195) opi psram: Latency      : 0x01 (Fixed)
I (200) opi psram: VCC          : 0x00 (1.8V)
I (205) opi psram: SRF          : 0x01 (Fast Refresh)
I (211) opi psram: BurstType    : 0x01 (Hybrid Wrap)
I (217) opi psram: BurstLen     : 0x01 (32 Byte)
I (222) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (228) opi psram: DriveStrength: 0x00 (1/1)
I (233) spiram: Found 32MB SPI RAM device
I (237) spiram: Speed: 40MHz

assert failed: cache_ll_l1_get_bus /IDF/components/hal/esp32s3/include/hal/cache_ll.h:50 (0)

Backtrace:0x40375c52:0x3fceb1700x40379885:0x3fceb190 0x4037f591:0x3fceb1b0 0x420057ed:0x3fceb2d0 0x403753ac:0x3fceb300 0x403ba265:0x3fceb330 0x403ba5bd:0x3fceb380 0x403b62a9:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED
0x40375c52: panic_abort at /Users/thomasgiles/esp/esp-idf/components/esp_system/panic.c:409

0x40379885: esp_system_abort at /Users/thomasgiles/esp/esp-idf/components/esp_system/esp_system.c:135

0x4037f591: __assert_func at /Users/thomasgiles/esp/esp-idf/components/newlib/assert.c:85

0x420057ed: cache_ll_l1_get_bus at /Users/thomasgiles/esp/esp-idf/components/hal/esp32s3/include/hal/cache_ll.h:50
 (inlined by) esp_spiram_init at /Users/thomasgiles/esp/esp-idf/components/esp_hw_support/port/esp32s3/spiram.c:122

0x403753ac: call_start_cpu0 at /Users/thomasgiles/esp/esp-idf/components/esp_system/port/cpu_start.c:391

I followed th error to line 50 of cache_ll

and it revealed:

[image]

I'll continue to try to debug this. Any other suggestions that you have please let me know!

many thanks,

Thomas

TGiles1998, 2022-04-25:
[image]

Will alter the sizes of the memory and see what happens.

Icarus113, 2022-04-26:
@TGiles1998 
In `cache_ll_l1_get_bus()`, what is `vaddr_end`? I suspect it exceeds `IRAM0_CACHE_ADDRESS_HIGH`.

TGiles1998, 2022-04-26:
@Icarus113 Hi, Im not sure how to obtain the value for vaddr_end. Should I increase DRAM0_CACHE_ADDRESS_HIGH? Yes sorry I am using VS Code to track the variables and it displayed the wrong script. I have now found the ESP32S3 instead which is the hardware I am using
[image]
.

TGiles1998, 2022-04-26:
Currently I am compiling the Hello World example from the get-started folder and using the 'flash monitor' command to see if it has been successful.

TGiles1998, 2022-04-28:
Hi @Icarus113, do you know where the limitation of 16MB for the PSRAM is in the code? Many thanks,

Thomas

TGiles1998, 2022-05-07:
Hi @Icarus113 ,

I've tried to use a smaller flash as I though this could be the issue with 32/32 flash/psram maxing out and causing errors. Running a 4/32 setup i got the error:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403ba57a
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0110,len:0x18d0
load:0x403b6000,len:0xb70
load:0x403b9000,len:0x31d0
entry 0x403b6240
I (37) boot: ESP-IDF v5.0-dev-2586-ga82e6e63d9 2nd stage bootloader
I (38) boot: compile time 21:36:00
I (38) boot: chip revision: 0
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : SLOW READ
I (51) boot.esp32s3: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
E (61) flash_parts: Incorrect MD5 checksum
E (66) boot: Failed to verify partition table
E (71) boot: load partition table error!

If you have any suggestions that would be great thanks!

Thomas

TGiles1998, 2022-05-14:
@Icarus113 I have changed the flash from 32MB octal to 8MB quad and I am still getting the same errors. Does the new patch allow for the dynamic allocation so 8MB of flash will leave 24MB of PSRAM in the virtual address? Below is the error:

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375894
0x40375894: esp_restart_noos at /Users/thomasgiles/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0110,len:0x18d8
load:0x403b6000,len:0xb70
load:0x403b9000,len:0x31d0
entry 0x403b6240
I (29) boot: ESP-IDF v5.0-dev-2586-ga82e6e63d9-dirty 2nd stage bootloader
I (29) boot: compile time 14:56:54
I (29) boot: chip revision: 0
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 8MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=07d50h ( 32080) map
I (104) esp_image: segment 1: paddr=00017d78 vaddr=3fc91390 size=02efch ( 12028) load
I (109) esp_image: segment 2: paddr=0001ac7c vaddr=40374000 size=0539ch ( 21404) load
I (120) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=172d4h ( 94932) map
I (141) esp_image: segment 4: paddr=000372fc vaddr=4037939c size=07ff4h ( 32756) load
I (148) esp_image: segment 5: paddr=0003f2f8 vaddr=50000000 size=00010h (    16) load
I (154) boot: Loaded app from partition at offset 0x10000
I (154) boot: Disabling RNG early entropy source...
I (170) opi psram: vendor id    : 0x0d (AP)
I (170) opi psram: dev id       : 0x03 (generation 4)
I (170) opi psram: density      : 0x07 (256 Mbit)
I (174) opi psram: good-die     : 0x01 (Pass)
I (179) opi psram: Latency      : 0x01 (Fixed)
I (184) opi psram: VCC          : 0x00 (1.8V)
I (190) opi psram: SRF          : 0x01 (Fast Refresh)
I (195) opi psram: BurstType    : 0x01 (Hybrid Wrap)
I (201) opi psram: BurstLen     : 0x01 (32 Byte)
I (206) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (212) opi psram: DriveStrength: 0x00 (1/1)
I (217) spiram: Found 32MB SPI RAM device
I (222) spiram: Speed: 40MHz

assert failed: cache_ll_l1_get_bus /IDF/components/hal/esp32s3/include/hal/cache_ll.h:50 (0)

Backtrace:0x40375c12:0x3fceb1700x4037982d:0x3fceb190 0x4037f539:0x3fceb1b0 0x420057f1:0x3fceb2d0 0x4037536c:0x3fceb300 0x403ba241:0x3fceb330 0x403ba599:0x3fceb380 0x403b62a9:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED
0x40375c12: panic_abort at /Users/thomasgiles/esp/esp-idf/components/esp_system/panic.c:409

0x4037982d: esp_system_abort at /Users/thomasgiles/esp/esp-idf/components/esp_system/esp_system.c:135

0x4037f539: __assert_func at /Users/thomasgiles/esp/esp-idf/components/newlib/assert.c:85

0x420057f1: cache_ll_l1_get_bus at /Users/thomasgiles/esp/esp-idf/components/hal/esp32s3/include/hal/cache_ll.h:50
 (inlined by) esp_spiram_init at /Users/thomasgiles/esp/esp-idf/components/esp_hw_support/port/esp32s3/spiram.c:115

0x4037536c: call_start_cpu0 at /Users/thomasgiles/esp/esp-idf/components/esp_system/port/cpu_start.c:391

ELF file SHA256: 6baf2c38e07f435a

Can you suggest any other methods to try and sort this to increase the PSRAM size?

Many thanks,

Thomas

TGiles1998, 2022-05-14:
@Icarus113 For the ESP32-S3-WROOM-2 with 32MB of flash and 8 MB of Flash, how does the dynamic mapping work for this?

TGiles1998, 2022-05-14:
@Icarus113 Would it be possible to implement some sort of if statement that if 32MB PSRAM is found, the flash size is deducted from its mappable length and the remaining can only be accessed?

TGiles1998, 2022-06-05:
@Icarus113 Just following up on the status of 32MB of PSRAM? Thanks, Thomas

TGiles1998, 2022-06-30:
@Icarus113 @igrr  Is the 32MB PSRAM mapping available now?

Icarus113, 2022-07-01:
@TGiles1998  Hi, sorry for the late reply. Although we don't support 32MB PSRAM, I think I can do some similar experiments to achieve this:
> One solution is:
The firmware can map as much. The rest of the PSRAM physical region will not be used. For example, 16MB of your PSRAM is mapped, and the rest 16MB can't be accessed.

I'll deal with this in these days, if nothing interrupts..

TGiles1998, 2022-07-11:
Hi @Icarus113 just wondering if there has been an update?

Many thanks,

Thomas

Icarus113, 2022-07-18:
Hi @TGiles1998 , would you help use current IDF github master branch to test it with 32MB size PSRAM? 
    
As we don't sell modules with 32MB psram now, I don't have a 32MB psram board.    
I tried to limit the vaddr range to 2MB, then map it to a 8MB PSRAM. And this works fine on current master branch. The way is to map as vaddr range allows. So in my situation, it'll start mapping from 0x3c030000 to 0x3c230000. In your situation, it should map from a vaddr (depends on your rodata size), to 0x3E000000.

TGiles1998, 2022-07-18:
@Icarus113 Thank you very very much! I will test this and let you know how I progress.

TGiles1998, 2022-07-26:
Hi, I've just tried it and it produced this:

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375abc
0x40375abc: esp_restart_noos at /Users/thomasgiles/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:152 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:4
load:0x3fce3810,len:0x1648
load:0x403c9700,len:0xbd0
load:0x403cc700,len:0x2d68
entry 0x403c9900
I (43) boot: ESP-IDF v5.0-dev-4379-g36f49f361c 2nd stage bootloader
I (43) boot: compile time 21:11:20
I (43) boot: chip revision: V001
I (46) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (53) boot.esp32s3: Boot SPI Speed : 20MHz
I (58) boot.esp32s3: SPI Mode       : DIO
I (63) boot.esp32s3: SPI Flash Size : 2MB
I (68) boot: Enabling RNG early entropy source...
I (73) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (91) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (99) boot:  2 factory          factory app      00 00 00010000 00100000
I (106) boot: End of partition table
I (110) boot_comm: chip revision: 1, min. application chip revision: 0
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08ae8h ( 35560) map
I (140) esp_image: segment 1: paddr=00018b10 vaddr=3fc91900 size=02ef0h ( 12016) load
I (145) esp_image: segment 2: paddr=0001ba08 vaddr=40374000 size=04610h ( 17936) load
I (154) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=19fech (106476) map
I (196) esp_image: segment 4: paddr=0003a014 vaddr=40378610 size=092d4h ( 37588) load
I (212) esp_image: segment 5: paddr=000432f0 vaddr=50000000 size=00010h (    16) load
I (218) boot: Loaded app from partition at offset 0x10000
I (218) boot: Disabling RNG early entropy source...
I (232) octal_psram: vendor id    : 0x0d (AP)
I (233) octal_psram: dev id       : 0x03 (generation 4)
I (233) octal_psram: density      : 0x07 (256 Mbit)
I (238) octal_psram: good-die     : 0x01 (Pass)
I (243) octal_psram: Latency      : 0x01 (Fixed)
I (248) octal_psram: VCC          : 0x00 (1.8V)
I (253) octal_psram: SRF          : 0x00 (Slow Refresh)
I (259) octal_psram: BurstType    : 0x01 ( Wrap)
I (265) octal_psram: BurstLen     : 0x03 (1024 Byte)
I (270) octal_psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (276) octal_psram: DriveStrength: 0x03 (1/8)
I (282) esp_psram: Found 32MB SPI RAM device
I (286) esp_psram: Speed: 40MHz

assert failed: mmu_hal_map_region /IDF/components/hal/mmu_hal.c:87 (mmu_ll_check_valid_ext_vaddr_region(mmu_id, vaddr, len))

Backtrace: 0x40375e4e:0x3fceb110 0x4037a299:0x3fceb130 0x4037fa15:0x3fceb150 0x4037cfb6:0x3fceb270 0x42007d45:0x3fceb2a0 0x403753e4:0x3fceb2d0 0x403cd762:0x3fceb330 0x403cdb6d:0x3fceb380 0x403c9969:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED
0x40375e4e: panic_abort at /Users/thomasgiles/esp/esp-idf/components/esp_system/panic.c:409

0x4037a299: esp_system_abort at /Users/thomasgiles/esp/esp-idf/components/esp_system/esp_system.c:135

0x4037fa15: __assert_func at /Users/thomasgiles/esp/esp-idf/components/newlib/assert.c:78

0x4037cfb6: mmu_hal_map_region at /Users/thomasgiles/esp/esp-idf/components/hal/mmu_hal.c:87 (discriminator 1)

0x42007d45: esp_psram_init at /Users/thomasgiles/esp/esp-idf/components/esp_psram/esp_psram.c:170

0x403753e4: call_start_cpu0 at /Users/thomasgiles/esp/esp-idf/components/esp_system/port/cpu_start.c:397

ELF file SHA256: 457f1d804004c1da

TGiles1998, 2022-07-26:
@Icarus113  How can I ensure that I have downloaded the most recent repo and what menuconfig parameters should I include. Many, thanks, Thomas

Icarus113, 2022-07-27:
@TGiles1998 Hi, this makes me think of a corner case. As you use a 32MB one, which reaches the end of the external DRAM vaddr range. Could you help just update your local repo to current IDF github head, [this commit](36f49f361c001b49c538364056bc5d2d04c6f321). This should include the recent psram code change. Then try it again? Just make sure we are using same version to test.

TGiles1998, 2022-07-27:
Hi, I made sure my idf was at this commit: 36f49f361c001b49c538364056bc5d2d04c6f321

[image]

Then this was the result:

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375abc
0x40375abc: esp_restart_noos at /Users/thomasgiles/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:152 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:4
load:0x3fce3810,len:0x1648
load:0x403c9700,len:0xbd0
load:0x403cc700,len:0x2d68
entry 0x403c9900
I (43) boot: ESP-IDF v5.0-dev-4379-g36f49f361c 2nd stage bootloader
I (43) boot: compile time 10:24:33
I (43) boot: chip revision: V001
I (46) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (53) boot.esp32s3: Boot SPI Speed : 20MHz
I (58) boot.esp32s3: SPI Mode       : DIO
I (63) boot.esp32s3: SPI Flash Size : 2MB
I (68) boot: Enabling RNG early entropy source...
I (73) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (84) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (91) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (99) boot:  2 factory          factory app      00 00 00010000 00100000
I (106) boot: End of partition table
I (110) boot_comm: chip revision: 1, min. application chip revision: 0
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08ae8h ( 35560) map
I (140) esp_image: segment 1: paddr=00018b10 vaddr=3fc91900 size=02ef0h ( 12016) load
I (145) esp_image: segment 2: paddr=0001ba08 vaddr=40374000 size=04610h ( 17936) load
I (154) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=19fech (106476) map
I (196) esp_image: segment 4: paddr=0003a014 vaddr=40378610 size=092d4h ( 37588) load
I (212) esp_image: segment 5: paddr=000432f0 vaddr=50000000 size=00010h (    16) load
I (218) boot: Loaded app from partition at offset 0x10000
I (218) boot: Disabling RNG early entropy source...
I (232) octal_psram: vendor id    : 0x0d (AP)
I (233) octal_psram: dev id       : 0x03 (generation 4)
I (233) octal_psram: density      : 0x07 (256 Mbit)
I (238) octal_psram: good-die     : 0x01 (Pass)
I (243) octal_psram: Latency      : 0x01 (Fixed)
I (248) octal_psram: VCC          : 0x00 (1.8V)
I (253) octal_psram: SRF          : 0x00 (Slow Refresh)
I (259) octal_psram: BurstType    : 0x01 ( Wrap)
I (265) octal_psram: BurstLen     : 0x03 (1024 Byte)
I (270) octal_psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (276) octal_psram: DriveStrength: 0x03 (1/8)
I (282) esp_psram: Found 32MB SPI RAM device
I (286) esp_psram: Speed: 40MHz

assert failed: mmu_hal_map_region /IDF/components/hal/mmu_hal.c:87 (mmu_ll_check_valid_ext_vaddr_region(mmu_id, vaddr, len))

Backtrace: 0x40375e4e:0x3fceb110 0x4037a299:0x3fceb130 0x4037fa15:0x3fceb150 0x4037cfb6:0x3fceb270 0x42007d45:0x3fceb2a0 0x403753e4:0x3fceb2d0 0x403cd762:0x3fceb330 0x403cdb6d:0x3fceb380 0x403c9969:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED
0x40375e4e: panic_abort at /Users/thomasgiles/esp/esp-idf/components/esp_system/panic.c:409

0x4037a299: esp_system_abort at /Users/thomasgiles/esp/esp-idf/components/esp_system/esp_system.c:135

0x4037fa15: __assert_func at /Users/thomasgiles/esp/esp-idf/components/newlib/assert.c:78

0x4037cfb6: mmu_hal_map_region at /Users/thomasgiles/esp/esp-idf/components/hal/mmu_hal.c:87 (discriminator 1)

0x42007d45: esp_psram_init at /Users/thomasgiles/esp/esp-idf/components/esp_psram/esp_psram.c:170

0x403753e4: call_start_cpu0 at /Users/thomasgiles/esp/esp-idf/components/esp_system/port/cpu_start.c:397

ELF file SHA256: 8ea6fbb2143eb839

Let me know how i can help and thank you so much for continuing to help solve this problem!

Icarus113, 2022-07-27:
@TGiles1998  Hi, thanks for letting me know. I will prepare a commit to solve this case. Once that commit is synced here, I'll let you know. Then we can know what's going on next.

Icarus113, 2022-08-01:
@TGiles1998 Hi, would you help update your local repo, including this [commit](https://github.com/espressif/esp-idf/commit/f325ad22117964cff3c9476f422b339214248cfe). With this one, ending at the end of the vaddr is ok. And let's see what's gonna happen next.

TGiles1998, 2022-08-01:
@Icarus113 
[image]

This was the result:

I (0) cpu_start: ApESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:4
load:0x3fce3810,len:0x164c
load:0x403c9700,len:0xbd0
load:0x403cc700,len:0x2d78
entry 0x403c9900
I (41) boot: ESP-IDF v5.0-dev-4488-gf325ad2211 2nd stage bootloader
I (41) boot: compile time 19:08:42
I (41) boot: chip revision: V001
I (44) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (51) boot.esp32s3: Boot SPI Speed : 20MHz
I (56) boot.esp32s3: SPI Mode       : DIO
I (61) boot.esp32s3: SPI Flash Size : 2MB
I (66) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (75) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (89) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (97) boot:  2 factory          factory app      00 00 00010000 00100000
I (104) boot: End of partition table
I (109) boot_comm: chip revision: 1, min. application chip revision: 0
I (116) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08b08h ( 35592) map
I (138) esp_image: segment 1: paddr=00018b30 vaddr=3fc91700 size=02f00h ( 12032) load
I (144) esp_image: segment 2: paddr=0001ba38 vaddr=40374000 size=045e0h ( 17888) load
I (152) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=19df8h (105976) map
I (194) esp_image: segment 4: paddr=00039e20 vaddr=403785e0 size=09098h ( 37016) load
I (210) esp_image: segment 5: paddr=00042ec0 vaddr=50000000 size=00010h (    16) load
I (216) boot: Loaded app from partition at offset 0x10000
I (216) boot: Disabling RNG early entropy source...
I (230) octal_psram: vendor id    : 0x0d (AP)
I (230) octal_psram: dev id       : 0x03 (generation 4)
I (230) octal_psram: density      : 0x07 (256 Mbit)
I (235) octal_psram: good-die     : 0x01 (Pass)
I (241) octal_psram: Latency      : 0x01 (Fixed)
I (246) octal_psram: VCC          : 0x00 (1.8V)
I (251) octal_psram: SRF          : 0x00 (Slow Refresh)
I (257) octal_psram: BurstType    : 0x01 ( Wrap)
I (262) octal_psram: BurstLen     : 0x03 (1024 Byte)
I (268) octal_psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (274) octal_psram: DriveStrength: 0x03 (1/8)
I (279) esp_psram: Found 32MB SPI RAM device
I (284) esp_psram: Speed: 40MHz
I (288) cpu_start: Pro cpu up.
I (292) cpu_start: Starting app cpu, entry point is 0x4037527c
0x4037527c: call_start_cpu1 at /Users/thomasgiles/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: ApESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:4
load:0x3fce3810,len:0x164c
load:0x403c9700,len:0xbd0
load:0x403cc700,len:0x2d78
entry 0x403c9900
I (41) boot: ESP-IDF v5.0-dev-4488-gf325ad2211 2nd stage bootloader
I (41) boot: compile time 19:08:42
I (41) boot: chip revision: V001
I (44) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (51) boot.esp32s3: Boot SPI Speed : 20MHz
I (56) boot.esp32s3: SPI Mode       : DIO
I (61) boot.esp32s3: SPI Flash Size : 2MB
I (66) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (75) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (89) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (97) boot:  2 factory          factory app      00 00 00010000 00100000
I (104) boot: End of partition table
I (109) boot_comm: chip revision: 1, min. application chip revision: 0
I (116) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08b08h ( 35592) map
I (138) esp_image: segment 1: paddr=00018b30 vaddr=3fc91700 size=02f00h ( 12032) load
I (144) esp_image: segment 2: paddr=0001ba38 vaddr=40374000 size=045e0h ( 17888) load
I (152) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=19df8h (105976) map
I (194) esp_image: segment 4: paddr=00039e20 vaddr=403785e0 size=09098h ( 37016) load
I (210) esp_image: segment 5: paddr=00042ec0 vaddr=50000000 size=00010h (    16) load
I (216) boot: Loaded app from partition at offset 0x10000
I (216) boot: Disabling RNG early entropy source...
I (230) octal_psram: vendor id    : 0x0d (AP)
I (230) octal_psram: dev id       : 0x03 (generation 4)
I (230) octal_psram: density      : 0x07 (256 Mbit)
I (235) octal_psram: good-die     : 0x01 (Pass)
I (241) octal_psram: Latency      : 0x01 (Fixed)
I (246) octal_psram: VCC          : 0x00 (1.8V)
I (251) octal_psram: SRF          : 0x00 (Slow Refresh)
I (257) octal_psram: BurstType    : 0x01 ( Wrap)
I (262) octal_psram: BurstLen     : 0x03 (1024 Byte)
I (268) octal_psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (274) octal_psram: DriveStrength: 0x03 (1/8)
I (279) esp_psram: Found 32MB SPI RAM device
I (284) esp_psram: Speed: 40MHz
I (288) cpu_start: Pro cpu up.
I (292) cpu_start: Starting app cpu, entry point is 0x4037527c
0x4037527c: call_start_cpu1 at /Users/thomasgiles/esp/esp-idf/components/esp_system/port/cpu_start.c:154

It kept on pausing and repeating these: Didn't get any further.

Thanks,

Thomas

Icarus113, 2022-08-02:
@TGiles1998  Hi, it did step further actually. PSRAM is mapped to virtual memory already. It fails when starting the second core. 
    
Could you use single core mode (by enabling [this config](https://github.com/espressif/esp-idf/blob/master/components/freertos/Kconfig#L13)) first, to see if the [psram memory test](https://github.com/espressif/esp-idf/blob/master/components/esp_system/port/cpu_start.c#L439) (it'll run by default, if `CONFIG_SPIRAM_MEMTEST` is on) succeeds?
    
I'm guessing your PSRAM introduced some issues to its connected IOs. As the main Flash share the IOs with PSRAM, except for their CSs. So running flash code isn't stable anymore.

TGiles1998, 2022-08-02:
@Icarus113 Apologies I meant further in terms of giving no more messages after the second core start, I apologise how that came over on messages. This is the result from FREERTOS UNICORE:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3810,len:0x164c
load:0x403c9700,len:0xb7c
load:0x403cc700,len:0x2cf8
entry 0x403c98e4
I (29) boot: ESP-IDF v5.0-dev-4488-gf325ad2211 2nd stage bootloader
I (29) boot: compile time 18:58:45
I (29) boot: chip revision: V001
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0832ch ( 33580) map
I (121) esp_image: segment 1: paddr=00018354 vaddr=3fc90b00 size=024d8h (  9432) load
I (124) esp_image: segment 2: paddr=0001a834 vaddr=40374000 size=057e4h ( 22500) load
I (136) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=16118h ( 90392) map
I (160) esp_image: segment 4: paddr=00036140 vaddr=403797e4 size=072f8h ( 29432) load
I (169) esp_image: segment 5: paddr=0003d440 vaddr=50000000 size=00010h (    16) load
I (175) boot: Loaded app from partition at offset 0x10000
I (175) boot: Disabling RNG early entropy source...
I (189) octal_psram: vendor id    : 0x0d (AP)
I (189) octal_psram: dev id       : 0x03 (generation 4)
I (190) octal_psram: density      : 0x07 (256 Mbit)
I (195) octal_psram: good-die     : 0x01 (Pass)
I (200) octal_psram: Latency      : 0x01 (Fixed)
I (205) octal_psram: VCC          : 0x00 (1.8V)
I (210) octal_psram: SRF          : 0x00 (Slow Refresh)
I (216) octal_psram: BurstType    : 0x01 ( Wrap)
I (221) octal_psram: BurstLen     : 0x03 (1024 Byte)
I (227) octal_psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (233) octal_psram: DriveStrength: 0x03 (1/8)
I (238) esp_psram: Found 32MB SPI RAM device
I (243) esp_psram: Speed: 40MHz
I (247) cpu_start: Pro cpu up.
I (251) cpu_start: Single core mode
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3810,len:0x164c
load:0x403c9700,len:0xb7c
load:0x403cc700,len:0x2cf8
entry 0x403c98e4
I (29) boot: ESP-IDF v5.0-dev-4488-gf325ad2211 2nd stage bootloader
I (29) boot: compile time 18:58:45
I (29) boot: chip revision: V001
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32s3: Boot SPI Speed : 40MHz
I (45) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0832ch ( 33580) map
I (121) esp_image: segment 1: paddr=00018354 vaddr=3fc90b00 size=024d8h (  9432) load
I (124) esp_image: segment 2: paddr=0001a834 vaddr=40374000 size=057e4h ( 22500) load
I (136) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=16118h ( 90392) map
I (160) esp_image: segment 4: paddr=00036140 vaddr=403797e4 size=072f8h ( 29432) load
I (169) esp_image: segment 5: paddr=0003d440 vaddr=50000000 size=00010h (    16) load
I (175) boot: Loaded app from partition at offset 0x10000
I (175) boot: Disabling RNG early entropy source...
I (189) octal_psram: vendor id    : 0x0d (AP)
I (189) octal_psram: dev id       : 0x03 (generation 4)
I (190) octal_psram: density      : 0x07 (256 Mbit)
I (195) octal_psram: good-die     : 0x01 (Pass)
I (200) octal_psram: Latency      : 0x01 (Fixed)
I (205) octal_psram: VCC          : 0x00 (1.8V)
I (210) octal_psram: SRF          : 0x00 (Slow Refresh)
I (216) octal_psram: BurstType    : 0x01 ( Wrap)
I (222) octal_psram: BurstLen     : 0x03 (1024 Byte)
I (227) octal_psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (233) octal_psram: DriveStrength: 0x03 (1/8)
I (238) esp_psram: Found 32MB SPI RAM device
I (243) esp_psram: Speed: 40MHz
I (247) cpu_start: Pro cpu up.
I (251) cpu_start: Single core mode

It just repeated the above. Thanks, Thomas

Icarus113, 2022-08-22:
@TGiles1998  Hi, thanks for the updated info. 

The latest two attached logs aren't related to virtual address mapping. I think the way:
> The firmware can map as much. The rest of the PSRAM physical region will not be used. 

is working.

On the other hand, it's very hard to debug it remotely, if it's a non-official PSRAM chip. What I can do is limited..
We met similar conditions on Flash chips:

> Espressif is only doing the reviewing of changes of formats. The tests should be done by the authors, before mass production.

ESP32DE, 2022-08-27:
Hiya @TGiles1998 

which 32MB (32*1024*1024) Quad PSRAM you use?

cheers

TGiles1998, 2022-08-27:
Hi @ESP32DE, I am using custom hardware using the APS25608N-OBR-BD, I have got the 16MB APS12808L-3OBM-BA to work successfully and APS12808L-OBM-BA.

TGiles1998, 2022-08-27:
@ESP32DE @Icarus113 I switched from using octal Flash to quad flash due to lack of availability for octal flash that is less than 16MB.

TGiles1998, 2022-08-27:
@ESP32DE @Icarus113 I have made my own module the same size as an ESP32S3 Wroom that can have octal flash and PSRAM, just trying to get 32MB of PSRAM to work. Only available octal PSRAM I can find is 1.8V.

ESP32DE, 2022-08-27:
Hi @TGiles1998
so you are successfull with the Double-Data-Rate OPI Xccela PSRAM 16MB APS12808L-3OBM-BA (200MHz) && APS12808L-OBM-BA (133MHz) ? the psram check and psram test successfull - how much you drive the clk in combine with Octal Flash? I ask to exclude later complication. 

> "switched from using octal Flash to quad flash due to lack of availability for octal flash that is less than 16MB"

i suppose the test was successful at 40, 80 MHz. can you go up to 108 MHz successful with your board? 
are you dependent on 1.8 / 3.3 VD_SDIO domain on your board or can you choose Flash/PSRAM supply 1.8/3.3 free? ( 1.2V ) which combination you want use? Octal / Quad ? 

> "just trying to get 32MB of PSRAM to work. Only available octal PSRAM I can find is 1.8V"

ok. i will study your used HW parts ( Datasheet ) and will come back to this.
maybe I can help you without getting the parts, let's see. 
FYI: I use different manufacturer(s) and a few things that are not yet free on the market.
also working to push it to the limit and [more](https://twitter.com/eMbeddedHome/status/1441762148577857536) ( #[PSRAMBANK](https://twitter.com/search?q=psrambank&src=typed_query&f=live) ) 
in any case, a few code hacks are necessary. Esp-IDF is very strictly tailored to current market availability. 
but well documented and open to your own extensions.

TGiles1998, 2022-08-28:
@ESP32DE I am successful with both 16MB PSRAM, this was achieved with no current modifications to the ESP-IDF. I can choose 1.8v/3.3v with my hardware, 32MB octal PSRAM is the goal, however, quad 32MB PSRAM would also be fantastic. I welcome any help, that would be fantastic thank you!

huming2207, 2022-08-28:
Hi @TGiles1998 

> with no current modifications to the ESP-IDF

I'm also thinking to put on a 16MB PSRAM for UI rendering. May I ask what's your setup? Is it with latest master branch ESP-IDF + auto detect RAM size, i.e. set `CONFIG_SPIRAM_TYPE_AUTO` + enabling auto mode, i.e. set `CONFIG_SPIRAM_MODE_OCT`? Are you using APS12808L-3OBMx PSRAM chip?

Regards,
Jackson

ESP32DE, 2023-08-10:
> @ESP32DE I am successful with both 16MB PSRAM, this was achieved with no current modifications to the ESP-IDF. I can choose 1.8v/3.3v with my hardware, 32MB octal PSRAM is the goal, however, quad 32MB PSRAM would also be fantastic. I welcome any help, that would be fantastic thank you!

in 2023 we go 512 and Q2/24 1G MP  @Icarus113 :)
i come back to this cause @huming2207 triggered me on twitter with this theme :)
give [me few days](https://twitter.com/eMbeddedHome/status/1688590802396626950) - i am waiting for revisioned PCB + hardware and we burn this.
  
@TGiles1998 how you go on with the 128 versions? just for yours or do you have online store or any?
i mean with this, how you tracert your bus line - you use 0r or like @huming2207 without them?
i use [this in 01005 version](https://twitter.com/eMbeddedHome/status/1683590163698417665) 

engineering without castellated holes for an carier board RM 1.27
[image]

engineering with castellared holes pin and site compatible for solo2 and s3wroom
[image]

but PCBA could not smt a beta  so fast :( so switched faster to 0201 design and they build a few test things which i can share

@Icarus113 13.Oct 2023 i want drive a 256 and 512 so if any .. the hardware is here and i am on the code and near ready - 
and not public also a 1024 ( i can not name the manufacturer, but is identical with 0x0d and rewriteabel to any, its an EONP)
there also on desk MPC/MCP with sharing bus and psram&flash combined also psram with more dies and bankswitching by adresses .... does this be the solution? -..

best wishes
;-)

huming2207, 2023-08-11:
Yep I've also mentioned by @ESP32DE . 

> I'm also thinking to put on a 16MB PSRAM for UI rendering. May I ask what's your setup? Is it with latest master branch ESP-IDF + auto detect RAM size, i.e. set `CONFIG_SPIRAM_TYPE_AUTO` + enabling octal mode, i.e. set `CONFIG_SPIRAM_MODE_OCT`? Are you using APS12808L-3OBMx PSRAM chip?

I've got this working last year and it seems to be okay on 80MHz + octal mode: https://github.com/huming2207/jumbo-s3/tree/master

I was planning to try APS25608 as well, but then I've been caught up on some other stuff I need to deal with and I haven't got free time to work on this hack later. Also since that was my own personal project, it's also quite expensive for me to buy chips from Mouser. Thus I'd use my money to deal with something more cost-effective instead. But I still wish to see some designs (or even better if someone sells modules) in any one of these combos:

- ESP32S3 + one 32MB PSRAM chips and 8MB flash
- ESP32S3 + two (or three, if possible???) 16MB PSRAM chips and one 8MB flash
- ESP32S3 + one 32MB PSRAM chip and one 32MB flash **// this will be the best option 😅**

I also understand that Espressif might have put more labours working on ESP32P4 so they probably don't have time or not interested in working on large external PSRAM support for ESP32S3?  I can also wait till ESP32P4 to come on the market and I hope that one have SiP or modules with larger PSRAM. 

On the other hand, for comparison on LCSC:

- Allwinner has D1s and V3s costs around US$3.5-4.5, with 64MB DDR RAM, runs Linux or RT-Thread, or a half-cooked FreeRTOS SDK;
- ESP32-S3-WROOM-1-N8R8 is also around US$3.5-4 
- ESP32-S3-WROOM-2-N32R8V is around US$6.8-10+

Of course D1s and V3s have much worse official community support, no wireless PHY and consumes much more power. These are the main factors why we are still sticking with ESP32S3. But the RAM for those Allwinner SiPs are much larger so then for some scenarios like making HMI devices with a high-res or weird displays (like large e-paper) is way easier. Anyway, I'd wish Espressif should consider adding the support for large PSRAM for ESP32S3 or in the future for ESP32P4.   

For @ESP32DE : 

> or like @huming2207 without them? ( very hard to stable the bus > 40 MHz )

To get 80MHz, I guess you need to:

1. Keep the trace short, or you probably need to match the impedance for the trace I assume?
2. You probably also need to make those traces for those eight data lines and one clock line to be in same length. My colleagues at that time didn't think that's necessary, but I've done that for mine anyway. 

Regards,
Jackson

ESP32DE, 2023-08-15:
@huming2207 
fast reply ( summerbreak ~25 Aug'23) to keep you updated
128,256 and 512 is on desk and second 1024 in the delivery status
[image]
best wishes
;-)

ESP32DE, 2023-08-18:
@huming2207
thanks for the Dataline tips, i did so short i can do between the SoC and IC's, no problem 80Mhz and higher.  
[128Mbit work's](url) ( combine with 256Mbit flash ) 
[image]
[image]

ESP-IDF: v5.0.2-dirty

next step 256Mbit 
best wishes
;-)

ESP32DE, 2023-08-20:
@huming2207 

[image]

ESP-IDF: v5.0.2-dirty
successful

next step 512Mbit
best wishes
;-)

ESP32DE, 2023-08-21:
update 

[image]

512Mbit, it has been detected.
maximal 32MB Psram allocated, 
will check few possible ways now.
one is update to the newest ESP-IDF Version
but i stay for more handwrite codes testings on 5.0.2 dirty  for now, is more familiar to me.

best wishes
;-)