espressif/esp-idf #8748: [CRASH][v4.4][ESP32-S3] root_port_handle_events() aborts when device unplugged

chipweinberger, 2022-04-08:
ESP32-S3
ESP-IDF release/v4.4 (https://github.com/espressif/esp-idf/commit/c29343eb94d2f2ca17b3a5b38c82452e556147f2) from a few days ago.

Logs + Backtrace:
```
E (12638) USBH: Device 1 gone
W (12638) pd usb host service: [client task] device gone! deviceHandle 0x3de4e6e0
I (12648) pd midi 
abort() was called at PC 0x420541ef on core 0
0x420541ef: root_port_handle_events at /components/usb/hub.c:785

Backtrace:0x403762d3:0x3fcb55800x40380a05:0x3fcb55b0 0x4038b825:0x3fcb55e0 0x420541ef:0x3fcb5660 0x42055072:0x3fcb5690 0x4204e428:0x3fcb56c0 0x4201eb27:0x3fcb56f0 0x40385b5e:0x3fcb5720 

0x403762d3: panic_abort at /components/esp_system/panic.c:402

0x40380a05: esp_system_abort at /components/esp_system/esp_system.c:128

0x4038b825: abort at /components/newlib/abort.c:46

0x420541ef: root_port_handle_events at /components/usb/hub.c:785

0x42055072: hub_process at /components/usb/hub.c:986

0x4204e428: usb_host_lib_handle_events at /components/usb/usb_host.c:514

0x4201eb27: PDUsbHostService::usbHostTaskLoop(void*) at /jcommon/usb/PDUsbHostService.cpp:49

0x40385b5e: vPortTaskWrapper at /components/freertos/port/xtensa/port.c:131
```

Aborts here, in hub.c (last case)
```
        case HCD_PORT_EVENT_OVERCURRENT: {
            bool pass_event_to_usbh = false;
            HUB_DRIVER_ENTER_CRITICAL();
            switch (p_hub_driver_obj->dynamic.driver_state) {
                case HUB_DRIVER_STATE_ROOT_POWERED: //This occurred before enumeration
                case HUB_DRIVER_STATE_ROOT_ENUM_FAILED: //This occurred after a failed enumeration.
                    //Therefore, there's no device and we can go straight to port recovery
                    p_hub_driver_obj->dynamic.flags.actions |= HUB_DRIVER_FLAG_ACTION_PORT_RECOVER;
                    break;
                case HUB_DRIVER_STATE_ROOT_ENUM:
                    //This occurred during enumeration. Therefore, we need to recover the failed enumeration
                    p_hub_driver_obj->dynamic.flags.actions |= HUB_DRIVER_FLAG_ACTION_ENUM_EVENT;
                    p_hub_driver_obj->single_thread.enum_ctrl.stage = ENUM_STAGE_CLEANUP_FAILED;
                    break;
                case HUB_DRIVER_STATE_ROOT_ACTIVE:
                    //There was an enumerated device. We need to indicate to USBH that the device is gone
                    pass_event_to_usbh = true;
                    break;
                default:
                    abort();    //Should never occur
                    break;
            }
```

chipweinberger, 2022-04-08:
I'm hitting this pretty offten... very annoying.

chipweinberger, 2022-04-08:
I logged every call I make to USBH.

In these logs I hit the case where it does not crash, but instead usb_host_uninstall() fails.

<details>
  <summary>Detailed Logs. Plug in midi USB device, then unplug.</summary>
```
I (3962) pd usbh: openSingleDevice called
I (3972) pd usbh: [init] usb_host_install()
I (4002) pd usbh: [init] xTaskCreate(usbHostTaskLoop)

I (4002) pd usbh: [host task] started...
I (4022) pd usbh: [host task] usb_host_lib_handle_events()

I (4022) pd usbh: [init] usb_host_transfer_alloc(buf: 264, isoc: 0, ...)
I (4032) pd usbh: [init] usb_host_transfer_alloc(buf: 264, isoc: 0, ...)
I (4052) pd usbh: [init] usb_host_transfer_alloc(buf: 264, isoc: 0, ...)
I (4062) pd usbh: [init] usb_host_transfer_alloc(buf: 264, isoc: 0, ...)
I (4062) pd usbh: [init] usb_host_client_register(...)
I (4072) pd usbh: [init] clientHandle 0x3de4e68c
I (4072) pd usbh: [init] xTaskCreate(clientEventsLoop)

I (4072) pd usbh: [openSingleDevice] waiting usb connection....

//
// << This is where I plugged in the device >>
//

I (4072) pd usbh: [client task] started...
I (4092) pd usbh: [client task] usb_host_client_handle_events(client:0x3de4e68c)

I (8432) pd usbh: [host task] usb_host_lib_handle_events()
I (8492) pd usbh: [host task] usb_host_lib_handle_events()
I (8492) pd usbh: [host task] usb_host_lib_handle_events()
I (8492) pd usbh: [host task] usb_host_lib_handle_events()
I (8502) pd usbh: [host task] usb_host_lib_handle_events()
I (8502) pd usbh: [host task] usb_host_lib_handle_events()
I (8512) pd usbh: [host task] usb_host_lib_handle_events()
I (8522) pd usbh: [host task] usb_host_lib_handle_events()
I (8532) pd usbh: [host task] usb_host_lib_handle_events()
I (8532) pd usbh: [host task] usb_host_lib_handle_events()
I (8542) pd usbh: [host task] usb_host_lib_handle_events()
I (8542) pd usbh: [host task] usb_host_lib_handle_events()
I (8542) pd usbh: [host task] usb_host_lib_handle_events()
I (8562) pd usbh: [host task] usb_host_lib_handle_events()
I (8562) pd usbh: [host task] usb_host_lib_handle_events()
I (8572) pd usbh: [host task] usb_host_lib_handle_events()

I (8562) pd usbh: [client task] USB_HOST_CLIENT_EVENT - NEW_DEV! addr: 1

I (8582) pd usbh: [openSingleDevice] usb_host_device_open(0x3de4e68c, usbAddress: 1, ..)
I (8582) pd usbh: [openSingleDevice] deviceHandle 0x3de4e6e0

I (8582) pd usbh: [client task] usb_host_client_handle_events(client:0x3de4e68c)

I (8592) pd usbh [device]: usb_host_get_device_descriptor(dev:0x3de4e6e0)

Device descriptor:
Length: 18
Descriptor type: 1 (Device)
USB version: 0x110 (1.10)
Device class: 0x00 (Generic Device)
Device subclass: 0x00 ()
Device protocol: 0x00
EP0 max packet size: 16
VID: 0x09e8
PID: 0x0076
Revision number: 1.0
Manufacturer id: 1
Product id: 2
[... 147 lines cut ...]

I (13622) pd usbh: [uninitialize] called
I (13622) pd usbh: [uninitialize] Stopping xUsbClientTask...
I (13632) pd usbh: [uninitialize] usb_host_client_unblock(client:0x3de4e68c)
I (13642) pd usbh: [client task] vTaskDelete()

E (14042) pd midi usb: too many xfer fails. stopping xferInTask_loop until replugged

I (14642) pd usbh: [uninitialize] 0 devices need closing
I (14642) pd usbh: [uninitialize] usb_host_client_deregister(client:0x3de4e68c)
I (14652) pd usbh: [uninitialize] Stopping xUsbHostTask...
I (14652) pd usbh: [uninitialize] usb_host_lib_unblock()

I (14652) pd usbh: [host task] event: USB_HOST_LIB_EVENT_FLAGS - NO_CLIENTS
I (14662) pd usbh: [host task] usb_host_device_free_all()
I (14672) pd usbh: [host task] vTaskDelete()

I (15662) pd usbh: [uninitialize] usb_host_uninstall() 

 // usb_host_uninstall() fails
 // returns ESP_ERR_INVALID_STATE
 // p_host_lib_obj->dynamic.process_pending_flags != 0;

```
</details>

chipweinberger, 2022-04-08:
@Dazza0

chipweinberger, 2022-04-08:
I just tried not unintializing, leaving USBH fully "up", and doing a few unplugs & replugs. I still hit the same crash after a few tries.

chipweinberger, 2022-07-11:
Bump @Dazza0 @igrr 

This was caused by me accidentally submitting the same `usb_transfer_t` multiple times for different requests, without waiting for the transfer object to complete.

It could be worthwhile to check for this and return an error. Aborting at a "random" place is never very easy to debug. 

If not, feel free to close this.

thoughts?

Dazza0, 2022-07-20:
@chipweinberger I'm hesitant to make a copy of the `usb_transfer_t` as

- These objects can get quite large for Isochronous transfers since we need to copy all of the `usb_isoc_packet_desc_t` descriptors as well
- When the transfer is complete, the internal copy of the transfer object will need to overwrite the user's copy which could also lead to its own bugs.

IMO the following changes below. Please let me know these look OK.

- The internal `urb_t` object wraps the public `usb_transfer_t` object, thus I think the `urb_t` would be a good place to add a variable that tracks the transfer's current "progress" without needing to reveal the variable to the user.
- `usb_host_transfer_submit()` will check the "progress" variable of `urb_t` and return an error if the transfer was already submitted.
- Add a public function `usb_host_transfer_check_progress(usb_transfer_t *transfer)` to query the progress of a transfer. Users can call this before attempting to modify a transfer object.

chipweinberger, 2022-07-20:
That sounds like a good and easy solution. It'll help users find issues.

Btw, after fixing the above issue I mentioned, I recently managed to hit the same abort() again after making some other changes to my shutdown code & inserting and removing the USB device many times. It's on my todo list to debug it further and give you something more actionable.

chipweinberger, 2022-08-24:
@Dazza0, I am still sometimes hitting the abort in root_port_handle_events() in v4.4.2.

> It's on my todo list to debug it further

The abort happens because we are in "HUB_DRIVER_STATE_ROOT_RECOVERY" state, which is not handled by root_port_handle_events() switch statement

Dazza0, 2022-09-02:
@chipweinberger  Seems like a double HCD Port event. The hub only goes into the `HUB_DRIVER_STATE_ROOT_RECOVERY` state when the root port has an error (e.g., `DISCONNECTION`, `ERROR`, `OVERCURRENT`).

However, while hub driver is recovering the root port, there's another `HCD_PORT_EVENT_...` occurring, thus leading to the switch case calling the abort: Could you log what `port_event` is happening when the `abort()` is called?

>  Note: this abort happens when I unplug the USB device.

Do you know if this is happening every time you unplug the USB device? Usually, when you first unblock the USB device, a `HCD_PORT_EVENT_DISCONNECTION` event occurs which puts Hub driver into recovery. However, once disconnected, there shouldn't be any further HCD Port events.

So maybe there's an edge case missing where an `HCD_PORT_EVENT_ERROR` occurs first, followed rapidly by a `HCD_PORT_EVENT_DISCONNECTION` (it could happen the disconnection happens slowly).

@chipweinberger  BTW, could you also open a separate Github issue to track this. I want to current issue to track the `usb_transfer_t` status feature mentioned above.

chipweinberger, 2022-09-02:
> is happening every time you unplug the USB device?

No. It's about 1 out of 20 unplugs (rough guess).

> Could you log what port_event

Yes, I'll try this today. Is there a specific line of code to insert this log?

chipweinberger, 2022-09-02:
**This issue is reserved for tracking the `usb_transfer_t` feature.**

closed 2022-11-11 by espressif-bot

chipweinberger, 2022-12-05:
commit: https://github.com/espressif/esp-idf/commit/e17e0668285f5bdea2a9211e3f456981fa9eb075