espressif/esp-idf #8749: WIFI/BT(BLE) Simultaneously?

bini2007, 2022-04-08:
I'm working on a project that uses both Wifi (specifically the http_client component) as well as BLE.

They both seem to work great separately.

However, once Wifi establishes a connection and starts working (a small http request every second or so), the BLE feels starved. It takes a very long time to get connected, reads take in the order of seconds, and I often get disconnected.

Any idea what could cause this behavior?

 

Either I must be doing something wrong, or there's an issue with the stack.
Is there an example that does WIFI and BLE at the same time? Is it expected to work fine?

 

I guess,

the problem is that they share a radio; in other words, while the ESP32 is sending/receiving a BT packet, it cannot listen or send a WiFi packet.

Thanks,

xiongweichao, 2022-04-11:
Hi @bini2007 

Please refer to this [coexistence document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/coexist.html). If this issue still exists, please provide the idf commit id and the demo (include sdkconfig) that reproduces the issue.

Thanks

xiongweichao, 2022-04-12:
Hi @bini2007 

Please provide the `.elf`  file for us to debug.

Thanks

bini2007, 2022-04-12:
the elf file can be downloaded in the link [https://github.com/bini2007/esp32-elf-debug/blob/main/esp32-welle.elf](https://github.com/bini2007/esp32-elf-debug/blob/main/esp32-welle.elf)

bini2007, 2022-04-15:
@[xiongweichao](https://github.com/xiongweichao)

We wonder how the debugging is going well using elf file.

bini2007, 2022-04-18:
We debugged using JTAG debugger on vscode.
So we captured such a following screen shot, however we don't know how to fix it.

[image]

The PC shows below codes in IDF source:
void* malloc(size_t size)
{
    return heap_caps_malloc_default(size);  <--- The problem point when connection to server(slave) device with ble.
}

Then any advice!

ESP-Marius, 2023-04-26:
Sorry for the delayed follow-up on this one.

If we look at the backtrace we can see that the crash seems when we try to take a heap lock to do a malloc. The lock seems to point to random value 0x0000ff08, instead of a writeable memory area. So we get a store-prohibited. 

Typically this is due to memory corruption, something overwrite this heap allocated lock. For some general suggestion on how to debug these errors you can see [heap-corruption-detection](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-corruption-detection).

Closing this issue for now as it is quite old, but feel free to reopen it again if you are still experiencing problems.

closed 2023-05-24 by Alvin1Zhang