espressif/esp-idf #8750: WPA2 Enterprise: ESP32 not connecting to AP when AP's channel bandwidth is left at "auto" after its power-on or reset

Illuminator7, 2022-04-08:
Environment:

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.4
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:  USB

Problem Description:

When enabling WPA2-Enterprise: ESP does not connect to AP when the AP's channel bandwidth is left at value "auto" (for some models like TP-Link) after a **power-on or a reset** but successfully connects when the AP's channel bandwidth is set specifically to either 20Mhz or 40Mhz (which is what ESP supports). The problem also does no longer appear even when switching back to "auto" after changing from 20Mhz or 40Mhz. The problem only appears during after power on or reset and channel BW is left at "auto". Other devices like android phones are able to successfully connect after AP's power on or reset and channel BW is left at "auto".

Expected Behavior:

ESP should be able to connect to AP even if AP's channel bandwidth is left at "auto" after a power-on or reset of the AP when Enterprise mode is enabled like it does when just using WPA/WPA2-Personal.

Actual Behavior:

ESP cannot establish a connection to AP if AP's channel bandwidth is left at "auto" after a power on or a reset.  

Steps to reproduce

1. Restart the Enterprise network's AP. This can revert the channel BW back to "auto".
2. Power on the radius server until communication is established between AP and the radius.
3. Modify the wifi_enterprise example project of IDFv4.4 through idf.py menuconfig to set the proper login credentials.
4. Run the example.

Code to reproduce this issue

Use IDFv4.4's wifi_enterprise example project.

Debug Logs

```
After a POR (power on/reset):
entry 0x40080694
I (26) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (26) boot: compile time 18:36:16
I (26) boot: chip revision: 1
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 1, min. application chip revision: 0
I (100) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=18734h (100148) map
I (145) esp_image: segment 1: paddr=0002875c vaddr=3ffb0000 size=037d4h ( 14292) load
I (151) esp_image: segment 2: paddr=0002bf38 vaddr=40080000 size=040e0h ( 16608) load
I (158) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=73ea0h (474784) map
I (331) esp_image: segment 4: paddr=000a3ec8 vaddr=400840e0 size=101b8h ( 65976) load
I (358) esp_image: segment 5: paddr=000b4088 vaddr=50000000 size=00010h (    16) load
I (368) boot: Loaded app from partition at offset 0x10000
I (369) boot: Disabling RNG early entropy source...
I (381) cpu_start: Pro cpu up.
I (381) cpu_start: Starting app cpu, entry point is 0x40081174
0x40081174: call_start_cpu1 at H:/ESP/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (395) cpu_start: Pro cpu start user code
[... 202 lines cut ...]
I (8823) example: ~~~~~~~~~~~
I (8823) example: IP:192.168.0.246
I (8823) example: MASK:255.255.255.0
I (8823) example: GW:192.168.0.254
I (8823) example: ~~~~~~~~~~~

```

Other items if possible
 
sdkconfig can be auto-generated when opening up menuconfig.  
WIFI Router used as AP: TP-Link Archer A6-v3 AC1200
   configuration for 2.4GHz: 
                         operation mode : AP
                         security: WPA/WPA2 Enterprise
                         version: WPA2
                         encryption: AES
                         wireless mode: mixed 802.11b/g/n
                         channel width: "auto" for replicating bug, either 20Mhz or 40Mhz to allow connection
                         channel: "auto"
 Radius Server used : Freeradius 3.0 on a Raspberry Pi 3B

To be able to actually communicate with Freeradius modify the settings of sdkconfig:
 - disable mbedtls crypto functions at the supplicant
 - ESP-TLS: allow potentially insecure connections  (don't really know this is need though)

AxelLin, 2022-04-09:
Please get failure log with debug messages.
CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y
CONFIG_WPA_DEBUG_PRINT=y

Illuminator7, 2022-04-11:
Hi,

Here are the log dumps (attached) for both connecting and not-connecting scenarios.

Thanks!

[not_connecting.txt](https://github.com/espressif/esp-idf/files/8460541/not_connecting.txt)
[connecting.txt](https://github.com/espressif/esp-idf/files/8460542/connecting.txt)

nishanth-radja, 2022-04-11:
@Illuminator7 can we have the sniffer capture when the issue happens?

Illuminator7, 2022-04-12:
I also figured that you might ask for the "connecting" sniff so here it is.

From the captures, it looks like AP isn't sending the Identity request after it acknowledges ESP's EAPOL start.
[connecting_wireshark.zip](https://github.com/espressif/esp-idf/files/8471970/connecting_wireshark.zip)

Illuminator7, 2022-04-13:
Hi @AxelLin @nishanth-radja , 

After some more digging, I found out that the router might be the problem. When I said before that other devices were able to connect to the AP after a restart or a power-cycle, this was because these devices that I used can support 5G. These devices were able to establish connections in the 5G channels. But when I turned off the 5G band of the AP, it was then no more devices were able to connect just like ESP, until I do the channel width setting changes that make ESP able to connect. It seems to me that this particular router has issues with its 2.4G band when set as Enterprise and ESP has no issues with its enterprise feature (at least, in my usage case). 

Thanks for the tips. I was able to narrow down the problem to the source.

closed 2022-04-13 by Illuminator7