espressif/esp-idf #8751: how to deal with extram_bss ?

franz-ms-muc, 2022-04-08:
Hello, 

regarding: 
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-allow-bss-seg-external-memory

it works very well if i use EXT_RAM_ATTR in Front of a Variable. 

e.g. 
`static uint8_t EXT_RAM_ATTR data[200]`

very well so far. 

but, what if i want the code to be more Flexible. 
e.g. if it want to port it to other MCUs, and have a more free Option. 

i like to make 3 Sections: 

#define ISO_CORE_VARS_ATTR   __attribute__((section(".ISO_Core_Task_data")))
#define ISO_BASE_VARS_ATTR   __attribute__((section(".ISO_Base_Task_data")))
#define ISO_CLIENT_VARS_ATTR __attribute__((section(".ISO_Client_Task_data")))

so, grouping Variables into 3 Sections, and then use a *.lf File to place these 3 sections into extram_bss like the Link above says.

so instead of using EXT_RAM_ATTR  i want to use the 3 Defines above. 

i have tried as follows, but it does not place the Variables into the RAM, but to Flash. 
i did something wrong, and need some help ... 
(not understood *.lf fully)

```
[sections:ISO_Core_Task_data_section] 
entries: 
    .ISO_Core_Task_data+ 
     
[sections:ISO_Base_Task_data_section] 
entries: 
    .ISO_Base_Task_data+ 
     
[sections:ISO_Client_Task_data_section] 
entries: 
    .ISO_Client_Task_data+ 
 
 
 
[scheme:ISO_Core_Task_data_scheme] 
entries: 
    ISO_Core_Task_data_section -> extram_bss 
     
[scheme:ISO_Base_Task_data_scheme] 
entries: 
    ISO_Base_Task_data_section -> extram_bss 
     
[scheme:ISO_Client_Task_data_scheme] 
entries: 
    ISO_Client_Task_data_section -> extram_bss 
 
 
 
[mapping:ISO_Core_Task_data_types] 
archive: * 
entries: 
	* (ISO_Core_Task_data_scheme); 
		ISO_Core_Task_data_section -> extram_bss 
 
[mapping:ISO_Base_Task_data_types] 
archive: * 
entries: 
	* (ISO_Base_Task_data_scheme); 
		ISO_Base_Task_data_section -> extram_bss 
		 
[mapping:ISO_Client_Task_data_types] 
archive: * 
entries: 
	* (ISO_Client_Task_data_scheme); 
		ISO_Client_Task_data_section -> extram_bss

```

o-marshmallow, 2022-04-11:
Hi @franz-ms-muc ,

Which chip are you using? 

According to the [Linker Script Template documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/linker-script-generation.html?highlight=linker%20lf#linker-script-template), the target token in the fragment file shall be the same as the one presented with `mapping[...]` inside the linker script (ld) file.

Thus, if you are using an ESP32 or an ESP32S2, can you try replacing the `extram_bss` in your Linker Fragment file by `extern_ram`?
As the corresponding section in the `sections.ld.in` is like this:
```
  .ext_ram.bss (NOLOAD) :
  {
    _ext_ram_bss_start = ABSOLUTE(.);

    mapping[extern_ram]

    . = ALIGN(4);
    _ext_ram_bss_end = ABSOLUTE(.);
  } > extern_ram_seg
```

Alvin1Zhang, 2022-05-06:
Thanks for reporting, there is a fix in https://github.com/espressif/esp-idf/commit/5bb59b00e72f8f91eb24d8c65bf9a7ea2b8a4f5f, feel free to reopen if the issue still happens. Thanks.

closed 2022-05-06 by Alvin1Zhang