espressif/esp-idf #8752: ESP32-C3 LIBUSB error (openocd, probe.rs)

TheButlah, 2022-04-08:
## Environment

- Development Kit: N/A (custom)
- Kit version (for WroverKit/PicoKit/DevKitC): N/A (custom)
- Module or chip used:  ESP32-C3-MINI-1
- IDF version: v4.4
- Build System:  idf.py
- Compiler version: (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System: Windows 10
- (Windows only) environment type: PowerShell.
- Using an IDE? Yes, esp-idf vscode extension
- Power Supply: USB

## Problem Description

I cannot open the built-in USB interface with LIBUSB, no matter whether I use the espressif provided drivers, default windows, or zadig drivers (I tried every combo).

### Expected Behavior

I can use libusb's open function on the device. This would allow me to use [probe.rs](https://probe.rs/).

### Actual Behavior

Depending on which drivers I use (WinUSB, libusbk, usbser, espressif), I get any of the following errors:
`LIBUSB_ERROR_NOT_FOUND`
`LIBUSB_ERROR_ACCESS`

I may have also gotten these, but I'm not sure:
`LIBUSB_ERROR_NOT_FOUND`
`LIBUSB_ERROR_NOT_SUPPORTED`

The strangest part was that when I unchecked "Ignore Hubs or Composite Parents" in Zadig, and made the composite parent use WinUSB, I actually successfully opened the device with libusb, which meant that the probe.rs debugger worked! The issue with this approach was that it removed the two USBCDC interfaces and now it was only a WinUSB interface, which meant no COM port. So I was unable to flash the device at all anymore (at least using the UART setting in esp-idf vscode).

### Steps to reproduce

Have a board using the built-in JTAG/Serial over USB (aka no UART chip, just uses D+ D- pins on esp32c3 directly)

Try using either openocd (which uses libusb), probe.rs debugger (uses libusb), or maybe manually try opening the device with libusb.

### Code to reproduce this issue

cargo.toml should use `probe-rs = "0.12.0"`

```rust
fn main() {
    // Get a list of all available debug probes.
    let probes = probe_rs::Probe::list_all();
    println!("{probes:?}");

    // Use the first probe found.
    let probe = probes[0].open().unwrap();
}
```

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## See also

My [conversation](https://matrix.to/#/!vhKMWjizPZBgKeknOo:matrix.org/$uRDW1d3xBMQhpPS0BSc7hy4Kg3Xs58ONsbJn0__77nQ?via=matrix.org&via=chat.berline.rs&via=unix.house) on the probe-rs Matrix group where I debugged probe.rs and determined that it was a driver issue with libusb.

My [conversation](https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$PF3UlMgOJv9MFAIFybSrZ74IrR4NsSNaFCO6yuxwTJU?via=matrix.org&via=tchncs.de&via=matrix.coredump.ch) on the esp-rs matrix group where I troubleshooted the espressif provided drivers.

TheButlah, 2022-04-08:
Here is the error from openocd when I try to start it:

```
debug_level: 2

adapter speed: 20000 kHz

WARNING: boards/esp32-wrover.cfg is deprecated, and may be removed in a future release.
         If your board is ESP32-WROVER-KIT, use board/esp32-wrover-kit-1.8v.cfg instead.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: no device found
❌ Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
❌ Error: no device found
❌ Error: unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'

Assertion failed!

Program: D:\Ryan\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin\openocd.exe
File: ../src/jtag/core.c, Line 343

Expression: jtag_trst == 0
[Stopped] : OpenOCD Server
```

dobairoland, 2022-04-08:
Hi. Have you used the Windows tool installer (https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/get-started/windows-setup.html#esp-idf-tools-installer)? It should be able to install all the necessary drivers. But before that, please remove all of the drivers you installed earlier.

As for openocd, you used the wrong `cfg` file. It should be `board/esp32c3-builtin.cfg`.

More information: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/index.html

igrr, 2022-04-08:
Is it okay if I move this issue into openocd-esp32 repository? Given that both openocd-esp32 and probe.rs fail, this is likely a hardware issue, and esp-idf codebase is likely not yet involved.

TheButlah, 2022-04-08:
sure, feel free to move it @igrr. 
@dobairoland, yes, I tried with every combination of esp windows installer, idf-env, and all four zadig driver options. I also made sure to uninstall drivers too when I tried new options. I tried every possible permutation for at least 3 hours :)