espressif/esp-idf #8754: Development board lost connection after flash-reset

HaochengZheng, 2022-04-09:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-S2
- IDF version (run ``git describe --tags`` to find it):v4.4
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc.exe 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes. I tried PowerShell, ESP-IDF VsCode Extension and PlatformIO
- Power Supply:         USB

## Problem Description

The build can finish successfully. After entering bootloader mode by pressing boot and reset, flash can finish successfully. Then I press the reset button, the board lost connection with the computer and does not execute the program I just flashed. I used blink example for both Arduino and ESP-IDF, all configured according to the board I'm using. After compiling and flashing with Arduino, the board disconnected and reconnected and the program starts running. With ESP-IDF, it only disconnects. I tried with two boards I have, one from waveshare and one from adafruit and they are behaving the same towards ESP-IDF (In PowerShell, ESP-IDF VSCode extension and platformIO)

### Expected Behavior
In the case mentioned above, LED starts blinking.
### Actual Behavior
Compiling & flashing with Arduino, blinks. With ESP-IDF framework, it does not.

HaochengZheng, 2022-04-09:
Update: I reinstalled ESP IDF with v4.2.2 and the problem disappeared. It's possible that v4.4 has some problem

moefear85, 2022-04-10:
can you upload some code? maybe there is a logic error somewhere?

Dazza0, 2022-07-08:
@HaochengZheng

> After compiling and flashing with Arduino, the board disconnected and reconnected and the program starts running.

This seems quite unusual. Generally when ESP-IDF or Arduino finishes flashing a board, it will issue a reset to the ESP board. But this reset should **not** cause a disconnection. Just to double check, are you actually seeing a device disconnection from your PC (i.e., like if the USB cable was disconnected form the ESP board).

> Then I press the reset button, the board lost connection with the computer and does not execute the program I just flashed. 

Likewise, on most ESP boards, pressing the reset button should not cause a disconnection.

Could you share the schematics for the ESP boards you are currently using?

closed 2022-07-25 by Alvin1Zhang