espressif/esp-idf #8756: CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY on ESP32-S3

diplfranzhoepfinger, 2022-04-09:
Hello, 

there is an Option 

CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY

described here: 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-allow-bss-seg-external-memory

which leads to the Define of EXT_RAM_ATTR, and then this stuff is placed in spiram. 

--> how to achieve the same in ESP32-S3 ? 

Thanks, 

Franz

FanhuaCloud, 2022-04-09:
This problem does exist, I use heap_caps_calloc Now, you can refer to the code below
`    // StaticTask_t xTaskBuffer;
    // StackType_t *xStack = heap_caps_calloc(1024 * 4, sizeof(StackType_t), MALLOC_CAP_SPIRAM);
    // xTaskCreateStatic(dump_haps, "dump_haps", 1024 * 4, NULL, 5, xStack, &xTaskBuffer);`

diplfranzhoepfinger, 2022-04-09:
that is very clear, MALLOC\_CAP\_SPIRAM is there. 

but my Question was really about EXT\_RAM\_ATTR. 

i some Safety-restraint Applications you are tight to static allocation versus dynamic allocation.

FanhuaCloud, 2022-04-09:
> that is very clear, MALLOC_CAP_SPIRAM is there. 
> 
> but my Question was really about EXT_RAM_ATTR. 
> 
> i some Safety-restraint Applications you are tight to static allocation versus dynamic allocation.

I am also worrying about this problem

Icarus113, 2022-04-11:
There will be a commit for this soon. Will update this issue when the commit is up.

franz-ms-muc, 2022-04-25:
in which release will this go ? 4.4 or 5.0 ? or both ?

franz-ms-muc, 2022-04-26:
> > There will be a commit for this soon. Will update this issue when the commit is up.
> 
> does "Done" mean the Commit is up ?

it seems so!!!
https://github.com/espressif/esp-idf/blob/master/components/esp_system/ld/esp32s3/sections.ld.in#L394

Alvin1Zhang, 2022-05-06:
Thanks for reporting, sorry for slow turnaround, there is a fix in https://github.com/espressif/esp-idf/commit/5bb59b00e72f8f91eb24d8c65bf9a7ea2b8a4f5f, feel free to reopen if the issue still happens. Thanks.

closed 2022-05-06 by Alvin1Zhang