espressif/esp-idf #8757: 100% BUG! uart_events_example crashes on esp32s2 when using UART_NUM_1

moefear85, 2022-04-10:
## Environment

- Development Kit:  AiThinker ESP-12K
- Module or chip used:  esp32s2
- IDF version (run ``git describe --tags`` to find it): v4.4-c29343e
- Build System:       idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it) :xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Linux
- Using an IDE?: Yes : VSCode esp-idf plugin
- Power Supply:         external 3.3V

## Problem Description

I'm using the uart_events_example project. All I did was modify the involved uart to UART_NUM_1, and thus modified the pins used because the defaults (UART_PIN_NO_CHANGE) don't produce any output, thus probably behave counter-intuitively. Anyways, everything seems to work when logging is off. But when I modify sdkconfig to allow logging. Initially it looks like everything is ok, but the moment any character is sent over serial to the RX pin of UART1, it immediately crashes.

### Expected Behavior

Should not crash

### Actual Behavior

Crashes when logging is enabled

### Steps to reproduce

1. Open VSCode ESP-IDF extension, start a new project, based on the uart_events_example.
2. Modify the sdkconfig as attached below
3. Modify the uart and pin numbers as shown in the code below
4. Flash to chip
5. Open up any serial terminal connected to uart1
6. Type in any char
7. BOOM! It crashes!

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
/* UART Events Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/uart.h"
#include "driver/gpio.h"
#include "esp_log.h"

static const char *TAG = "uart_events";

/**
 * This example shows how to use the UART driver to handle special UART events.
 *
 * It also reads data from UART0 directly, and echoes it to console.
 *
 * - Port: UART0
 * - Receive (Rx) buffer: on
 * - Transmit (Tx) buffer: off
 * - Flow control: off
 * - Event queue: on
 * - Pin assignment: TxD (default), RxD (default)
 */

[... 134 lines cut ...]
assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

Backtrace:0x400297a2:0x3ffd5cd00x40030e41:0x3ffd5d00 0x4003b75a:0x3ffd5d30 0x40032a91:0x3ffd5e70 0x4008a835:0x3ffd5ed0 0x40087a5f:0x3ffd5f30 
0x400297a2: panic_abort at /home/moefear/source/esp-idf-v4.4_c29343e/components/esp_system/panic.c:402

0x40030e41: esp_system_abort at /home/moefear/source/esp-idf-v4.4_c29343e/components/esp_system/esp_system.c:128

0x4003b75a: __assert_func at /home/moefear/source/esp-idf-v4.4_c29343e/components/newlib/assert.c:85

0x40032a91: xQueueSemaphoreTake at /home/moefear/source/esp-idf-v4.4_c29343e/components/freertos/queue.c:1549 (discriminator 1)

0x4008a835: uart_read_bytes at /home/moefear/source/esp-idf-v4.4_c29343e/components/driver/uart.c:1286

0x40087a5f: uart_event_task at /home/moefear/temp/uart_events_test/build/../main/uart_events_example_main.c:57 (discriminator 9)

ELF file SHA256: 28c59bf5a473fe7f

CPU halted.
```

## Other items if possible
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8459956/sdkconfig.txt)
[uart_events_test.elf.txt](https://github.com/espressif/esp-idf/files/8459963/uart_events_test.elf.txt)

* Note: the elf file has the .txt extension added purely so github allows it to upload. please rename.

moefear85, 2022-04-10:
some testing revealed it only happens when "debug without optimizations" is chosen.

ESP-Marius, 2022-04-11:
@moefear85 is it possible that this is just stack overflow issue? `2048` seems too small for your `uart_event_task`, try increasing it to e.g. `4096` and see if that helps.

closed 2022-04-11 by moefear85