espressif/esp-idf #8758: fflush(stdout) has no effect

moefear85, 2022-04-10:
whenever I use printf(...) to print text that doesn't include any newline character, the output is delayed, because it is buffered. So I would use fflush(stdout) to immediately print to the screen. Since a while back, I think since I upgraded to v4.4, fflush(stdout) no longer has any effect. Is this a bug, or has its behavior been changed?

Alvin1Zhang, 2022-04-11:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

igrr, 2022-04-11:
@moefear85 flush(stdout) should still work, but it alone isn't sufficient to ensure that the text is sent out over UART. You also need `fsync(fileno(stdout))` to flush the underlying device buffers. Please see this comment: https://github.com/espressif/esp-idf/issues/2820#issuecomment-446051814

I have checked the code and didn't find substantial changes in this regard between v4.4 and v4.3. If you have some code to reproduce the difference in the behavior, please share it.

moefear85, 2022-04-11:
ok I didn't know about fsync(). I just attempted to try it, so I just created a fresh empty project to test it, and it is working right now using purely fflush(stdout) without fsync(), so I guess there are some hidden factors (maybe specific sdkconfig settings) that affect it. Although I've experienced it in more than one project, I don't know exactly which one. I'll come back to this when I face it again. I read #2820, however in my case, I'm just using printf, I'm not writing to any files or cards.

closed 2022-04-11 by igrr