espressif/esp-idf #8759: scan_evt timeout

bestyun-xyz, 2022-04-11:
using ESP-IDF v4.4 release ,edit with bt multi connect sample.
it work well,using wifi and bt at same time.
but a few days later,I will get log fllow
`scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout
scan_evt timeout`
...so many...
and then ,the mcu will restart
`Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x40085cb1  PS      : 0x00060635  A0      : 0x80085fee  A1      : 0x3ffd17e0  
A2      : 0x3ffc06e4  A3      : 0x3ffc05ac  A4      : 0x3ff000cc  A5      : 0x3ff00094  
A6      : 0x00000000  A7      : 0x15e3a9ad  A8      : 0x80085ca1  A9      : 0x3ffd17c0  
A10     : 0x3ff00094  A11     : 0x00000030  A12     : 0x3ff5d008  A13     : 0xe420cf97  
A14     : 0x00000064  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x40085cae:0x3ffd17e00x40085feb:0x3ffd1800 0x400859d2:0x3ffd1830 0x40085a4b:0x3ffd1850 0x40085b53:0x3ffd1870 0x4014d2da:0x3ffd18a0 0x400d6315:0x3ffd18c0 0x4016851d:0x3ffd18e0 0x401685df:0x3ffd1900 0x40168672:0x3ffd1920 0x4009b675:0x3ffd1940 0x401c9376:0x3ffd1960 0x401c93dd:0x3ffd1980 0x40099986:0x3ffd19a0 0x4009473d:0x3ffd19d0 

Core  1 register dump:
PC      : 0x401c6232  PS      : 0x00060335  A0      : 0x800d3a69  A1      : 0x3ffcd0f0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x800949f1  A5      : 0x3ffcd020  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x801269d2  A9      : 0x3ffcd0c0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x800949f1  A13     : 0x3ffcd000  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001e  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x401c622f:0x3ffcd0f00x400d3a66:0x3ffcd110 0x40092e99:0x3ffcd130 0x4009473d:0x3ffcd150 □`
I find other issus like this,but didn't resolve。how to fix this.
when the bt scan timeout,I can do something like restart bt moudle in callback.

juanaviladev, 2022-04-16:
Many issues were opened with this same log several versions ago. The bluetooth driver is not open source and it doesn't seem like the team is actively working on resolving the bug, it's very frustrating.

bestyun-xyz, 2022-04-18:
> Many issues were opened with this same log several versions ago. The bluetooth driver is not open source and it doesn't seem like the team is actively working on resolving the bug, it's very frustrating.

There's no way but to set up a watchdog to check the bt status.

bestyun-xyz, 2022-06-29:
I have add the code bellow,but it still doesn't work。
"Scan start success" will print normally.
I have no choice but to restart at 4 am.How to resolve the problem??????????
`
case ESP_GAP_BLE_SCAN_START_COMPLETE_EVT:
        //scan start complete event to indicate scan start successfully or failed
        if (param->scan_start_cmpl.status == ESP_BT_STATUS_SUCCESS) {
            ESP_LOGI(TAG, "Scan start success");
            clear_bt_watchdog();
        }else{
            ESP_LOGE(TAG, "Scan start failed");
        }
        break;
`

nishanth-radja, 2022-07-12:
@bestyun-xyz What steps were followed for this issue.Is ble and wifi scan performed in loop?
Can you attach the sdk config also.

WyattSmithTritium, 2022-11-07:
I also have this problem and need a solution to fix or at least catch the error.

idf v4.4.2

vik-gokhale, 2022-11-08:
Hi @WyattSmithTritium ,
Can you please share 
1. the logs from the device which have the crash dump and 
2. '.elf' file that is present in the "build" folder of your application?

Regards,
Vikram

juanaviladev, 2022-11-08:
> Hi @WyattSmithTritium , Could you please share the following?
> 
>     1. the logs from the device which has the crash dump and
> 
>     2. the corresponding '.elf' file that is present in the "build" folder of your application on which the crash occurred.
> 
> 
> Regards, Vikram

Logs are in the OP. In any case, this issue has been reported on multiple times for more than 1 year and no solution has been given in any case, beyond restarting with a watchdog.

WyattSmithTritium, 2022-11-11:
@juanaviladev 

Can you please share how to set up a BLE watchdog for this event?

tanhuisheng2, 2022-12-14:
I also have the same issue on ESP32 idf-v4.4.1 , and add a workaround to let it restart automatically.

juanaviladev, 2023-01-06:
@tanhuisheng2 @WyattSmithTritium 

Sorry guys, I don't have the watchdog programmed, but the idea is to reboot the device every X seconds so that if the error occurs, it stays blocked for as little time as possible.

It's a very dumb solution, but once the `scan_evt` error comes out, it can only be resolved by rebooting...

esp-cjh, 2024-10-24:
@thorrak This bug has been fixed on v4.4.7

closed 2024-12-23 by Alvin1Zhang