espressif/esp-idf #8760: idf.py gdbgui dependency problem

skycamdk, 2022-04-11:
## Environment
- IDF version: v4.4
- Operating System: Linux

## Problem Description
I'm unable to start gdbgui. It looks as if there is a dependency incompatibility. Here is the command and the result...
```
idf.py gdbgui  --gdbgui-port 5000
Executing action: gdbgui
gdbgui started as a background task 40029
Executing action: post_debug
Traceback (most recent call last):
  File "/home/trace/.espressif/python_env/idf4.4_py3.8_env/bin/gdbgui", line 5, in <module>
    from gdbgui.backend import main
  File "/home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/gdbgui/backend.py", line 25, in <module>
    from flask import (
  File "/home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/flask/__init__.py", line 19, in <module>
    from jinja2 import Markup, escape
ImportError: cannot import name 'Markup' from 'jinja2' (/home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/jinja2/__init__.py)
"gdbgui" exited with 1
```

I have followed the installation guide and all dependencies seems to be correct installed in correct versions
```
Detecting the Python interpreter
Checking "python" ...
Python 3.8.10
"python" has been detected
Installing ESP-IDF tools
Selected targets are: esp32, esp32c3, esp32s3, esp32s2
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r2-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r2-8.4.0 (already installed)
Skipping riscv32-esp-elf@esp-2021r2-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20211220 (already installed)
Installing Python environment and packages
Python 3.8.10
pip 22.0.4 from /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/pip (python 3.8)
Installing Python packages from /home/trace/esp/esp-idf/requirements.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 4)) (62.0.0)
Requirement already satisfied: click>=7.0 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 8)) (8.1.2)
Requirement already satisfied: pyserial>=3.3 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 12)) (36.0.2)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 18)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 19)) (0.28)
Requirement already satisfied: idf-component-manager>=0.2.99-beta in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 20)) (1.0.1)
Requirement already satisfied: gdbgui==0.13.2.0 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 22)) (0.13.2.0)
Requirement already satisfied: pygdbmi<=0.9.0.2 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 25)) (0.9.0.2)
Requirement already satisfied: python-socketio<5 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 28)) (4.6.1)
Requirement already satisfied: kconfiglib==13.7.1 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 30)) (13.7.1)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 33)) (1.5.4)
Requirement already satisfied: bitstring>=3.1.6 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 34)) (3.1.9)
Requirement already satisfied: ecdsa>=0.16.0 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 35)) (0.17.0)
Requirement already satisfied: construct==2.10.54 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from -r /home/trace/esp/esp-idf/requirements.txt (line 39)) (2.10.54)
Requirement already satisfied: Pygments<3.0,>=2.2.0 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/trace/esp/esp-idf/requirements.txt (line 22)) (2.11.2)
Requirement already satisfied: Flask-SocketIO<3.0,>=2.9 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/trace/esp/esp-idf/requirements.txt (line 22)) (2.9.6)
Requirement already satisfied: gevent<2.0,>=1.2.2 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/trace/esp/esp-idf/requirements.txt (line 22)) (1.5.0)
Requirement already satisfied: Flask-Compress<2.0,>=1.4.0 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/trace/esp/esp-idf/requirements.txt (line 22)) (1.11)
Requirement already satisfied: Flask<1.0,>=0.12.2 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from gdbgui==0.13.2.0->-r /home/trace/esp/esp-idf/requirements.txt (line 22)) (0.12.5)
Requirement already satisfied: cffi>=1.12 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from cryptography>=2.1.4->-r /home/trace/esp/esp-idf/requirements.txt (line 12)) (1.15.0)
Requirement already satisfied: six in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from idf-component-manager>=0.2.99-beta->-r /home/trace/esp/esp-idf/requirements.txt (line 20)) (1.16.0)
Requirement already satisfied: tqdm in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from idf-component-manager>=0.2.99-beta->-r /home/trace/esp/esp-idf/requirements.txt (line 20)) (4.64.0)
Requirement already satisfied: requests in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from idf-component-manager>=0.2.99-beta->-r /home/trace/esp/esp-idf/requirements.txt (line 20)) (2.27.1)
Requirement already satisfied: pyyaml in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from idf-component-manager>=0.2.99-beta->-r /home/trace/esp/esp-idf/requirements.txt (line 20)) (6.0)
Requirement already satisfied: semantic-version>="2.8" in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from idf-component-manager>=0.2.99-beta->-r /home/trace/esp/esp-idf/requirements.txt (line 20)) (2.9.0)
Requirement already satisfied: schema in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from idf-component-manager>=0.2.99-beta->-r /home/trace/esp/esp-idf/requirements.txt (line 20)) (0.7.5)
Requirement already satisfied: requests-toolbelt in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from idf-component-manager>=0.2.99-beta->-r /home/trace/esp/esp-idf/requirements.txt (line 20)) (0.9.1)
[... cut ...]
Requirement already satisfied: contextlib2>=0.5.5 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from schema->idf-component-manager>=0.2.99-beta->-r /home/trace/esp/esp-idf/requirements.txt (line 20)) (21.6.0)
Requirement already satisfied: MarkupSafe>=2.0 in /home/trace/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages (from Jinja2>=2.4->Flask<1.0,>=0.12.2->gdbgui==0.13.2.0->-r /home/trace/esp/esp-idf/requirements.txt (line 22)) (2.1.1)
```

Thank you for any hints on getting gdbgui up and running...

dobairoland, 2022-04-11:
Hi @skycamdk. Thanks for the report. It looks like the dependencies broke recently with release of new versions.

Once you have run `. export.sh`, could you please also run the following command?

```sh
pip install "jinja2<3.1"  "itsdangerous<2.1"
```

Does it solve your issue?

dobairoland, 2022-04-12:
Great! And thanks for the feedback. Lets keep this open until we merge a permanent fix.

dobairoland, 2022-05-09:
The linked fixes address this.