espressif/esp-idf #8761: SmartConfig (SC_TYPE_ESPTOUCH_V2) weird behaviour

UnTraDe, 2022-04-11:
## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version 4.3.2v
- Build System: idf.py

## Problem Description

SmartConfig (SC_TYPE_ESPTOUCH_V2) ignores BSSID and/or SSID encoded on the packet.

### Expected Behavior

Encoded BSSID and SSID on a SmartConfig transmission should be received on the ESP32.

### Actual Behavior

Where:
A = The WiFi a device transmits the SmartConfig on
B = A second WiFi in range
C = A third WiFi in range
X = Random SSID or BSSID that doesn't exists in air

| SSID | BSSID |     Received on ESP32      |
|------|-------|----------------------------|
| A    | A     | SSID: A BSSID: A           |
| A    | X     | SmartConfig never finishes |
| A    | B     | SSID: C BSSID: A           |
| X    | A     | SSID: A BSSID: A           |
| B    | A     | SSID: A BSSID: A           |
| B    | B     | SSID: C BSSID: A           |
| B    | C     | SmartConfig never finishes |

### Steps to reproduce

Modify the sample in `examples\wifi\smart_config` to use `SC_TYPE_ESPTOUCH_V2` and run it with the table above.

### Code to reproduce this issue

https://gist.github.com/UnTraDe/2187e5f78166b999f498a0375d8bfb67

mhdong, 2022-04-20:
Hi @UnTraDe From the code you provide to reproduce this issue, we know that you have opened the log. Could you provide the log for us to analysis？ And you can disable the esp_touch_v2_enable_crypt to see whether it is caused by encryption.

mhdong, 2022-06-09:
Hi @UnTraDe  We cannot edit SSID and BSSID. We can only use AP SSID and BSSID which connected to smart config device.

UnTraDe, 2022-06-09:
I have multiple AP's on the same VLAN. It means that they share the same SSID, but each have their own BSSID. The problem is that when I use and app or a library to broadcast SmartConfig over UDP, the packet will be trasmitted from *ALL* the APs. This means the for most packets found in the air the BSSID encoded on the SC packets will not much the BSSID of the underlying 802.11 it is transmitted on, because in all the provided libraries / EspTouch apps you can only enter one BSSID. 
Ideally it shouldn't matter and SC implemention in the ESP should return the *encoded* BSSID (and SSID + Password), but the implemention seems to rely on the BSSID in the 802.11 packet header

UnTraDe, 2022-07-14:
> Hi @UnTraDe Sorry to reply you so late. Smartconfig doesn't support this method now. You can try to use blufi to complete the same function. You can refer to https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/blufi.

Hi, you're saying that SmartConfig doesn't support multi-AP WiFi networks?

mhdong, 2022-07-14:
Hi @UnTraDe  We can only use AP SSID and BSSID which connected to smart config device. It does not mean that SmartConfig doesn't support multi-AP. It means that you can't SmartConfig connected To A, want ESP32 connect to B, even if A and B have same ssid.

UnTraDe, 2022-07-14:
> Hi @UnTraDe We can only use AP SSID and BSSID which connected to smart config device. It does not mean that SmartConfig doesn't support multi-AP. It means that you can't SmartConfig connected To A, want ESP32 connect to B, even if A and B have same ssid.

Thanks for the clarification! 
So, what is the purpose of filling the BSSID field in the SmartConfig packet if it is taken from the underlying 802.11 anyways?

mhdong, 2022-08-09:
Hi @UnTraDe  The purpose of filling the BSSID field in the SmartConfig packet is to make sure that ESP32 and smart config device connect to the same AP.