espressif/esp-idf #8762: usb host: cdc_acm_host_install() give user callback of "New device connected event"

zhuyeaini9, 2022-04-12:
Hi all:
when I use the cdc_acm_host_install() function,I find it use the default callback of usb event: usb_event_cb
`   // Register USB Host client
    usb_host_client_handle_t usb_client = NULL;
    const usb_host_client_config_t client_config = {
        .is_synchronous = false,
        .max_num_event_msg = 3,
        .async.client_event_callback = **usb_event_cb**,
        .async.callback_arg = NULL
    };`
this callback can handle new device connect event:
`static void usb_event_cb(const usb_host_client_event_msg_t *event_msg, void *arg)
{
    switch (event_msg->event) {
    case USB_HOST_CLIENT_EVENT_NEW_DEV:
        ESP_LOGD(TAG, "New device connected");
        break;
......
}`

but seems no chance for user to handle it!
and I'd like to open the device after receive the event "New device connected".
any advice is welcome,correct me if I'm wrong.

chegewara, 2022-04-12:
The code you are referring to is just an example, so you can edit it in your app.
For example you can add event loop in `usb_event_cb` or QueueSend or just call your own function.

zhuyeaini9, 2022-04-21:
Hi all:
I find one possible issue of the new device connect event.
in the main function,I use one cdc_acm_driver_config with new device callback,
then do the loop without open it.I want to open it in the new device callback.
and the open device code is same as original.I remove the open device code in the original
new connect event.
correct me if I'm wrong.
//----------------------------------------------------------------------------------------------
static void usb_event_cb(const usb_host_client_event_msg_t *event_msg, void *arg)
{
    switch (event_msg->event) {
    case USB_HOST_CLIENT_EVENT_NEW_DEV:
        ESP_LOGD(TAG, "New device connected");
        if (p_cdc_acm_obj->new_dev_cb) {
            /*
            usb_device_handle_t new_dev;
            if (usb_host_device_open(p_cdc_acm_obj->cdc_acm_client_hdl, event_msg->new_dev.address, &new_dev) != ESP_OK) {
                ESP_LOGW(TAG, "Couldn't open the new device");
                break;
            }
            assert(new_dev);
            */
            p_cdc_acm_obj->new_dev_cb(NULL);
            //usb_host_device_close(p_cdc_acm_obj->cdc_acm_client_hdl, new_dev);
        }
        break;
.................
.................
}
//---------------------------------------------------------------------
main()
{
cdc_acm_host_driver_config_t cdc_acm_driver_config = {
        .driver_task_stack_size = 4096,
        .driver_task_priority = 10,
        .xCoreID = 0,
        .new_dev_cb = **new_usb_connect**};
    ESP_LOGI(TAG, "Installing CDC-ACM driver");
    ESP_ERROR_CHECK(cdc_acm_host_install(&cdc_acm_driver_config));

while (1)
    {
        if (xSemaphoreTake(device_disconnected_sem, portMAX_DELAY))
        {
            //delete vcp;
            //goto open_device;
        }
    }
}
//---------------------------------------------------------------------
void new_usb_connect(usb_device_handle_t usb_dev)
{
    const cdc_acm_host_device_config_t dev_config = {
        .connection_timeout_ms = 10000,
        .out_buffer_size = 64,
        .event_cb = handle_event,
        .data_cb = handle_rx,
        .user_arg = NULL};

    FT23x *vcp;
    try
    {
        ESP_LOGI(TAG, "Opening FT232 UART device");
        vcp = new FT23x(FTDI_FT231_PID, &dev_config);
    }
    catch (esp_err_t err)
    {
        ESP_LOGE(TAG, "The required device was not opened.\nExiting...");
        return;
    }

    ESP_LOGI(TAG, "Setting up line coding");
    cdc_acm_line_coding_t line_coding = {
        .dwDTERate = EXAMPLE_BAUDRATE,
        .bCharFormat = EXAMPLE_STOP_BITS,
        .bParityType = EXAMPLE_PARITY,
        .bDataBits = EXAMPLE_DATA_BITS,
    };
    ESP_ERROR_CHECK(vcp->line_coding_set(&line_coding));

    vTaskDelay(100);
    char cmd[] = {"RM"};
    ESP_ERROR_CHECK(vcp->tx_blocking((uint8_t *)cmd, strlen(cmd)));
}
//---------------------------------------------------------------------------------
but it report error:
I (313) VCP example: Installing USB Host
I (343) VCP example: Installing CDC-ACM driver
I (4053) VCP example: Opening FT232 UART device
E (9053) cdc_acm: cdc_acm_reset_transfer_endpoint(241):
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4037908c
0x4037908c: _esp_error_check_failed at C:/code/feature_usb_host_cmd/esp-idf/components/esp_system/esp_err.c:42

file: "/IDF/examples/peripherals/usb/host/cdc/common/cdc_acm_host/cdc_acm_host.c" line 839
func: cdc_acm_host_close
expression: usb_host_interface_release(p_cdc_acm_obj->cdc_acm_client_hdl, cdc_dev->dev_hdl, cdc_dev->data.intf_desc->bInterfaceNumber)

abort() was called at PC 0x4037908f on core 0
0x4037908f: _esp_error_check_failed at C:/code/feature_usb_host_cmd/esp-idf/components/esp_system/esp_err.c:43

Backtrace:0x40375bce:0x3fce29e00x40379099:0x3fce2a00 0x4037fab6:0x3fce2a20 0x4037908f:0x3fce2a90 0x42007f5a:0x3fce2ab0 0x42006fad:0x3fce2ad0 0x42007309:0x3fce2b80 0x42009327:0x3fce2bb0 0x4200737e:0x3fce2be0 0x4037c265:0x3fce2c00
0x40375bce: panic_abort at C:/code/feature_usb_host_cmd/esp-idf/components/esp_system/panic.c:409

0x40379099: esp_system_abort at C:/code/feature_usb_host_cmd/esp-idf/components/esp_system/esp_system.c:135

0x4037fab6: abort at C:/code/feature_usb_host_cmd/esp-idf/components/newlib/abort.c:46

0x4037908f: _esp_error_check_failed at C:/code/feature_usb_host_cmd/esp-idf/components/esp_system/esp_err.c:43

0x42007f5a: cdc_acm_host_close at C:/code/feature_usb_host_cmd/esp-idf/examples/peripherals/usb/host/cdc/common/cdc_acm_host/cdc_acm_host.c:839 (discriminator 1)

0x42006fad: CdcAcmDevice::close() at c:\code\feature_usb_host_cmd\esp-idf\examples\peripherals\usb\host\cdc\cdc_acm_vcp_test\build/../../common/cdc_acm_host/include/usb/cdc_acm_host.h:302
 (inlined by) CdcAcmDevice::~CdcAcmDevice() at c:\code\feature_usb_host_cmd\esp-idf\examples\peripherals\usb\host\cdc\cdc_acm_vcp_test\build/../../common/cdc_acm_host/include/usb/cdc_acm_host.h:281
 (inlined by) esp_usb::FT23x::FT23x(unsigned short, cdc_acm_host_device_config_t const*, unsigned char) at c:\code\feature_usb_host_cmd\esp-idf\examples\peripherals\usb\host\cdc\cdc_acm_vcp_test\build/../main/ftdi_usb.hpp:40
 (inlined by) new_usb_connect(usb_device_handle_s*) at c:\code\feature_usb_host_cmd\esp-idf\examples\peripherals\usb\host\cdc\cdc_acm_vcp_test\build/../main/cdc_acm_vcp.cpp:89

0x42007309: usb_event_cb at C:/code/feature_usb_host_cmd/esp-idf/examples/peripherals/usb/host/cdc/common/cdc_acm_host/cdc_acm_host.c:1033

0x42009327: usb_host_client_handle_events at C:/code/feature_usb_host_cmd/esp-idf/components/usb/usb_host.c:761

0x4200737e: cdc_acm_client_task at C:/code/feature_usb_host_cmd/esp-idf/examples/peripherals/usb/host/cdc/common/cdc_acm_host/cdc_acm_host.c:214

0x4037c265: vPortTaskWrapper at C:/code/feature_usb_host_cmd/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141

tore-espressif, 2022-04-22:
In your `new_usb_connect` callback you can't call any CDC-ACM driver API. You should only flag to your main task that a new device is connected and the main task can open the CDC device.

I'll make it clearer in the code and docu when this makes it to our `master` branch.

zhuyeaini9, 2022-04-22:
got it.
I use task to complete it.
thanks.

closed 2022-04-22 by zhuyeaini9

Darshnik253, 2022-11-08:
Hello @zhuyeaini9 @tore-espressif ,
Currently working on ESP32S3 Dev Kit C1, it is working as a USB Host and another ESP32S3 Dev kit C1 as a USB Device.
Using example of CDC_ACM_HOST version 4.4.2,
If i connected USB device with USB host then host recognized device and generate NEW_Device_Connected event. But whenever i disconnect and reconnect usb device, then device does not recognized by USB Host.

I have tried with above reference code and given link for new device detection 
https://github.com/tore-espressif/esp-idf/commit/0428efa4ad6fd9e4fc9e2abcf58afae331bba57b 

But not USB Host not found VID and PID or cdc_acm_host_open() api not opening usb port.

Can anyone help me out on this to open usb port w/o rebooting the host?

tore-espressif, 2022-11-08:
To everyone seeing this in the future: Please open new issues for new problems, if no other issue describes the problem you see, rather than replying here, we've locked this not because we don't care about issues but because we do care about them, and tracking a number of discussions in a long thread makes it hard for us to address them all.

Darshnik253, 2022-11-08:
Thanks @tore-espressif for the suggestion. As you mention i have open new issue "ESP32S3 USB CDC Communication"
Which describes the problem. So can you help me out in it?