espressif/esp-idf #8763: latest idf.py missing include path?

jonshouse1, 2022-04-12:
I've installed the latest esp-idf and esp-iot-solution code.

I can't get idf.py to build my project, I can see the issue, an include (-I /home/jon/esp-idf/components/driver/include) seems to be missing. I do not know enough about the build system to fix it, the project used to build on the older idf versions.

Using 
ESP-IDF v5.0-dev-2393-gfaed2a44aa

Have I missed a step, does anyone know how to fix this?, thanks.

```
[6/374] Building C object esp-idf/touch_panel/CMakeFiles/__idf_touch_panel.dir/calibration/touch_calibration.c.obj
FAILED: esp-idf/touch_panel/CMakeFiles/__idf_touch_panel.dir/calibration/touch_calibration.c.obj
/home/jon/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -Iconfig -I/home/jon/esp/esp-iot-solution/components/display/touch_panel -I/home/jon/esp/esp-iot-solution/components/display/touch_panel/xpt2046 -I/home/jon/esp/esp-iot-solution/components/display/touch_panel/ft5x06 -I/home/jon/esp/esp-iot-solution/components/display/touch_panel/ns2016 -I/home/jon/esp/esp-iot-solution/components/display/touch_panel/calibration/basic_painter -I/home/jon/esp/esp-iot-solution/components/display/touch_panel/calibration/basic_painter/fonts -I/home/jon/esp/esp-iot-solution/components/display/touch_panel/calibration -I/home/jon/esp-idf/components/newlib/platform_include -I/home/jon/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/jon/esp-idf/components/freertos/esp_additions/include/freertos -I/home/jon/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/jon/esp-idf/components/freertos/esp_additions/include -I/home/jon/esp-idf/components/esp_hw_support/include -I/home/jon/esp-idf/components/esp_hw_support/include/soc -I/home/jon/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/jon/esp-idf/components/esp_hw_support/port/esp32/. -I/home/jon/esp-idf/components/esp_hw_support/port/esp32/private_include -I/home/jon/esp-idf/components/heap/include -I/home/jon/esp-idf/components/log/include -I/home/jon/esp-idf/components/lwip/include/apps -I/home/jon/esp-idf/components/lwip/include/apps/sntp -I/home/jon/esp-idf/components/lwip/lwip/src/include -I/home/jon/esp-idf/components/lwip/port/esp32/include -I/home/jon/esp-idf/components/lwip/port/esp32/include/arch -I/home/jon/esp-idf/components/soc/include -I/home/jon/esp-idf/components/soc/esp32/. -I/home/jon/esp-idf/components/soc/esp32/include -I/home/jon/esp-idf/components/hal/esp32/include -I/home/jon/esp-idf/components/hal/include -I/home/jon/esp-idf/components/hal/platform_port/include -I/home/jon/esp-idf/components/esp_rom/include -I/home/jon/esp-idf/components/esp_rom/include/esp32 -I/home/jon/esp-idf/components/esp_rom/esp32 -I/home/jon/esp-idf/components/esp_common/include -I/home/jon/esp-idf/components/esp_system/include -I/home/jon/esp-idf/components/esp_system/port/soc -I/home/jon/esp-idf/components/esp_system/port/include/private -I/home/jon/esp-idf/components/xtensa/include -I/home/jon/esp-idf/components/xtensa/esp32/include -I/home/jon/esp-idf/components/vfs/include -I/home/jon/esp-idf/components/esp_wifi/include -I/home/jon/esp-idf/components/esp_event/include -I/home/jon/esp-idf/components/esp_netif/include -I/home/jon/esp-idf/components/esp_eth/include -I/home/jon/esp-idf/components/esp_phy/include -I/home/jon/esp-idf/components/esp_phy/esp32/include -I/home/jon/esp-idf/components/esp_timer/include -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/ili9341 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/ili9342 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/st7789 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/st7796 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/nt35510 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/ili9806 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/ili9486 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/ili9488 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/ssd1351 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/rm68120 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/ssd1306 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/ssd1307 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/ssd1322 -I/home/jon/esp/esp-iot-solution/components/display/screen/controller_driver/ssd1963 -I/home/jon/esp/esp-iot-solution/components/display/screen/interface_driver -I/home/jon/esp/esp-iot-solution/components/display/screen -I/home/jon/esp/esp-iot-solution/components/bus/include -I/home/jon/esp-idf/components/nvs_flash/include -I/home/jon/esp-idf/components/spi_flash/include -mlongcalls -Wno-frame-address    -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=/home/jon/test/esp32_vncc=. -fmacro-prefix-map=/home/jon/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0-dev-2393-gfaed2a44aa\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/touch_panel/CMakeFiles/__idf_touch_panel.dir/calibration/touch_calibration.c.obj -MF esp-idf/touch_panel/CMakeFiles/__idf_touch_panel.dir/calibration/touch_calibration.c.obj.d -o esp-idf/touch_panel/CMakeFiles/__idf_touch_panel.dir/calibration/touch_calibration.c.obj   -c /home/jon/esp/esp-iot-solution/components/display/touch_panel/calibration/touch_calibration.c
In file included from /home/jon/esp/esp-iot-solution/components/display/screen/interface_driver/scr_interface_driver.h:23,
                 from /home/jon/esp/esp-iot-solution/components/display/screen/screen_driver.h:18,
                 from /home/jon/esp/esp-iot-solution/components/display/touch_panel/calibration/touch_calibration.c:18:
/home/jon/esp/esp-iot-solution/components/bus/include/i2s_lcd_driver.h:17:10: fatal error: driver/i2s.h: No such file or directory
 #include "driver/i2s.h"
          ^~~~~~~~~~~~~~
compilation terminated.
[7/374] Performing build step for 'bootloader'
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/jon/test/esp32_vncc/sdkconfig
-- Adding linker script /home/jon/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/jon/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/jon/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/jon/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/jon/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/jon/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script /home/jon/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32 esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /home/jon/esp-idf/components/bootloader /home/jon/esp-idf/components/bootloader_support /home/jon/esp-idf/components/efuse /home/jon/esp-idf/components/esp32 /home/jon/esp-idf/components/esp_common /home/jon/esp-idf/components/esp_hw_support /home/jon/esp-idf/components/esp_rom /home/jon/esp-idf/components/esp_system /home/jon/esp-idf/components/esptool_py /home/jon/esp-idf/components/freertos /home/jon/esp-idf/components/hal /home/jon/esp-idf/components/log /home/jon/esp-idf/components/bootloader/subproject/main /home/jon/esp-idf/components/bootloader/subproject/components/micro-ecc /home/jon/esp-idf/components/newlib /home/jon/esp-idf/components/partition_table /home/jon/esp-idf/components/soc /home/jon/esp-idf/components/spi_flash /home/jon/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jon/test/esp32_vncc/build/bootloader
[1/1] cd /home/jon/test/esp32_vncc/build/bootloader/esp-idf/esptool_py && /home/jon/.espressif/python_env/idf5.0_py3.8_env/bin/python /home/jon/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/jon/test/esp32_vncc/build/bootloader/bootloader.bin
Bootloader binary size 0x65e0 bytes. 0xa20 bytes (9%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

0xjakob, 2022-04-12:
@jonshouse1 Did you change to the newest version recently? If yes, have you tried `idf.py fullclean` and `idf.py reconfigure` before building again?

o-marshmallow, 2022-04-12:
Hi @jonshouse1 ,

Are you trying to port one of your project wirtten for a previous verison of IDF? In IDF v5.0, `driver` component is not automatically linked anymore. Please add the following parameter to your project's `idf_component_register` call, in your `CMakeLists.txt`:
```
 PRIV_REQUIRES driver
```
so you'll get something like this:
```
idf_component_register(SRCS "your_src"
                    INCLUDE_DIRS  "your_include_path"
                    PRIV_REQUIRES driver)
```

Check the [Migration Guide for more info](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/build-system.html#component-dependencies-that-shall-be-explicit)

igrr, 2022-04-12:
Given that the component in question is in esp-iot-solution (esp-iot-solution/components/bus), I suppose this CMakeLists.txt issue is something for Espressif to fix.

jonshouse1, 2022-04-12:
> Hi @jonshouse1 ,
>
> Are you trying to port one of your project wirtten for a previous verison of IDF? In IDF v5.0, `driver` component is not automatically linked anymore. Please add the following [...]

Thanks, I will go and read that.  
The build instructions seem to raise the bar somwehat, I have little to no desire to learn the innards and eccentricities of yet another over clever build system.  From a "dumb user who only knows a bit of C" perspective "It worked, I upgraded, it broke" :-(

Current CMakeLists.txt is

```
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
include($ENV{IOT_SOLUTION_PATH}/component.cmake)
project(ethernet_basic)

```

jonshouse1, 2022-04-12:
> Given that the component in question is in esp-iot-solution (esp-iot-solution/components/display/), I suppose this CMakeLists.txt issue is something for Espressif to fix.

I hope so.  Interestingly it seems to be the IDF drivers include that is missing rather than the IOT, I was wondering if I missed a step installing the new IDF but I can't see it if I did ?

I tried changing the order of the two includes in CMakeLists.txt.  It still errors but this time for i2c.h rather than i2s.h.

I assume it needs these as iot-solution has no headers of the same name.
```
jon@jon-VirtualBox:~/esp-idf$ pwd
/home/jon/esp-idf
jon@jon-VirtualBox:~/esp-idf$ find . |grep 'driver/i2c.h'
./components/driver/include/driver/i2c.h
```

igrr, 2022-04-12:
> From a "dumb user who only knows a bit of C" perspective "It worked, I upgraded, it broke" :-(

Unfortunately the reason is that in ESP-IDF `master` branch, development of the upcoming v5.0 release is happening. Since this is a major version change from v4.4, we are using this as an opportunity to implement some of the pending breaking changes in IDF. `esp-iot-solution` project hasn't had time yet to adapt to these breaking changes, hence you are seeing this error.

I simple solution might be to stick with release/v4.4 branch of ESP-IDF as the latest "stable" version for the time being.

If you want to try to fix the issue, I think it might be sufficient to add `REQUIRES driver` in the following esp-iot-solution file:
https://github.com/espressif/esp-iot-solution/blob/ce1d460938824199535dd84b6e3f5427ab410dc3/components/bus/CMakeLists.txt#L2
so the file would end up looking like
```
idf_component_register(SRC_DIRS "." 
                        INCLUDE_DIRS "include" 
                        REQUIRES driver)
```

igrr, 2022-04-12:
> OK, I followed the guide here, Step 2:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html
Using git you seem to end up with latest version of everything. Would it not make more sense for the links in the document to point to the latest stable release. Or did I do it wrong, if so I am not sure how ?

I'm sorry that it's confusing! You didn't do anything wrong.

Each ESP-IDF documentation version contains instructions for setting up ESP-IDF from the related ESP-IDF branch. `latest` documentation corresponds to the `master` branch; `stable` documentation corresponds to the latest stable release (v4.4 at the time of writing).

For example, the getting started guide link for the stable version is here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/index.html

Different versions of ESP-IDF are explained in the [versions chapter](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html#which-version-should-i-start-with).

We also try to add notices at the top of the getting started guide to announce the version of IDF the documentation is written for:

[image]
[image]

> I tried the change to iot-solution CMakeLists. It builds a lot further but I guess something is broken.

Yep, more changes are necessary to get esp-iot-solution to build with IDF v5.0.

By the way, according to the esp-iot-solution readme, the `master` branch of that project is supposed to work with IDF release/v4.3 branch: https://github.com/espressif/esp-iot-solution#versions. So I am not sure it is going to work with IDF v4.4 as I have suggested above. To see installation instructions for release/v4.3 branch of ESP-IDF, please go to https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/get-started/index.html.

jonshouse1, 2022-04-12:
> By the way, according to the esp-iot-solution readme, the `master` branch of that project is supposed to work with IDF release/v4.3 branch: https://github.com/espressif/esp-iot-solution#versions. So [...]

Ok ,thanks.
The problem is partially mine. I have always struggled with this style of documentation, I would just prefer two plain text files "IDF Install latest.txt" "IDF install stable.txt" - no visuals, no menus, no html,  just plain text!  I know that puts me in tiny tiny minority as well as shows my age ;-)  The "better structured" documentation seems for the under 30s the less usable it seems to be for the  over 50s ... I guess it is a "Pre GUI" and "Post GUI" brain. Ironic considering the project, no touch screens in my childhood :-)

Thanks everyone for the help, I will start again with a fresh install and the version 4.3 link provided.

jonshouse1, 2022-04-13:
Project did not build on 4.3 (IP_EVENT_TH_LOST_IP undeclared) but does build on 4.4

The point of the exercise was to re-test my code to see if it was any more stable.  My previous build environment I made some changes to the IDF code I was using to manually patch some things. I am doing a clean install of the development tools to see if the following issue is now fixed, I suspect the changes made improve it but ultimately do not fix it.
https://github.com/espressif/esp-idf/issues/7380

Do any fixes get back ported into older IDF versions, if not then isn't the iot-solution code pointless as important changes to the the IDF versions render it unable to build the iot-solution code ? Going back in IDF version enables iot code to build sure, but then the bugs remain (in this case a lack of stability with ethernet PHY and SPI when used at the same time).
 
I don't see why if Espressif want extra device support they don't simply stick it all in the IDF rather than a separate poorly supported iot-solution project. I understand the wisdom of developing the code in different trees, but then it would seem more sensible to merge and test (in a semi automated manner) into a single stable IDF and release that, after all they manage to do that with freertos, though that is pretty mature.

igrr, 2022-04-13:
> Do any fixes get back ported into older IDF versions

@jonshouse1 Sure, we can backport https://github.com/espressif/esp-idf/commit/a6ca3d39a855376649d915e0b0d54438295846dd to v4.3.x release branch. I'll ping the developer of that fix.

Regarding your last point, we would of course also like to see esp-iot-solution being maintained more actively and being kept up to date with esp-idf. That is not something esp-idf project team has a lot of influence on, though.

jonshouse1, 2022-04-13:
> > Do any fixes get back ported into older IDF versions
> 
> @jonshouse1 Sure, we can backport [a6ca3d3](https://github.com/espressif/esp-idf/commit/a6ca3d39a855376649d915e0b0d54438295846dd) to v4.3.x release branch. I'll ping the developer of that fix.

And 4.4 ! 

> 
> Regarding your last point, we would of course also like to see esp-iot-solution being maintained more actively and being kept up to date with esp-idf. That is not something esp-idf project team has a lot of influence on, though.

Hmmm, at the end of the day if the IDF team chose to do so they could import the current iot-solution code, fix it up and simply wait for the people maintaining iot to switch over to your tree or push changes forward to yourselves, but maybe such thinking is why I have few friends ;-)

igrr, 2022-04-13:
> And 4.4 !

Yes, of course. The fix mentioned wasn't backported initially because it also introduced a breaking change in how Ethernet driver is configured. We'll see about making a smaller fix that doesn't introduce breaking changes to the API and will backport that to release/v4.4 and release/v4.3.

I'm going to close this issue since the original problem is now resolved. Please don't hesitate to open a new issue if you run into anything else!

closed 2022-04-13 by igrr

jonshouse1, 2022-04-13:
>We'll see about making a smaller fix that doesn't introduce breaking changes to the API and will backport that to release/v4.4 and release/v4.3.

Great news.

>I'm going to close this issue since the original problem is now resolved.

Ok, sounds fair.
The support and help from everyone who posted was much appreciated, thanks.