espressif/esp-idf #8764: crash after calling esp_restart

chegewara, 2022-04-12:
Hi,
sorry if its not a bug and i am just stupid, but since when esp32 is still running my code after calling `esp_restart()`?

The problem i have is the app crash after calling `esp_restart()`:
Logs:
```
about to call esp_restart()

<---- HERE i call esp_restart()

[ 37172][I][WiFi_Task.cpp:47] wifi_event_handler(): station disconnected
E (37175) TRANSPORT_BASE: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (37176) MQTT_CLIENT: Poll read error: 119, aborting connection
DEBUG: MQTT_Task: ln64: MQTT Disconnect

assertion "pxQueue" failed: file "IDF/components/freertos/queue.c", line 770, function: xQueueGenericSend

abort() was called at PC 0x401acfeb on core 1

Backtrace:0x4010efbd:0x3ffeff500x40094a81:0x3ffeff70 0x4009ac69:0x3ffeff90 0x401acfeb:0x3fff0010 0x400955c2:0x3fff0040 0x400da6d9:0x3fff0080 0x40140c4d:0x3fff00a0 0x40140c91:0x3fff00c0 0x40140f26:0x3fff00e0 0x40141d61:0x3fff0100
```
Backtrace decoded:
```
Decoding stack results
0x4010efbd: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x40094a81: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 129
0x4009ac69: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c line 46
0x401acfeb: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c line 62
0x400955c2: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 770
0x400da6d9: mqtt_event_handler(esp_mqtt_event_t*) at src/MQTT_Task.cpp line 163
0x40140c4d: esp_mqtt_dispatch_event at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c line 922
0x40140c91: esp_mqtt_dispatch_event_with_msgid at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c line 913
0x40140f26: esp_mqtt_abort_connection at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c line 686
0x40141d61: esp_mqtt_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c line 1488
```

I am using arduino in this case, but i believe its esp-idf problem. As we can see it is still trying to run `xQueueGenericSend`, but queue handle is no longer valid and its causing crash.

- board: esp32
- arduino-esp32: v2.0.0

Thanks

igrr, 2022-04-12:
@chegewara could you please check what's inside `mqtt_event_handler(esp_mqtt_event_t*) at src/MQTT_Task.cpp line 163`? Looks like possible part of your application code, judging by the path?

When esp_restart is called, it doesn't immediately stop all other tasks in the system. Some functions which are registered to run before shutdown are executed first. For example, if Wi-Fi was enabled, esp_wifi_stop (or similar) will be executed and the network interfaces will be deinitialized. If your event handler triggers MQTT disconnection, you need to make sure it's not going to try to reconnect when the disconnection is caused by the system getting restarted.

chegewara, 2022-04-12:
Hi @igrr 
it is 
```
    case MQTT_EVENT_DISCONNECTED:
        xQueueSendToBack(Indicator_Queue, &INDICATOR_MODE, 0xFFFFFFF); // line 163 in my code
        break;
```` 
And i can assure you this code is working very good unless i call `esp_restart`.

It is just a part of story i described here, because its easier to show ith eapressif mqtt native code sending `MQTT_EVENT_DISCONNECTED` after calling `esp_restart`, because initially it was just crashing in another task i have completely not related to code running `esp_restart`. Thats why i decided to add this `vTaskDelete` workaround:
```
        extern bool about_reset;
        if(about_reset) vTaskDelete(NULL);
        if (uxQueueMessagesWaiting(LED_Mode_Queue) != 0) <---- it was crashing HERE
```
This solved my initial problem, but crash jumped to mqtt task described above. And funny thing is that when i add similar workaround in that mqtt event (MQTT_EVENT_DISCONNECTED) the crash is again in this line ignoring `vTaskDelete`:
` if (uxQueueMessagesWaiting(LED_Mode_Queue) != 0) <---- it was crashing HERE`

BTW it is app which is running bunch of tasks, most pinned to core 0

Thanks

chegewara, 2022-04-12:
Hi,
yes, i thought about making example based on hello_world anyway. 
What i had problem to understand the code continue after this:
```
void IRAM_ATTR esp_restart(void)
{
    for (int i = SHUTDOWN_HANDLERS_NO - 1; i >= 0; i--) {
        if (shutdown_handlers[i]) {
            shutdown_handlers[i]();
        }
    }

#ifdef CONFIG_FREERTOS_SMP
    //Note: Scheduler suspension behavior changed in FreeRTOS SMP
    vTaskPreemptionDisable(NULL);
#else
    // Disable scheduler on this core.
    vTaskSuspendAll();
#endif // CONFIG_FREERTOS_SMP
...
```
but probably `vTaskSuspendAll` does not stop freertos code responsible for handling queues. 
I believe it is freertos problem, not the esp-idf, but i just wanted to make sure its really a problem not something stupid i do in my code.

Maybe i will just call `assert(0)` instead, because in both cases it is something that work, but logs make me nervous.

Thanks

igrr, 2022-04-12:
`vTaskSuspendAll()` disables task switching on the current CPU (where esp_restart was called). A bit later in `esp_restart_noos`, the other CPU is stalled and shortly after it is reset.

There is no guarantee that immediately after esp_restart is called, all execution of the code will stop. Shutdown handlers invoked by esp_restart may run for some time. During this time, other tasks in your code also get a chance to run, depending on their priorities.

Let me illustrate this with an example:
1. Suppose that esp_restart is called from a task with priority 1.
2. One of the `shutdown_handlers` shuts down the network interfaces. This results in a disconnection event being reported to the event loop.
3. The event loop task is usually of a higher priority than 1, so it will preempt the task originally calling `esp_restart` and run the event callback. The restart process will not continue until that task finishes dispatching events and gets blocked.
4. If the event callback sends something to another queue, and the task blocked on that queue has higher priority than 1, that other task will get unblocked and will execute. And so on.
5. Only when all higher priority tasks finish their work, execution will return to the priority 1 task and restart process will continue.

closed 2022-05-05 by chegewara